raspberrypi/linux #3836: Kernel Module Request: CHACHA20POLY1305

gstreeter, 2020-09-04:
Please add the following kernel modules to the build in order to support the CHACHA20POLY1305 ESP crypto algorithm for IPSec connections:

chacha20poly1305
chacha_generic
poly1305_generic

These modules are configured with the CONFIG_CRYPTO_CHACHA20POLY1305 kernel build option

Justification: IPSec encapsulated security protocol (ESP) crypto is carried out by the Linux kernel itself. AES used to be the de-facto standard IPSec encryption algorithm but the newer CHACHA20-POLY1305 AEAD algorithm is supplanting AES for mobile and low power devices such as the RPI which lack hardware instruction support for AES. CHACHA20-POLY1305 has a smaller processor demand than AES which should give higher bandwidth.

I've complied the modules locally (branch 5.4.y) and confirmed they are used with a Strongswan configured IPSec VPN tunnel which otherwise cannot be opened when using a CHACHA20-POLY1305 ESP configuration

pelwell, 2020-09-07:
This option adds ~30K of modules, which isn't too bad. What do others users need to do to make use of this?

gstreeter, 2020-09-07:
A user would need an IPSec IKE client or server depending on their use-case. I use the Pi as home-based VPN server so I can connect to my desktop and NAS while away. I use the Strongswan IKE server compiled from source on the Pi. The client is available on the Google and Apple apps stores. The Windows 10 native VPN client also works with it.

https://www.strongswan.org/

pelwell, 2020-09-07:
Thanks - see 9a1dd17.

gstreeter, 2020-09-08:
Thanks - have pulled the update and confirmed modules now included in the build.

closed 2020-09-08 by pelwell