raspberrypi/linux #3839: zram kernel oops on  PI4-8GB board in 32bit

paul-1, 2020-09-06:
**Describe the bug**
Enabling zram swap space causes kernel oops.  This only happens on the Pi4-8GB board.  The 4GB board (And other Pi boards) works fine.

**To reproduce**
```
modprobe zram
grep MemFree /proc/meminfo | awk '{printf("%dK\n", $2/4)}' > /sys/block/zram0/disksize  (Although any size will cause problem)
mkswap /dev/zram0
```

**System**
* Which model of Raspberry Pi? `Pi4-8GB (Raspberry Pi 4 Model B Rev 1.4)`
* Which OS and version (`cat /etc/rpi-issue`)?  
```
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Sep  3 2020 18:04:32
Copyright (c) 2012 Broadcom
version 68206b4397629e1a0892db7a9156aed86525172f (clean) (release) (start)
```

* Which kernel version (`uname -a`)?`Linux raspberrypi 5.4.61-v7l+ #1339 SMP Tue Sep 1 18:51:27 BST 2020 armv7l GNU/Linux`

**Logs**
```
[  145.988932] zram: Added device: zram0
[  149.980753] zram0: detected capacity change from 0 to 2029449216
[  155.416345] 8<--- cut here ---
[  155.418648] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  155.421051] pgd = 51d67c8d
[  155.423360] [00000000] *pgd=133a5003, *pmd=1fff40003
[  155.425727] Internal error: Oops: 207 [#1] SMP ARM
[  155.428075] Modules linked in: zram zsmalloc cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc joydev hid_asus brcmfmac brcmutil v3d sha256_generic libsha256 vc4 gpu_sched cec bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) drm_kms_helper videobuf2_dma_contig videobuf2_vmalloc raspberrypi_hwmon videobuf2_memops cfg80211 videobuf2_v4l2 rfkill videobuf2_common drm snd_soc_core drm_panel_orientation_quirks videodev snd_bcm2835(C) snd_compress snd_pcm_dmaengine mc snd_pcm vc_sm_cma(C) snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6 nf_defrag_ipv6
[  155.436263] CPU: 2 PID: 628 Comm: mkswap Tainted: G         C        5.4.61-v7l+ #1339
[  155.439032] Hardware name: BCM2711
[  155.441802] PC is at zs_map_object+0xb4/0x1e4 [zsmalloc]
[  155.444565] LR is at 0x483e7b98
[  155.447357] pc : [<bf3aa104>]    lr : [<483e7b98>]    psr: 60000013
[  155.450156] sp : d3dadaa8  ip : 00000000  fp : d3dadaf4
[  155.452942] r10: d3ce4000  r9 : 00000000  r8 : d8c00000
[  155.455719] r7 : c1204fc8  r6 : d3d63800  r5 : 00000000  r4 : 02f0f490
[  155.458490] r3 : dbb0f490  r2 : 00000002  r1 : c13369c0  r0 : fafc3001
[  155.461244] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  155.464011] Control: 30c5383d  Table: 14368100  DAC: 55555555
[  155.466787] Process mkswap (pid: 628, stack limit = 0xc863875d)
[  155.469577] Stack: (0xd3dadaa8 to 0xd3dae000)
[  155.472356] daa0:                   d40273c0 d30d8000 00000000 deb0f490 d3dadadc d3dadac8
[  155.475195] dac0: c0658110 483e7b98 d3dadb18 00000000 c1204fc8 d30d8000 d3d899c0 d3db4000
[  155.478051] dae0: bf3c6400 d3ce4000 d3dadb4c d3dadaf8 bf3c3998 bf3aa05c c02a7920 c08e1378
[  155.480964] db00: 00000001 d33b0000 ffffc786 d8fb6ac0 00000000 c1204fc8 0000004a 483e7b98
[  155.483873] db20: 00000008 d3c73300 d3db4000 00001000 00000000 00000000 00001000 00001000
[  155.486832] db40: d3dadbac d3dadb50 bf3c4184 bf3c3660 00000001 d3c73300 c020bcb8 bf3c6198
[  155.489799] db60: 00000000 00000000 00000000 c1204fc8 d8fb6ac0 00001000 00000000 483e7b98
[  155.492752] db80: c020bc5c c1204fc8 d3c73300 d3dadbb4 d33b0000 00000000 ffffffff d8fb6ac0
[  155.495731] dba0: d3dadbec d3dadbb0 c06778a4 bf3c3fdc c038d5fc 00000000 00000000 00000000
[  155.498731] dbc0: 00000000 483e7b98 c1204fc8 d3c73300 00000008 00000800 d3dade60 00000000
[  155.501794] dbe0: d3dadc5c d3dadbf0 c0677adc c06777b0 c135d9ec c066eccc 00000000 c1204fc8
[  155.504891] dc00: d3c73300 00001000 d8fb6ac0 00000000 00000000 d8fb6ac0 d3dadc5c d3dadc28
[  155.507969] dc20: c066eccc c066eb20 d3dadc33 483e7b98 d3dadc5c d352fe40 d3c73300 00000801
[  155.511074] dc40: 00000800 d3dade60 00000000 d8fb6ac0 d3dadc84 d3dadc60 c0450978 c0677a94
[  155.514209] dc60: 00000001 d352fe40 d352fe40 00000800 d4b2c508 d3dade60 d3dadcec d3dadc88
[  155.517340] dc80: c0450c60 c04507f0 d3dade60 d8fb6ac0 d3dadcd4 c0455224 00001000 c1204fc8
[  155.520472] dca0: c044efa4 d3dade60 00000800 d4b2c508 00078f6e 00000000 c03929d4 483e7b98
[  155.523621] dcc0: 20000013 d4b2c508 00000000 d8fb6ac0 d3dade60 c0455224 d3dade60 00000000
[  155.526806] dce0: d3dadd24 d3dadcf0 c0451160 c04509b4 c044efa4 d3dadd00 ffffffff d4b2c618
[  155.529998] dd00: 00000000 d3dadd90 d4b2c618 00000001 d3dade60 00000000 d3dadd34 d3dadd28
[  155.533159] dd20: c0454514 c045107c d3dadd4c d3dadd38 c0391684 c04544fc d8fb6ac0 00000000
[... 50 lines cut ...]
[  155.611569]  r4:d4b2c618
[  155.613033] [<c0392f28>] (generic_writepages) from [<c04544c4>] (blkdev_writepages+0x18/0x1c)
[  155.614511]  r6:d4b2c508 r5:d4b2c618 r4:d3dade60
[  155.616006] [<c04544ac>] (blkdev_writepages) from [<c0394fc0>] (do_writepages+0x30/0x8c)
[  155.617494] [<c0394f90>] (do_writepages) from [<c038ae50>] (__filemap_fdatawrite_range+0xf0/0x12c)
[  155.618990]  r5:c1204fc8 r4:d4b2c618
[  155.620471] [<c038ad60>] (__filemap_fdatawrite_range) from [<c038c2e4>] (file_write_and_wait_range+0x60/0xb4)
[  155.621992]  r10:00000076 r9:00000000 r8:00000000 r7:00000000 r6:d3acb600 r5:d4b2c618
[  155.623493]  r4:ffffffff
[  155.624989] [<c038c284>] (file_write_and_wait_range) from [<c0453ed0>] (blkdev_fsync+0x2c/0x54)
[  155.626515]  r9:d3dac000 r8:c02011c4 r6:d4b2c508 r5:7fffffff r4:ffffffff
[  155.628040] [<c0453ea4>] (blkdev_fsync) from [<c0448374>] (vfs_fsync_range+0x68/0x98)
[  155.629571]  r6:00000000 r5:00000000 r4:d3acb600
[  155.631095] [<c044830c>] (vfs_fsync_range) from [<c044842c>] (do_fsync+0x4c/0x74)
[  155.632619]  r7:00000076 r6:00000000 r5:d3acb600 r4:d3acb600
[  155.634138] [<c04483e0>] (do_fsync) from [<c0448738>] (sys_fsync+0x1c/0x20)
[  155.635674]  r6:00000000 r5:00000003 r4:00000000
[  155.637206] [<c044871c>] (sys_fsync) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[  155.638739] Exception stack(0xd3dadfa8 to 0xd3dadff0)
[  155.640242] dfa0:                   00000000 00000003 00000003 b6ecf808 0000006c 00000000
[  155.641759] dfc0: 00000000 00000003 00000000 00000076 00000000 00000000 00000000 00033c98
[  155.643296] dfe0: 00033dd0 bec37178 00012cf0 b6e558a4
[  155.644808] Code: e0844084 e1a04204 e0883004 e5939018 (e5991000)
[  155.646338] ---[ end trace 41670e3195a789be ]---
```

thwint, 2020-09-20:
I have a very similar issue (the same?) also with a PI4-8GB board in 32bit. A slightly different stack trace though

yutayu, 2020-11-06:
kern  :err   : [ 2218.163204] zram: Decompression failed! err=-22, page=33126
kern  :err   : [ 2222.218258] zram: Decompression failed! err=-22, page=113756
kern  :err   : [ 2495.277005] zram: Decompression failed! err=-22, page=140201

 uname -a
Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux

I think similer. It happens after newest kernel update.

pelwell, 2020-11-06:
Does disabling the upper 4GB of RAM using `total_mem=4096` avoid the problem? (I'm not suggesting this as a fix!)

paul-1, 2020-11-06:
Yes, loads fine that way

pelwell, 2020-11-06:
And am I right in thinking, given the issue title, that the 64-bit kernel has no such problems?

Should we be unable to find a solution to this, would saying "use the 64-bit kernel" be an option for you?

paul-1, 2020-11-06:
Correct 64bit kernels load zram fine.   I opened the issue after ensuring RaspiOS had the same problem, but my work is purely for the custom OS piCorePlayer. Which loads the whole base OS on a Ramdrive.  We had been running 64bit kernels with 32bit user space on piCorePlayer.  But I'll be launching pCP7 soon, and we will now have a 32bit userland and a 64 bit userland.   Right or wrong, we use the kernel mode to know what environment we are running.  So now using the 64bit kernel with 32 bit userland is not a good option for us.

Our images still work on every rpi board, so those older boards with small memory sizes, we do see a benefit from using the zram compression.  So we ship images with Zram on.  If you cannot find a solution, I would likely build a trap in the zram startup script to not load if memory size is >4gb.   Let me know if I need to go that route.

Alexey-Tsarev, 2020-11-13:
It happened to me too. RPI4, 4Gb

```
Nov 12 21:31:31 laser-pi kernel: [215358.737362] zram: Decompression failed! err=-22, page=9711
Nov 12 21:31:31 laser-pi kernel: [215358.737435] zram: Decompression failed! err=-22, page=9711
Nov 12 21:31:31 laser-pi kernel: [215358.737448] Read-error on swap-device (254:1:77688)

cat /etc/os-release | head -n 1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

uname -a
Linux laser-pi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
```

yutayu, 2020-11-26:
[32375.779020] swapper/0: page allocation failure: order:0, mode:0xa20(GFP_ATOMIC), nodemask=(null),cpuset=/,mems_allowed=0
[32375.779048] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.72-v7+ #1356
[32375.779051] Hardware name: BCM2835
[32375.779054] Backtrace: 
[32375.779071] [<8010e1c0>] (dump_backtrace) from [<8010e538>] (show_stack+0x20/0x24)
[32375.779079]  r7:ffffffff r6:00000000 r5:60000113 r4:80e996f0
[32375.779088] [<8010e518>] (show_stack) from [<8090e830>] (dump_stack+0xd4/0x118)
[32375.779098] [<8090e75c>] (dump_stack) from [<802d97a4>] (warn_alloc+0xe4/0x178)
[32375.779106]  r10:00000000 r9:80e00000 r8:00000000 r7:ffffe000 r6:80bbf69c r5:00000000
[32375.779110]  r4:00000000 r3:00000000
[32375.779117] [<802d96c0>] (warn_alloc) from [<802da990>] (__alloc_pages_nodemask+0x1158/0x121c)
[32375.779121]  r3:00000000 r2:80bbf69c
[32375.779126]  r7:00000a20 r6:00000a20 r5:00000201 r4:00000200
[32375.779133] [<802d9838>] (__alloc_pages_nodemask) from [<802dace0>] (page_frag_alloc+0x178/0x194)
[32375.779141]  r10:80e00000 r9:00000000 r8:00000000 r7:00000a00 r6:00000a20 r5:00000b40
[32375.779144]  r4:b6b2bdc8
[32375.779153] [<802dab68>] (page_frag_alloc) from [<807bc2d8>] (__netdev_alloc_skb+0x150/0x180)
[32375.779158]  r7:00000a00 r6:00000b40 r5:b5f76000 r4:b6b2bdc8
[32375.779166] [<807bc188>] (__netdev_alloc_skb) from [<806d1c0c>] (rx_submit+0x44/0x234)
[32375.779172]  r8:00000000 r7:00000a00 r6:b28d1000 r5:00000000 r4:b5f76540
[32375.779178] [<806d1bc8>] (rx_submit) from [<806d286c>] (rx_complete+0x2a0/0x2b4)
[32375.779185]  r9:00000000 r8:00000000 r7:a6ee9600 r6:b28d1000 r5:00000000 r4:b5f76540
[32375.779193] [<806d25cc>] (rx_complete) from [<806dcdd4>] (__usb_hcd_giveback_urb+0x94/0x110)
[32375.779198]  r7:00000000 r6:00000000 r5:b676b000 r4:b28d1000
[32375.779206] [<806dcd40>] (__usb_hcd_giveback_urb) from [<806dcf40>] (usb_hcd_giveback_urb+0xf0/0xf4)
[32375.779211]  r7:b6b212cc r6:b28d1000 r5:b28d1000 r4:b3c918c0
[32375.779222] [<806dce50>] (usb_hcd_giveback_urb) from [<80709010>] (completion_tasklet_func+0x90/0xdc)
[32375.779227]  r7:b6b212cc r6:b28d1000 r5:b5e2c000 r4:b3c918c0
[32375.779236] [<80708f80>] (completion_tasklet_func) from [<807182fc>] (tasklet_callback+0x20/0x24)
[32375.779241]  r6:b66cdd44 r5:00000000 r4:b66cdd40
[32375.779250] [<807182dc>] (tasklet_callback) from [<80126d74>] (tasklet_action_common.constprop.0+0x64/0xf0)
[32375.779259] [<80126d10>] (tasklet_action_common.constprop.0) from [<80126e58>] (tasklet_hi_action+0x28/0x30)
[32375.779266]  r9:00000001 r8:b645f000 r7:00000100 r6:00000000 r5:00000001 r4:80e03080
[32375.779275] [<80126e30>] (tasklet_hi_action) from [<80102480>] (__do_softirq+0x190/0x414)
[32375.779282] [<801022f0>] (__do_softirq) from [<80126c04>] (irq_exit+0xdc/0x100)
[32375.779288]  r10:10c5387d r9:80e00000 r8:b645f000 r7:00000000 r6:00000001 r5:00000000
[32375.779291]  r4:ffffe000
[32375.779301] [<80126b28>] (irq_exit) from [<80181aa4>] (__handle_domain_irq+0x70/0xc0)
[32375.779305]  r5:00000000 r4:80da4264
[32375.779313] [<80181a34>] (__handle_domain_irq) from [<80102224>] (bcm2836_arm_irqchip_handle_irq+0x64/0xa4)
[32375.779319]  r9:80e00000 r8:80e9ebcb r7:80e01f14 r6:ffffffff r5:60000013 r4:80109c10
[32375.779327] [<801021c0>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[32375.779331] Exception stack(0x80e01ee0 to 0x80e01f28)
[32375.779338] 1ee0: 80109c0c 00000000 60000093 60000093 80e00000 00000000 80e04fa8 80e04ff0
[32375.779344] 1f00: 80e9ebcb 80ba5a98 10c5387d 80e01f3c ffffe000 80e01f30 80e00000 80109c10
[32375.779348] 1f20: 60000013 ffffffff
[32375.779360] [<80109bdc>] (arch_cpu_idle) from [<8092d6a4>] (default_idle_call+0x34/0x48)
[32375.779368] [<8092d670>] (default_idle_call) from [<80155e2c>] (do_idle+0xec/0x16c)
[32375.779375] [<80155d40>] (do_idle) from [<80156194>] (cpu_startup_entry+0x28/0x2c)
[32375.779381]  r9:b73ff840 r8:80d69a40 r7:00000000 r6:80eb1000 r5:80e04f80 r4:000000ce
[32375.779388] [<8015616c>] (cpu_startup_entry) from [<80926a14>] (rest_init+0xb4/0xbc)
[32375.779399] [<80926960>] (rest_init) from [<80d00b78>] (arch_call_rest_init+0x18/0x1c)
[32375.779403]  r5:80e04f80 r4:80eb1054
[32375.779410] [<80d00b60>] (arch_call_rest_init) from [<80d010c4>] (start_kernel+0x4d0/0x510)
[32375.779417] [<80d00bf4>] (start_kernel) from [<00000000>] (0x0)
[32375.779420] Mem-Info:
[32375.779435] active_anon:54587 inactive_anon:64675 isolated_anon:32
                active_file:26489 inactive_file:12757 isolated_file:0
                unevictable:4 dirty:11 writeback:0 unstable:0
                slab_reclaimable:5241 slab_unreclaimable:8362
                mapped:45978 shmem:35489 pagetables:2859 bounce:0
                free:4359 free_pcp:518 free_cma:0
[32375.779446] Node 0 active_anon:218348kB inactive_anon:258700kB active_file:105956kB inactive_file:51028kB unevictable:16kB isolated(anon):128kB isolated(file):0kB mapped:183912kB dirty:44kB writeback:0kB shmem:141956kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[32375.779460] Normal free:17436kB min:53248kB low:57344kB high:61440kB active_anon:218348kB inactive_anon:258636kB active_file:106240kB inactive_file:50528kB unevictable:16kB writepending:44kB present:970752kB managed:947032kB mlocked:16kB kernel_stack:3200kB pagetables:11436kB bounce:0kB free_pcp:2068kB local_pcp:472kB free_cma:0kB
[32375.779464] lowmem_reserve[]: 0 0
[... cut ...] 372kB
[32375.779513] 74799 total pagecache pages
[32375.779520] 39 pages in swap cache
[32375.779526] Swap cache stats: add 841555, delete 841517, find 18715/460207
[32375.779531] Free swap  = 524288kB
[32375.779536] Total swap = 947028kB
[32375.779540] 242688 pages RAM
[32375.779545] 0 pages HighMem/MovableOnly
[32375.779549] 5930 pages reserved
[32375.779554] 16384 pages cma reserved

pi 3B 1GB 32bit

uname -a
Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux

Alexey-Tsarev, 2020-11-26:
@yutayu, your log message contains nothing about zram.

yutayu, 2020-11-26:
@Alexey-Tsarev 
This happens often when I use zram as swap.

lanmarc77, 2020-12-01:
Same here on a Pi Zero W. Sporadic:
[ 1388.690372] zram: Decompression failed! err=-22, page=73880

on Linux version 5.4.72+ (dom@buildbot) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #1356 Thu Oct 22 13:56:00 BST 2020
When I create a backup on my system I create an additional swapfile on the sd card. I have the feeling the zram decompression error only happens then. I will investigate further. Did never happen before the upgrade to that kernel version. Was on 5.4.51 before that upgrade.

yutayu, 2020-12-04:
https://askubuntu.com/questions/1295809/zram-decompression-failed-with-new-kernels
https://forum.armbian.com/topic/16056-problems-with-zram-on-h235-boards-zram-decompression-failed-err-22/?tab=comments#comment-113430
https://github.com/armbian/build/pull/2377

lanmarc77, 2020-12-08:
It seems to me as with the recent kernel update the bug is gone on my PI Zero W. I am now on:
Linux version 5.4.79+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 Mon Nov 23 13:18:15 GMT 2020

May be someone can confirm that between 72+ and 79+ something was done on corresponding code parts?

yutayu, 2020-12-08:
[39580.361213] zram: Decompression failed! err=-22, page=25684
uname -a
Linux raspberrypi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux

with 3B , Still it happens.
@lanmarc77

yutayu, 2020-12-08:
@lanmarc77 
as you say , I got these error after recent kernel update.

yutayu, 2020-12-09:
root@raspberrypi:/etc/modprobe.d# cat zsmalloc.conf
options	zsmalloc	pgtable_mapping=n

modprobe -D zsmalloc
insmod /lib/modules/5.4.79-v7+/kernel/mm/zsmalloc.ko pgtable_mapping=n 

I am trying this mobule option.

MartinConsultingServicesInc, 2021-01-01:
Just another "me too" ... I also find my PI4-8GB board crashes when I try to use zram.

yutayu, 2021-02-05:
uname -a
Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux

It seeems fixed.

lanmarc77, 2021-02-05:
@yutayu Thanks for the update, I will reenable it on my pi zero and report back in a week or so

yutayu, 2021-02-07:
https://github.com/raspberrypi/linux/commit/e91d8d78237de8d7120c320b3645b7100848f24d

lanmarc77, 2021-02-12:
No problems detected anymore. Works for me on a Pi Zero W

popcornmix, 2021-03-23:
@paul-1 okay to close?

paul-1, 2021-03-23:
Sure thing.

closed 2021-03-23 by paul-1