raspberrypi/linux #3842: 4k60p display issue (DRM VC4 driver crash / backtrace)

rhildinger, 2020-09-08:
I'm running into an issue trying to get 4K60p output from my RPi4 to my Acer Nitro XV273K monitor. I thought this might have been the typical EDID issue that many people have run into trying to get 4K60p to work, but this appears to be more of a corner case that involves a bug in the DRM VC4 driver causing it to get stuck in a loop and eventually crash an/or restart.

The specific symptom in this case is that when I set "hdmi_enable_4kp60=1" in config.txt and reboot, the display will show the initial boot screen with the 4 raspberry pi logos and boot logs, but as soon as the VC4 driver takes over, the display goes blank and stays blank. I did eventually discover that If I wait for 8 or 9 minutes, the display does in fact come back on and show the RPi OS desktop at 4k60p. This behavior is consistent and repeatable.

**Hardware list:**
- RPi 4 8GB
- Official PSU
- Official microHDMI to HDMI
- Acer Nitro XV273K Monitor (supports HDMI 2.0 input with support for 4K60p full RGB)

**OS:**
- Fresh install of August 2020 Raspberry Pi OS (32-bit)

**Steps:**
- Install clean image of August 2020 RPi OS to SD card
- boot SD card, desktop comes up normally at 4K30p
- Use raspi-config to enable 4K60p, reboot
- Initial boot screen comes up in 4k60p, then display goes blank
- After 8-9 minutes, desktop finally appears in 4K60p

**Analysis:**
I spent most of the day yesterday trying to debug this issue. My initial thought was that this was an EDID parsing / matching issue, but after some investigation I determined that wasn't the case and that the VC4 DRM driver is selecting the appropriate modeline. After enabling the DRM debug logs in the kernel via the "drm.debug=0x14" kernel command line flag, I could see that the DRM driver is getting stuck in a loop with the following error logs popping out 30 seconds or so:

```
[  247.534702] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:52:crtc-0] flip_done timed out
[  257.774879] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:54:HDMI-A-1] flip_done timed out
[  257.774970] [drm:vc4_crtc_atomic_flush [vc4]] [CRTC:52] crtc_atomic_flush.
[  268.015082] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:52:crtc-0] flip_done timed out
```

This goes on for several minutes until the VC4 DRM driver apparently crashes or restarts just past the 7 minute mark. The following backtrace is dumped at this point:

```
[  432.577342] [drm:vc4_crtc_atomic_check [vc4]] [CRTC:52] crtc_atomic_check.
[  432.577529] [drm:drm_atomic_commit [drm]] committing f6ee61ec
[  442.738703] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:52:crtc-0] flip_done timed out
[  452.978880] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:54:HDMI-A-1] flip_done timed out
[  463.219017] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:31:plane-0] flip_done timed out
[  463.219109] [drm:drm_atomic_helper_commit_modeset_disables [drm_kms_helper]] disabling [ENCODER:53:TMDS-53]
[  463.219231] [drm:vc4_fkms_encoder_disable [vc4]] Encoder_disable
[  463.219321] [drm:drm_atomic_helper_commit_modeset_disables [drm_kms_helper]] disabling [CRTC:52:crtc-0]
[  463.219401] [drm:vc4_crtc_disable [vc4]] [CRTC:52] vblanks off.
[  463.219483] [drm:vc4_fkms_disable_vblank [vc4]] [CRTC:52] disable_vblank.
[  463.219560] [drm:vc4_plane_atomic_disable [vc4]] [PLANE:31:plane-0] plane disable 3840x2160@44 +0,0
[  463.219636] [drm:vc4_plane_set_blank [vc4]] [PLANE:31:plane-0] primary plane blank
[  463.219765] [drm:drm_atomic_helper_commit_modeset_disables [drm_kms_helper]] modeset on [CRTC:52:crtc-0]
[  463.219875] [drm:vc4_crtc_mode_set_nofb [vc4]] Setting mode for display num 2 mode name , clk 594000, h(disp 3840, start 4016, end 4104, total 4400, skew 0) v(disp 2160, start 2168, end 2178, total 2250, scan 0), vrefresh 60, par 0, flags 0x0005
[  464.258959] ------------[ cut here ]------------
[  464.258990] WARNING: CPU: 0 PID: 511 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0x108/0x128
[  464.259002] Firmware transaction timeout
[  464.259006] Modules linked in: cmac rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc evdev hid_apple brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill vc4 cec v3d gpu_sched raspberrypi_hwmon drm_kms_helper drm bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig drm_panel_orientation_quirks videobuf2_vmalloc videobuf2_memops vc_sm_cma(C) videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_soc_core videodev snd_compress mc snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[  464.259253] CPU: 0 PID: 511 Comm: Xorg Tainted: G         C        5.4.51-v7l+ #1333
[  464.259261] Hardware name: BCM2711
[  464.259268] Backtrace: 
[  464.259289] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[  464.259302]  r6:d30d0000 r5:00000000 r4:c129c8f8 r3:0e24d4f2
[  464.259321] [<c020d748>] (show_stack) from [<c0a39a44>] (dump_stack+0xe0/0x124)
[  464.259338] [<c0a39964>] (dump_stack) from [<c0221c70>] (__warn+0xec/0x104)
[  464.259351]  r8:0000003f r7:00000009 r6:c0e2a724 r5:00000000 r4:d30d1a64 r3:0e24d4f2
[  464.259367] [<c0221b84>] (__warn) from [<c0221d0c>] (warn_slowpath_fmt+0x84/0xc0)
[  464.259379]  r9:c0e2a724 r8:0000003f r7:c08a2dec r6:00000009 r5:c0e2a744 r4:c1204f88
[  464.259394] [<c0221c8c>] (warn_slowpath_fmt) from [<c08a2dec>] (rpi_firmware_transaction+0x108/0x128)
[  464.259406]  r9:ef96b340 r8:00000000 r7:00000000 r6:ffffff92 r5:ef96b340 r4:c1204f88
[  464.259421] [<c08a2ce4>] (rpi_firmware_transaction) from [<c08a2ec8>] (rpi_firmware_property_list+0xbc/0x174)
[  464.259431]  r7:c1204f88 r6:dec04000 r5:00001000 r4:4000000e
[  464.259513] [<c08a2e0c>] (rpi_firmware_property_list) from [<bf25e008>] (vc4_crtc_mode_set_nofb+0x2d4/0x35c [vc4])
[  464.259526]  r10:d4270a00 r9:d3dedc40 r8:ef323614 r7:c1204f88 r6:d3ded440 r5:00000000
[  464.259534]  r4:ef323600
[  464.259685] [<bf25dd34>] (vc4_crtc_mode_set_nofb [vc4]) from [<bf3435c8>] (drm_atomic_helper_commit_modeset_disables+0x294/0x4b8 [drm_kms_helper])
[  464.259697]  r10:ef322600 r9:d3ded440 r8:bf27ad80 r7:d3ded800 r6:d3ded440 r5:00000001
[... 71 lines cut ...]
[  464.507794] [<bf2c9d40>] (drm_ioctl [drm]) from [<c0410860>] (do_vfs_ioctl+0xbc/0x804)
[  464.507807]  r10:00000036 r9:d30d0000 r8:0000000c r7:d42ada80 r6:d3ea7af0 r5:bef43568
[  464.507815]  r4:c1204f88
[  464.507830] [<c04107a4>] (do_vfs_ioctl) from [<c0410fec>] (ksys_ioctl+0x44/0x6c)
[  464.507841]  r10:00000036 r9:d30d0000 r8:0000000c r7:c06864a2 r6:d42ada80 r5:bef43568
[  464.507849]  r4:d42ada81
[  464.507864] [<c0410fa8>] (ksys_ioctl) from [<c041102c>] (sys_ioctl+0x18/0x1c)
[  464.507876]  r8:c02011c4 r7:00000036 r6:c06864a2 r5:bef43568 r4:b6fbd968 r3:00000001
[  464.507891] [<c0411014>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[  464.507900] Exception stack(0xd30d1fa8 to 0xd30d1ff0)
[  464.507912] 1fa0:                   b6fbd968 bef43568 0000000c c06864a2 bef43568 00000001
[  464.507924] 1fc0: b6fbd968 bef43568 c06864a2 00000036 01026cb8 0000003c 00000034 00000000
[  464.507934] 1fe0: b6d6a08c bef4354c b6d5088c b69dc51c
[  464.507944] ---[ end trace 2904cc9e799aa98a ]---
[  464.508026] [drm:vc4_crtc_atomic_flush [vc4]] [CRTC:52] crtc_atomic_flush.
[  464.508117] [drm:drm_atomic_helper_commit_modeset_enables [drm_kms_helper]] enabling [CRTC:52:crtc-0]
[  464.508202] [drm:vc4_crtc_enable [vc4]] [CRTC:52] vblanks on.
```

**Immediately after this backtrace is dumped, the VC4 DRM driver seems to correct itself and successfully brings up the desktop at 4k60p.** 

I've reached the limit of my ability to debug this issue, so I need some guidance from the RPi experts...

Thanks!
-Robert Hildinger

alanbork, 2020-12-13:
did this ever get fixed?

Bibinoo, 2021-01-10:
Hi,
I have the same issue unfortunatelly. I has never happened with the September 2020 firmwate but after installing the last one 3rd Dec 2020 I have this crashes as well with :

[165973.223398] WARNING: CPU: 0 PID: 1075 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xec/0x128
[165973.223404] Firmware transaction timeout
[165973.223410] Modules linked in: rfcomm tun fuse veth xt_conntrack nf_conntrack_netlink br_netfilter bridge binfmt_misc overlay bnep hci_uart btbcm bluetooth ecdh_generic ecc ipt_REJECT nf_reject_ipv4 xt_owner nft_counter 8021q garp stp llc nft_chain_nat xt_nat xt_tcpudp xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype nft_compat nf_tables nfnetlink brcmfmac brcmutil v3d sha256_generic gpu_sched cfg80211 bcm2835_codec(C) i2c_bcm2835 rfkill sg v4l2_mem2mem raspberrypi_hwmon bcm2835_v4l2(C) vc4 bcm2835_isp(C) bcm2835_mmal_vchiq(C) cec videobuf2_dma_contig vc_sm_cma(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_kms_helper videodev snd_bcm2835(C) mc drm drm_panel_orientation_quirks snd_soc_core rpivid_mem snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight gpio_fan uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[165973.223897] CPU: 0 PID: 1075 Comm: InputThread Tainted: G         C        5.10.4-v7l+ #1387
[165973.223901] Hardware name: BCM2711
[165973.223906] Backtrace:
[165973.223918] [<c0b584dc>] (dump_backtrace) from [<c0b58870>] (show_stack+0x20/0x24)
[165973.223925]  r7:ffffffff r6:00000000 r5:68000013 r4:c12e69bc
[165973.223932] [<c0b58850>] (show_stack) from [<c0b5cc00>] (dump_stack+0xcc/0xf8)
[165973.223941] [<c0b5cb34>] (dump_stack) from [<c0220af4>] (__warn+0xfc/0x114)
[165973.223947]  r10:dec08008 r9:00000009 r8:c0997ce8 r7:00000040 r6:00000009 r5:c0997ce8
[165973.223952]  r4:c0e98e20 r3:c1205094
[165973.223958] [<c02209f8>] (__warn) from [<c0b59014>] (warn_slowpath_fmt+0xa4/0xd8)
[165973.223964]  r7:00000040 r6:c0e98e20 r5:c1205048 r4:c0e98e40
[165973.223971] [<c0b58f74>] (warn_slowpath_fmt) from [<c0997ce8>] (rpi_firmware_transaction+0xec/0x128)
[165973.223977]  r9:c1a6f540 r8:00000048 r7:00000000 r6:ffffff92 r5:c1a6f540 r4:c1205048
[165973.223984] [<c0997bfc>] (rpi_firmware_transaction) from [<c0997de0>] (rpi_firmware_property_list+0xbc/0x170)
[165973.223989]  r7:c1205048 r6:dec08000 r5:00001000 r4:dec08054
[165973.224032] [<c0997d24>] (rpi_firmware_property_list) from [<bf3900e8>] (vc4_plane_set_blank+0x110/0x168 [vc4])
[165973.224038]  r10:c94cea00 r9:00000000 r8:00000000 r7:ca53e000 r6:caac8a20 r5:c1205048
[165973.224042]  r4:caac8840
[165973.224101] [<bf38ffd8>] (vc4_plane_set_blank [vc4]) from [<bf3902c8>] (vc4_plane_atomic_update+0x34/0x38 [vc4])
[165973.224106]  r8:c94ce000 r7:bf3afb08 r6:caac8840 r5:00000007 r4:c5f551c0
[165973.224175] [<bf390294>] (vc4_plane_atomic_update [vc4]) from [<bf299e88>] (drm_atomic_helper_commit_planes+0xe8/0x28c [drm_kms_helper])
[165973.224240] [<bf299da0>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf39298c>] (vc4_atomic_complete_commit+0x1b0/0x680 [vc4])
[165973.224246]  r10:cc365b40 r9:00000000 r8:ca53e500 r7:00000000 r6:ca53e000 r5:c5f551c0
[165973.224250]  r4:00000001
[165973.224305] [<bf3927dc>] (vc4_atomic_complete_commit [vc4]) from [<bf392fd8>] (vc4_atomic_commit+0x17c/0x1cc [vc4])
[165973.224311]  r10:cc365b40 r9:00000000 r8:ca53e500 r7:00000000 r6:00000000 r5:ca53e000
[165973.224315]  r4:c5f551c0 r3:00000001
[165973.224423] [<bf392e5c>] (vc4_atomic_commit [vc4]) from [<bf1978e0>] (drm_atomic_commit+0x54/0x60 [drm])
[165973.224429]  r9:000003ac r8:ca834c40 r7:00000000 r6:ca53e000 r5:c5f551c0 r4:00000000
[165973.224540] [<bf19788c>] (drm_atomic_commit [drm]) from [<bf29aa24>] (drm_atomic_helper_update_plane+0xfc/0x134 [drm_kms_helper])
[165973.224545]  r7:00000000 r6:caac8840 r5:c94ce000 r4:c5f551c0
[165973.224655] [<bf29a928>] (drm_atomic_helper_update_plane [drm_kms_helper]) from [<bf1a2070>] (__setplane_atomic+0x104/0x144 [drm])
[165973.224661]  r10:00400000 r9:00400000 r8:00000000 r7:000003ac r6:ca834c40 r5:cc365b40
[165973.224665]  r4:bf29a928 r3:000003ac
[165973.224809] [<bf1a1f6c>] (__setplane_atomic [drm]) from [<bf1a23c0>] (drm_mode_cursor_universal+0x174/0x294 [drm])
[165973.224815]  r10:cc951d64 r9:ca53e000 r8:caac8840 r7:c1205048 r6:ca834c40 r5:cc951db8
[165973.224819]  r4:cc365b40
[165973.224960] [<bf1a224c>] (drm_mode_cursor_universal [drm]) from [<bf1a25dc>] (drm_mode_cursor_common+0xfc/0x250 [drm])
[165973.224965]  r10:cc951e64 r9:00000000 r8:ca834c40 r7:c297bf00 r6:ca834c54 r5:ca834c74
[165973.224969]  r4:cc951db8
[165973.225111] [<bf1a24e0>] (drm_mode_cursor_common [drm]) from [<bf1a2af0>] (drm_mode_cursor_ioctl+0x78/0x94 [drm])
[165973.225116]  r10:cc951e64 r9:bf1a2a78 r8:00000002 r7:c297bf00 r6:ca53e000 r5:00000000
[165973.225120]  r4:c1205048
[165973.225262] [<bf1a2a78>] (drm_mode_cursor_ioctl [drm]) from [<bf18546c>] (drm_ioctl_kernel+0xb8/0xfc [drm])
[165973.225268]  r7:00000000 r6:c297bf00 r5:ca53e000 r4:c1205048
[165973.225410] [<bf1853b4>] (drm_ioctl_kernel [drm]) from [<bf185694>] (drm_ioctl+0x1e4/0x3a8 [drm])
[165973.225416]  r10:0000001c r9:cc951e64 r8:c297bf00 r7:000000a3 r6:bf1a2a78 r5:c1205048
[165973.225420]  r4:bf1baa6c
[165973.225499] [<bf1854b0>] (drm_ioctl [drm]) from [<c044fa7c>] (sys_ioctl+0x1d4/0x8ec)
[165973.225505]  r10:0000000c r9:cc950000 r8:00000000 r7:cb625900 r6:cb625901 r5:c1205048
[165973.225509]  r4:c01c64a3
[... cut ...]
[165973.225521] Exception stack(0xcc951fa8 to 0xcc951ff0)
[165973.225527] 1fa0:                   b6f89968 b0a640c8 0000000c c01c64a3 b0a640c8 00000370
[165973.225533] 1fc0: b6f89968 b0a640c8 c01c64a3 00000036 00000001 00000372 001fd000 00201894
[165973.225538] 1fe0: b6d3508c b0a640ac b6d1b88c b69a751c
[165973.225543]  r10:00000036 r9:cc950000 r8:c0200204 r7:00000036 r6:c01c64a3 r5:b0a640c8
[165973.225547]  r4:b6f89968
[165973.225553] ---[ end trace 01a71a45c4bc2439 ]---

I have no c;lue how to fix this

aromanro, 2021-01-21:
Hi,

I have a similar issue, reproduced on both compute module 3 and 3+. 

It's for a custom DSI panel with 4 lanes, kms vc4:

```
Jan 19 17:02:55 raspberrypi kernel: [    8.491571] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
Jan 19 17:02:55 raspberrypi kernel: [    8.501716] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.502085] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.771856] HG_LTP08_lcd_driver: loading out-of-tree module taints kernel.
Jan 19 17:02:55 raspberrypi kernel: [    8.772984] Probing!
Jan 19 17:02:55 raspberrypi kernel: [    8.785852] Backlight pin set to 28
Jan 19 17:02:55 raspberrypi kernel: [    8.791585] Reset pin set to 13
Jan 19 17:02:55 raspberrypi kernel: [    8.796824] DSI Device init for HG_LTP08!
Jan 19 17:02:55 raspberrypi kernel: [    8.806632] Probed!
Jan 19 17:02:55 raspberrypi kernel: [    8.822476] debugfs: Directory '3f902000.hdmi' with parent 'vc4-hdmi' already present!
Jan 19 17:02:55 raspberrypi kernel: [    8.829375] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
Jan 19 17:02:55 raspberrypi kernel: [    8.830486] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.830797] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.833205] vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.833497] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.833734] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.834047] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.834379] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.834802] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.835052] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
Jan 19 17:02:55 raspberrypi kernel: [    8.836167] checking generic (3eaf0000 10a800) vs hw (0 ffffffff)
Jan 19 17:02:55 raspberrypi kernel: [    8.836182] fb0: switching to vc4drmfb from simple
Jan 19 17:02:55 raspberrypi kernel: [    8.836947] Console: switching to colour dummy device 80x30
Jan 19 17:02:55 raspberrypi kernel: [    8.837109] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Jan 19 17:02:55 raspberrypi kernel: [    8.837118] [drm] Driver supports precise vblank timestamp query.
Jan 19 17:02:55 raspberrypi kernel: [    8.839186] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
Jan 19 17:02:55 raspberrypi kernel: [    8.853832] Preparing!
Jan 19 17:02:55 raspberrypi kernel: [    9.796500] Enabling!
Jan 19 17:02:55 raspberrypi kernel: [   20.316460] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:69:crtc-1] flip_done timed out
Jan 19 17:02:55 raspberrypi kernel: [   20.317141] Console: switching to colour frame buffer device 100x80
Jan 19 17:02:55 raspberrypi kernel: [   30.556485] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:69:crtc-1] flip_done timed out
Jan 19 17:02:55 raspberrypi kernel: [   40.796449] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:49:DSI-1] flip_done timed out
Jan 19 17:02:55 raspberrypi kernel: [   51.036446] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:63:plane-1] flip_done timed out
Jan 19 17:02:55 raspberrypi kernel: [   61.276438] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:69:crtc-1] flip_done timed out
Jan 19 17:02:55 raspberrypi kernel: [   61.292286] vc4-drm soc:gpu: fb0: vc4drmfb frame buffer device
```

The panel worked with older kernel versions, for example with 4.19.127-v7+ kernel that comes from a Raspbian Lite image from May 2020, but with the kernel 5.4,83-v7+ that comes with the RaspberryOS image from January 2021, it does not work anymore.

Looks like the panel device driver is probed, prepared and enabled ok, then the problem appears. The screen stays blank and eventually the panel driver is disabled.

aromanro, 2021-01-22:
I have more info on this: after changing the driver code for 5.10.y (there are some changes in there that prevented compiling it), it seems to work, so probably along the way the DSI vc4 implementation was broken and in 5.10.y is fixed.

6by9, 2021-01-22:
Any discussion of DSI displays on CM3 is going a long way off topic for a thread on 4k60 on Pi4.

If you are wishing to use vc4-kms-v3d then I strongly recommend you use the rpi-5.10.y branch.
rpi-5.4.y had an early version of the patches that added support for the Pi4, and they did cause some regressions with Pi0-3. 5.10 is now working pretty well on all 

The backtrace from rpi_firmware_transaction is just that the firmware hasn't replied within a timeout period, typically because something else has managed to crash it. The callstack is generally meaningless.

I haven't checked 4k60 myself recently, but it was working. There were a large number of tweaks to get the HDMI state machine stable, and AFAIK it is in the most recent releases.

alanbork, 2021-01-22:
By release do you mean the official image downloadable from the website , or a testing version via rpi-update? I have a copy downloaded in November that's having issues with 4k60, and I'm wondering if I need a new install from scratch, or should try rpi-update.

alanbork, 2021-04-15:
anybody found a solution to this? I've personally found that 3840x2160-60 works but have yet to get 4096x2160-60 to ever show a picture using the pi4, across several different cables, pi4s, and monitors/tvs. The pi tries to display it, meaning that all the config.txt settings are probably right, but the displays don't like the signal.

6by9, 2021-04-15:
I'd need to check the actual register sizes.
4096 needs a 13bit register to store the width properties, whilst 3840 fits into 12bits. Checking the vc4-kms-v3d driver, it does appear to be a 13bit register (for both width and height), but it's possible the masking is incorrect in the firmware.

4k60 support is nearly resolved for vc4-kms-v3d.
I will have a very quick check of the firmware to see if there is an obvious masking error, but vc4-kms-v3d is going to be the better supported solution longer term. I don't have a display that accepts 4096x2160 @ 60Hz to test with.

Please could you dump the EDID so that I can see what configuration we end up with, even if I can't see the results. Github should take zipped attachments, or base64 encode it.

alanbork, 2021-04-15:
I think that question is meant for me? I dumped both edid's for a separate bug report I posted (hard crash/lockup on recent firmware for 4096x2160): https://github.com/raspberrypi/linux/issues/4271.  Sorry if that's a duplicate; though it's not obviously so: The original poster's backtrace shows he was trying 3840x and failing, so he should definitely try the newest firmware to see if it works for him.

alanbork, 2021-04-15:
sorry it looks like you are asking for the binary not the decoded edid?

6by9, 2021-04-15:
> sorry it looks like you are asking for the binary not the decoded edid?

Yes please. Accessible as (normally) /sys/class/drm/card1-HDMI-A-1/edid or /sys/class/drm/card1-HDMI-A-2/edid

alanbork, 2021-04-15:
not the result of tvservice -d or whatever the flag is?

alanbork, 2021-04-15:
[tvservice dumped edids 4k tvs.zip](https://github.com/raspberrypi/linux/files/6320003/tvservice.dumped.edids.4k.tvs.zip)

6by9, 2021-04-15:
Running those through edid-decode,
sonyedid
```
----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: SNY
    Model: 62211
    Serial Number: 16843009
    Made in: week 1 of 2016
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 95 cm x 54 cm
    Gamma: 2.20
    RGB color display
    First detailed timing is preferred timing
  Color Characteristics:
    Red  : 0.6250, 0.3398
    Green: 0.2802, 0.5947
    Blue : 0.1552, 0.0703
    White: 0.2832, 0.2978
  Established Timings I & II:
    DMT 0x04:   640x480    59.940 Hz   4:3    31.469 kHz  25.175 MHz
    DMT 0x09:   800x600    60.317 Hz   4:3    37.879 kHz  40.000 MHz
    DMT 0x10:  1024x768    60.004 Hz   4:3    48.363 kHz  65.000 MHz
  Standard Timings:
    DMT 0x23:  1280x1024   60.020 Hz   5:4    63.981 kHz 108.000 MHz
    DMT 0x53:  1600x900    60.000 Hz  16:9    60.000 kHz 108.000 MHz (RB)
    DMT 0x15:  1152x864    75.000 Hz   4:3    67.500 kHz 108.000 MHz
    DMT 0x3a:  1680x1050   59.954 Hz  16:10   65.290 kHz 146.250 MHz
  Detailed Timing Descriptors:
    DTD 1:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz (952 mm x 535 mm)
                 Hfront   88 Hsync  44 Hback 148 Hpol P
                 Vfront    4 Vsync   5 Vback  36 Vpol P
    DTD 2:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz (952 mm x 535 mm)
                 Hfront  110 Hsync  40 Hback 220 Hpol P
                 Vfront    5 Vsync   5 Vback  20 Vpol P
    Display Product Name: 'SONY TV  *00'
  Display Range Limits:
    Monitor ranges (GTF): 48-62 Hz V, 14-70 kHz H, max dotclock 300 MHz
  Extension blocks: 1
Checksum: 0x53

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Underscans PC formats by default
  Basic audio support
  Supports YCbCr 4:4:4
  Supports YCbCr 4:2:2
  Native detailed modes: 0
  Video Data Block:
    VIC  93:  3840x2160   24.000 Hz  16:9    54.000 kHz 297.000 MHz
    VIC  94:  3840x2160   25.000 Hz  16:9    56.250 kHz 297.000 MHz
    VIC  95:  3840x2160   30.000 Hz  16:9    67.500 kHz 297.000 MHz
    VIC  98:  4096x2160   24.000 Hz 256:135  54.000 kHz 297.000 MHz
    VIC  31:  1920x1080   50.000 Hz  16:9    56.250 kHz 148.500 MHz
    VIC  16:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz
    VIC  20:  1920x1080i  50.000 Hz  16:9    28.125 kHz  74.250 MHz
    VIC   5:  1920x1080i  60.000 Hz  16:9    33.750 kHz  74.250 MHz
    VIC  19:  1280x720    50.000 Hz  16:9    37.500 kHz  74.250 MHz
    VIC   4:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz
    VIC  32:  1920x1080   24.000 Hz  16:9    27.000 kHz  74.250 MHz
    VIC  34:  1920x1080   30.000 Hz  16:9    33.750 kHz  74.250 MHz
    VIC  60:  1280x720    24.000 Hz  16:9    18.000 kHz  59.400 MHz
    VIC  62:  1280x720    30.000 Hz  16:9    22.500 kHz  74.250 MHz
    VIC  18:   720x576    50.000 Hz  16:9    31.250 kHz  27.000 MHz
[... 66 lines cut ...]
    VIC 102:  4096x2160   60.000 Hz 256:135  67.500 kHz 297.000 MHz
  HDR Static Metadata Data Block:
    Electro optical transfer functions:
      Traditional gamma - SDR luminance range
      SMPTE ST2084
      Hybrid Log-Gamma
    Supported static metadata descriptors:
      Static metadata type 1
  Detailed Timing Descriptors:
    DTD 3:  1920x1080i  60.000 Hz  16:9    33.750 kHz  74.250 MHz (952 mm x 535 mm)
                 Hfront   88 Hsync  44 Hback 148 Hpol P
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vback  +0.5 Even Field
Checksum: 0x1
```
VIC 102:  4096x2160   60.000 Hz 256:135  67.500 kHz 297.000 MHz
is listed as a YCbCr 4:2:0 only format. 4:2:0 is not supported on the Pi.

4096x2160 @ 24Hz should work and would be a useful test to confirm the width/height registers are programmed correctly.

The TCL EDID appears to say the same. 

The DRM driver should have filtered out all 4:2:0 modes because it knows they aren't supported by the hardware, so have you been manually selecting them?

You can add ```drm/debug=0x14``` to /boot/cmdline.txt (don't add any carriage returns) to get a load of debug in dmesg, which includes all the parsing of the EDID for which modes it believes the hardware (both source and sink) support.

alanbork, 2021-04-15:
what do you mean by manually selecting them? I didn't add something to config.txt if that's what you mean? I used stock modetest from the drm/freedesktop project.

6by9, 2021-04-15:
OK, I'll artificially load your EDID and see what the parsing is doing.
If edid-decode is giving the correct decode of those EDIDs, then I don't believe 4096x2160 @ 60Hz is possible with your display wanting it as YCbCr 4:2:0 (which the Pi doesn't support).

Something odd is happening if it does allow it, as the driver does NOT set ```ycbcr_420_allowed``` on the connector, so https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/drm_modes.c#L1184 should reject it.

alanbork, 2021-04-15:
ok, turns out that the tv has a per-port setting for what edid to send, and I had not enabled hdmi 2.0 on that port. my bad.

alanbork, 2021-04-15:
[sonyedid hdmi2.zip](https://github.com/raspberrypi/linux/files/6320165/sonyedid.hdmi2.zip)

alanbork, 2021-04-15:
having fixed that we are back to where I was before: 

root@raspberrypi:~# modetest | grep '#'
  #0 3840x2160 60.00 3840 4016 4104 4400 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: preferred, driver
  #1 4096x2160 60.00 4096 4184 4272 4400 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #2 4096x2160 59.94 4096 4184 4272 4400 2160 2168 2178 2250 593407 flags: phsync, pvsync; type: driver
  #3 4096x2160 50.00 4096 5064 5152 5280 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #4 4096x2160 24.00 4096 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #5 4096x2160 23.98 4096 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #6 3840x2160 59.94 3840 4016 4104 4400 2160 2168 2178 2250 593407 flags: phsync, pvsync; type: driver
  #7 3840x2160 50.00 3840 4896 4984 5280 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #8 3840x2160 30.00 3840 4016 4104 4400 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #9 3840x2160 29.97 3840 4016 4104 4400 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #10 3840x2160 25.00 3840 4896 4984 5280 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #11 3840x2160 24.00 3840 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #12 3840x2160 23.98 3840 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #13 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: userdef, driver
  #14 1920x1080 59.94 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
  #15 1920x1080i 30.00 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #16 1920x1080i 29.97 1920 2008 2052 2200 1080 1084 1094 1125 74176 flags: phsync, pvsync, interlace; type: driver
  #17 1920x1080 50.00 1920 2448 2492 2640 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  #18 1920x1080i 25.00 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #19 1920x1080 30.00 1920 2008 2052 2200 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #20 1920x1080 29.97 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #21 1920x1080 24.00 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #22 1920x1080 23.98 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #23 1680x1050 59.88 1680 1728 1760 1840 1050 1053 1059 1080 119000 flags: phsync, nvsync; type: driver
  #24 1600x900 60.00 1600 1624 1704 1800 900 901 904 1000 108000 flags: phsync, pvsync; type: driver
  #25 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  #26 1152x864 75.00 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
  #27 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #28 1280x720 59.94 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #29 1280x720 50.00 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #30 1280x720 30.00 1280 3040 3080 3300 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #31 1280x720 29.97 1280 3040 3080 3300 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #32 1280x720 24.00 1280 3040 3080 3300 720 725 730 750 59400 flags: phsync, pvsync; type: driver
  #33 1280x720 23.98 1280 3040 3080 3300 720 725 730 750 59341 flags: phsync, pvsync; type: driver
  #34 1024x768 60.00 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  #35 800x600 60.32 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  #36 720x576 50.00 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  #37 720x576i 25.00 720 732 795 864 576 580 586 625 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #38 720x480 60.00 720 736 798 858 480 489 495 525 27027 flags: nhsync, nvsync; type: driver
  #39 720x480 59.94 720 736 798 858 480 489 495 525 27000 flags: nhsync, nvsync; type: driver
  #40 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #41 720x480i 29.97 720 739 801 858 480 488 494 525 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #42 640x480 60.00 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
  #43 640x480 59.94 640 656 752 800 480 490 492 525 25175 flags: nhsync, nvsync; type: driver
  #0 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: userdef, driver

modetest -s 89:3840x2160-60  WORKS
 modetest -s 89:4096x2160-50  WORKS
modetest -s 89:4096x2160-59.94 no signal, locks modetest

6by9, 2021-04-15:
I'll look at your updated EDID in a mo, but modetest with the old EDID reports modes as
```
Connectors:
id      encoder status          name            size (mm)       modes   encoders
89      88      connected       HDMI-A-1        950x540         38      88
  modes:
        index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  #0 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: preferred, driver
  #1 4096x2160 24.00 4096 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #2 4096x2160 23.98 4096 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #3 3840x2160 30.00 3840 4016 4104 4400 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #4 3840x2160 29.97 3840 4016 4104 4400 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #5 3840x2160 25.00 3840 4896 4984 5280 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #6 3840x2160 24.00 3840 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #7 3840x2160 23.98 3840 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #8 1920x1080 59.94 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
  #9 1920x1080i 30.00 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #10 1920x1080i 29.97 1920 2008 2052 2200 1080 1084 1094 1125 74176 flags: phsync, pvsync, interlace; type: driver
  #11 1920x1080 50.00 1920 2448 2492 2640 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  #12 1920x1080i 25.00 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #13 1920x1080 30.00 1920 2008 2052 2200 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #14 1920x1080 29.97 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #15 1920x1080 24.00 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #16 1920x1080 23.98 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #17 1680x1050 59.88 1680 1728 1760 1840 1050 1053 1059 1080 119000 flags: phsync, nvsync; type: driver
  #18 1600x900 60.00 1600 1624 1704 1800 900 901 904 1000 108000 flags: phsync, pvsync; type: driver
  #19 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  #20 1152x864 75.00 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
  #21 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #22 1280x720 59.94 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #23 1280x720 50.00 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #24 1280x720 30.00 1280 3040 3080 3300 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #25 1280x720 29.97 1280 3040 3080 3300 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #26 1280x720 24.00 1280 3040 3080 3300 720 725 730 750 59400 flags: phsync, pvsync; type: driver
  #27 1280x720 23.98 1280 3040 3080 3300 720 725 730 750 59341 flags: phsync, pvsync; type: driver
  #28 1024x768 60.00 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  #29 800x600 60.32 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  #30 720x576 50.00 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  #31 720x576i 25.00 720 732 795 864 576 580 586 625 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #32 720x480 60.00 720 736 798 858 480 489 495 525 27027 flags: nhsync, nvsync; type: driver
  #33 720x480 59.94 720 736 798 858 480 489 495 525 27000 flags: nhsync, nvsync; type: driver
  #34 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #35 720x480i 29.97 720 739 801 858 480 488 494 525 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #36 640x480 60.00 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
  #37 640x480 59.94 640 656 752 800 480 490 492 525 25175 flags: nhsync, nvsync; type: driver
 ```
4096x2160 @ 60 is not one of those modes

alanbork, 2021-04-15:
> 4096x2160 @ 60 is not one of those modes

yes, that's what I see too, before enabling hdmi 2.0. very sorry about that mistake. but after enabling it and rebooting the pi 4096x2160 @ 60 is back in the list.

alanbork, 2021-04-15:
after adding the drm flag to cmdline, dmesg gives this in response to  "modetest -s 89:4096x2160-60" (but only after locking up for at least a minute).

[  204.633524] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out
[  214.873435] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out
[  225.113356] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:31:plane-0] flip_done timed out
[  226.153242] ------------[ cut here ]------------
[  226.153280] WARNING: CPU: 2 PID: 599 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xec/0x128
[  226.153295] Firmware transaction timeout
[  226.153309] Modules linked in: 8021q garp stp llc brcmfmac brcmutil sha256_generic cfg80211 rfkill v3d gpu_sched raspberrypi_hwmon i2c_bcm2835 bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem vc4 bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common cec drm_kms_helper videodev drm mc vc_sm_cma(C) drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
[  226.154026] CPU: 2 PID: 599 Comm: modetest Tainted: G         C        5.10.27-v7l+ #1409
[  226.154036] Hardware name: BCM2711
[  226.154046] Backtrace:
[  226.154076] [<c0b78f58>] (dump_backtrace) from [<c0b792ec>] (show_stack+0x20/0x24)
[  226.154092]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6abc
[  226.154108] [<c0b792cc>] (show_stack) from [<c0b7d67c>] (dump_stack+0xcc/0xf8)
[  226.154128] [<c0b7d5b0>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[  226.154142]  r10:dec03008 r9:00000009 r8:c09b2fbc r7:00000040 r6:00000009 r5:c09b2fbc
[  226.154153]  r4:c0ea0020 r3:c1205094
[  226.154169] [<c0220ab0>] (__warn) from [<c0b79a90>] (warn_slowpath_fmt+0xa4/0xd8)
[  226.154182]  r7:00000040 r6:c0ea0020 r5:c1205048 r4:c0ea0040
[  226.154200] [<c0b799f0>] (warn_slowpath_fmt) from [<c09b2fbc>] (rpi_firmware_transaction+0xec/0x128)
[  226.154215]  r9:c1a954c0 r8:00000030 r7:00000000 r6:ffffff92 r5:c1a954c0 r4:c1205048
[  226.154234] [<c09b2ed0>] (rpi_firmware_transaction) from [<c09b30b4>] (rpi_firmware_property_list+0xbc/0x170)
[  226.154247]  r7:c1205048 r6:dec03000 r5:00001000 r4:dec0303c
[  226.154344] [<c09b2ff8>] (rpi_firmware_property_list) from [<bf2bc85c>] (vc4_crtc_mode_set_nofb+0x2a0/0x328 [vc4])
[  226.154358]  r10:00000465 r9:c2e1ba00 r8:c2887800 r7:c309ea14 r6:c1205048 r5:c3402840
[  226.154368]  r4:00000001
[  226.154548] [<bf2bc5bc>] (vc4_crtc_mode_set_nofb [vc4]) from [<bf31a7d0>] (drm_atomic_helper_commit_modeset_disables+0x2d0/0x4b0 [drm_kms_helper])
[  226.154562]  r10:bf32d0cd r9:c37e6740 r8:c2887800 r7:bf2dbc9c r6:bf32b488 r5:00000001
[  226.154572]  r4:c3402840
[  226.154746] [<bf31a500>] (drm_atomic_helper_commit_modeset_disables [drm_kms_helper]) from [<bf2be8b0>] (vc4_atomic_complete_commit+0xc0/0x680 [vc4])
[  226.154760]  r10:c2887800 r9:00000000 r8:c2887d00 r7:00000000 r6:c2887800 r5:c37e6740
[  226.154769]  r4:00000001
[  226.154920] [<bf2be7f0>] (vc4_atomic_complete_commit [vc4]) from [<bf2befec>] (vc4_atomic_commit+0x17c/0x1cc [vc4])
[  226.154934]  r10:c2887800 r9:00000000 r8:c2887d00 r7:00000000 r6:00000000 r5:c2887800
[  226.154944]  r4:c37e6740 r3:00000001
[  226.155220] [<bf2bee70>] (vc4_atomic_commit [vc4]) from [<bf1c7928>] (drm_atomic_commit+0x54/0x60 [drm])
[  226.155234]  r9:c35001dc r8:00000001 r7:c3402c40 r6:c2887800 r5:c37e6740 r4:00000000
[  226.155626] [<bf1c78d4>] (drm_atomic_commit [drm]) from [<bf1ddf98>] (drm_client_modeset_commit_atomic+0x208/0x248 [drm])
[  226.155639]  r7:c3402c40 r6:00000001 r5:c28879c4 r4:c37e6740
[  226.156025] [<bf1ddd90>] (drm_client_modeset_commit_atomic [drm]) from [<bf1de0c0>] (drm_client_modeset_commit_locked+0x6c/0x198 [drm])
[  226.156040]  r10:ffffe000 r9:00000000 r8:c288789c r7:c229b900 r6:bf32d020 r5:c229b918
[  226.156049]  r4:c2887800
[  226.156431] [<bf1de054>] (drm_client_modeset_commit_locked [drm]) from [<bf1de220>] (drm_client_modeset_commit+0x34/0x50 [drm])
[  226.156445]  r9:00000000 r8:c288789c r7:c229b9b4 r6:bf32d020 r5:c229b900 r4:c2887800
[  226.156733] [<bf1de1ec>] (drm_client_modeset_commit [drm]) from [<bf3222c4>] (__drm_fb_helper_restore_fbdev_mode_unlocked+0xb0/0xd8 [drm_kms_helper])
[  226.156744]  r5:c229b900 r4:00000000
[  226.156932] [<bf322214>] (__drm_fb_helper_restore_fbdev_mode_unlocked [drm_kms_helper]) from [<bf322354>] (drm_fbdev_client_restore+0x24/0x2c [drm_kms_helper])
[  226.156945]  r7:bf1fe058 r6:c28878b0 r5:c2887800 r4:c229b900
[  226.157231] [<bf322330>] (drm_fbdev_client_restore [drm_kms_helper]) from [<bf1dd9f4>] (drm_client_dev_restore+0x84/0xc8 [drm])
[  226.157615] [<bf1dd970>] (drm_client_dev_restore [drm]) from [<bf1b2840>] (drm_lastclose+0x58/0x5c [drm])
[  226.157630]  r9:00000000 r8:c2646088 r7:c35396c0 r6:c33b31c0 r5:c2887874 r4:c2887800
[  226.158014] [<bf1b27e8>] (drm_lastclose [drm]) from [<bf1b2960>] (drm_release+0x11c/0x120 [drm])
[  226.158025]  r5:c2887874 r4:c2887800
[  226.158229] [<bf1b2844>] (drm_release [drm]) from [<c043ddec>] (__fput+0x90/0x25c)
[  226.158243]  r7:c2319190 r6:000a201f r5:c3406eb8 r4:c35396c0
[  226.158258] [<c043dd5c>] (__fput) from [<c043dfd0>] (____fput+0x18/0x1c)
[... cut ...]
[  226.501070] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[  226.501078] Exception stack(0xc3c71fb0 to 0xc3c71ff8)
[  226.501088] 1fa0:                                     00000000 c00464af becee354 00000000
[  226.501098] 1fc0: 00000007 becee354 c00464af 00000036 0045f50c 00462f60 eeeeeeee 00000001
[  226.501108] 1fe0: b6edb08c becee334 b6ec188c b6e1451c 60000010 00000003
[  226.501118]  r10:00000036 r9:c3c70000 r8:c0200204 r7:00000036 r6:c00464af r5:becee354
[  226.501125]  r4:00000007
[  226.501135] ---[ end trace c8f224ca16ad4cd7 ]---

6by9, 2021-04-16:
The HDMI2 analyser has decided to work again, so I do have a test system :-)

Using your EDID, selecting 4096x2160@50 via modetest works. @60 doesn't by default.
Add 
```
core_freq=600
core_freq_min=600
```
to /boot/config.txt and selecting 4096x2160@60 works.

So it looks to be an issue of the clock being slightly too low to be able to clock through the extra pixels of 4096 vs 3840.

@timg236 @popcornmix Is it worth looking again at the clock config, or do we just document that 4096x2160@60 needs an overclock?

popcornmix, 2021-04-16:
What clock config do you think could help?

Does kms behave any differently? (with the 4kp60 patchset)

It may just be that the extra pixels from the 4096 width are too much for 550MHz core/hvs clock.
Ideally we should filter those modes out of edid if core_freq <= 600 (or whatever we believe the minimum usable frequency is)
so a user could make them available with a core_freq overclock.

6by9, 2021-04-16:
I haven't tested further. I'll try full KMS with the 4k60 patches in a bit, but I suspect they'll be the same.

4096 is 6.7% more than 3840, so plausibly 586MHz. I don't recall what the clocking arrangement is now and therefore what clock frequencies are possible.

If we can determine the required conditions, then yes we can filter them out with an appropriate warning message. We already query the firmware with RPI_FIRMWARE_GET_DISPLAY_CFG to get the max pixel clock (ie whether hdmi_enable_4k60=1), so that can gain core_clock_freq checks too.

popcornmix, 2021-04-16:
> 4096 is 6.7% more than 3840, so plausibly 586MHz. I don't recall what the clocking arrangement is now and therefore what clock frequencies are possible.

Latest firmware sets PLLA to an integer multiple of the max of core,v3d,isp,h264,hevc. So you can specify arbitrary values, i.e. core_clock=586 will get you an accurate core clock (but you may effectively be also overclocking v3d,isp,h264,hevc).

If you don't specify over_voltage yourself you will get a default one based on the increase in clocks over default.

alanbork, 2021-04-16:
Overclocking fixes the lockup - 4096x2160-60 is displayed correctly, with no tearing. 

It's not a perfect solution, however, as 3840x2160-60, which was working perfectly, now has tearing. well, tearing isn't exactly the right word - it's like a fragment of the most recently flipped buffer is shown 2 frames early.  see video: https://www.youtube.com/watch?v=3nhIT1MwPKg. use 25% speed to get to the "good part", and then , and . to step frame by frame. 

what's happening: after swapbuffer gets called a small triangular segment of the back buffer shows up on screen immediately. because of double buffering, that full frame that the fragment came from will show up as expected after 2 more refresh cycles. 

the fragment is an odd triangular cutout:

[image]

6by9, 2021-04-16:
FKMS has issues with vsync synchronisation. It's not going to be fixed as vc4-kms-v3d is the better solution. 4k60 support there should be merged within a week or two - it is all working now.

Altering the core clock has shifted the timing of V3D completing vs display being rendered.

alanbork, 2021-04-16:
any chance of getting early access to that kms-v3d binary blob? with my lag
testing setup I could do some verification that the timing is
working properly.

popcornmix, 2021-04-16:
Can you build your own kernel? It's described [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md)
You'd want to build with this PR included:
https://github.com/raspberrypi/linux/pull/4284

6by9, 2021-04-16:
Or more technically that's a backport of the upstream commits
https://patchwork.kernel.org/project/dri-devel/list/?series=466407

alanbork, 2021-04-16:
I have under x86, but it's a bit beyond my enthusiasm to help out to set it
up for the pi. I was hoping more for a binary file or two I could copy to
/boot or whatever. But if that's too much trouble I understand.

popcornmix, 2021-04-17:
You can wait. I expect it will be merged (and hence appear in rpi-update kernel) in the next week or so.

mripard, 2021-07-06:
We got a lot of fixes that might fix the issue reported here, can you test if it's still there with an updated kernel? Thanks!

closed 2023-01-24 by popcornmix