raspberrypi/linux #3844: Idle clock issue in spi_bcm2835. How I can disable it?

yushb0602, 2020-09-09:
Is there a way to disable  idle clock in spi_bcm2835?

My workplace: Linux raspberrypi 4.19.118+ , Raspbery Zero W

When use spi_bcm2835 driver to communicate with SC16IS740 controller,  SC16IS740 asks for continual clock without IDLE clock, but spi_bcm2835 feed 8+1 clock cycles/byte.  So two pairs are not matched as the following snapshot:

[image]

After investigation, I notice the snippets in /drivers/spi/spi-bcm2835.c:
```
	/* Calculate the estimated time in us the transfer runs.  Note that
	 * there is 1 idle clocks cycles after each byte getting transferred
	 * so we have 9 cycles/byte.  This is used to find the number of Hz
	 * per byte per polling limit.  E.g., we can transfer 1 byte in 30 us
	 * per 300,000 Hz of bus clock.
	 */
	hz_per_byte = polling_limit_us ? (9 * 1000000) / polling_limit_us : 0;
	byte_limit = hz_per_byte ? spi_used_hz / hz_per_byte : 1;
```

I'm  struggling with this issue  for some weeks.  Please give helps . Thank you in advance.

pelwell, 2020-09-09:
I'm not sure I understand. The whole concept of time in SPI is governed by the SCLK signal. A longer or shorter clock pulse warps time in such a way that nothing should care, _unless_ the resulting waveform ends up violating some timing constraints.

The diagram you include is just showing the order the bits are transmitted - the SCLK pulses are shown to be of uniform width but that shouldn't be required, it just looks neat.

yushb0602, 2020-09-27:
The electronics engineer says that  SC16IS740 controller could handle the IDLE clock. We guess that some  special setting for SC16IS740.  we don't think it is a issue of raspberrypi

closed 2020-09-27 by yushb0602