raspberrypi/linux #3846: Failed to resolve overlay 'vc4-kms-v3d' (dterror: can't find symbol 'i2c2') on Linux 5.9.0-rc3

jeffersonchua, 2020-09-10:
How to fix this error ... and I've tried with Linux 5.8 as well but still encountered with the same issue with loading vc4 ...

005567.330: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d.dtbo
005572.052: dterror: can't find symbol 'i2c2'
005572.063: Failed to resolve overlay 'vc4-kms-v3d'

Hardware      : BCM2711
Revision        : d03114
Model           : Raspberry Pi 4 Model B Rev 1.4

# kernel (Linux 5.9.0-rc3) compiled with ...
make ARCH=arm CROSS_COMPILE=arm-rpi-linux-gnueabihf- KERNEL=kernel7l bcm2711_defconfig
make ARCH=arm CROSS_COMPILE=/v5/cross-bcm2709/bin/arm-rpi-linux-gnueabihf- KERNEL=kernel7l

# detailed log below ...
# vcdbg log msg
004282.162: brfs: File read: 448 bytes
004317.783: brfs: File read: /mfs/sd/config.txt
005428.640: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
005431.102: *** Restart logging
005431.122: brfs: File read: 448 bytes
005440.989: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005444.360: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005444.374: HDMI0: hdmi_pixel_encoding: 300000000
005444.387: HDMI1: hdmi_pixel_encoding: 300000000
005449.122: dtb_file 'bcm2711-rpi-4-b.dtb'
005459.534: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
005459.551: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xba66
005471.926: brfs: File read: 47718 bytes
005546.572: brfs: File read: /mfs/sd/config.txt
005546.657: dtparam: audio=on
005556.085: brfs: File read: 448 bytes
005567.330: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d.dtbo
005572.052: dterror: can't find symbol 'i2c2'
005572.063: Failed to resolve overlay 'vc4-kms-v3d'
005577.713: brfs: File read: 2481 bytes
005582.104: brfs: File read: /mfs/sd/cmdline.txt
005582.156: Read command line from file 'cmdline.txt':
005582.171: 'dwc_otg.lpm_enable=0 cma=128 dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 console=tty1 consoleblank=0 root=/dev/mmcblk0p2 quiet rootwait elevator=noop sdhci-bcm2708.sync_after_dma=0 ro x11'
006670.180: brfs: File read: 190 bytes
007261.572: brfs: File read: /mfs/sd/kernel7l.img
007261.594: Loading 'kernel7l.img' to 0x8000 size 0x6138e0
007261.618: Device tree loaded to 0x2eff4100 (size 0xbef2)
007263.893: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
012229.230: vchiq_core: vchiq_init_state: slot_zero = 0xefd80000, is_master = 1

jeffersonchua, 2020-09-10:
fixed with dtoverlay=vc4-kms-v3d-pi4, but X is still on swrast, and running glxgears complainted about:

libGL error: failed to create dri screen
libGL error: failed to load driver: vc4

... but that's a separate issue.

Jeff

pelwell, 2020-09-10:
Either your firmware is old or you haven't copied overlay_map.dtb into the overlays directory.

Or both.

jeffersonchua, 2020-09-10:
> Either your firmware is old or you haven't copied overlay_map.dtb into the overlays directory.
 
overlay_map.dtb is the missing piece! Fixed after copying it!

Any tips on how to compile and run vlc on console mode without X on rpi4? Now that omxplayer is no longer supported.

I tried ... mplayer (no sound) and video (a small video only on fbdev)

# modprobe snd_bcm2835
# modprobe vc4
# cvlc --mmal-display=hdmi-0 test.mp4

Thank you!
Jeff

pelwell, 2020-09-10:
To quote a wise man:

> ... but that's a separate issue.

jeffersonchua, 2020-09-10:
# export VDPAU_DRIVER=va_gl
# ALSA_CARD=0 cvlc --mmal-display=hdmi-0 test.mp4
[b65c043c] main generic error: cannot load plug-in /usr/lib/vlc/plugins/codec/libsvgdec_plugin.so: /opt/vc/lib/libEGL.so: undefined symbol: glDiscardFramebufferEXT
[b5683608] dbus_screensaver inhibit error: cannot connect to session bus: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[b56833b8] mmal_device generic error: VCSM init failed
[b563f480] mmal_codec decoder error: Could not find a MMAL decoder device
[afb46ca8] mmal_device generic error: VCSM init failed
[afb46d20] mmal_device generic error: VCSM init failed
[b2adce0c] main generic error: cannot load plug-in /usr/lib/vlc/plugins/video_output/libegl_x11_plugin.so: /opt/vc/lib/libEGL.so: undefined symbol: glPointSizePointerOES
[afb62550] fb vout display error: cannot get terminal mode (Inappropriate ioctl for device)
ALSA lib pcm_dmix.c:1094:(snd_pcm_dmix_open) unable to open slave
[00446410] alsa audio output error: cannot open ALSA device "default": No such device
[00446410] main audio output error: Audio output failed
[00446410] main audio output error: The audio device "default" could not be used:
No such device.
[00446410] main audio output error: module not functional
[b1130d80] main decoder error: failed to create audio output

jeffersonchua, 2020-09-11:
> To quote a wise man:
> 
> > ... but that's a separate issue.

Ok :+1: 

Thank you, and we'll close this.

Jeff.

floion, 2021-08-16:
Hi @pelwell I got this issue on kernel 5.10.52 and boot firmware 20210727:

> 005235.683: dtb_file 'bcm2711-rpi-4-b.dtb'
005240.977: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
005240.994: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x8e01
005253.577: brfs: File read: 36353 bytes
005266.496: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
005314.972: brfs: File read: 1667 bytes
005317.835: brfs: File read: /mfs/sd/config.txt
005318.194: brfs: File read: 164 bytes
005329.413: Failed to load overlay 'vc4-fkms-v3d'
005337.233: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
005340.099: dterror: no symbols found
005340.108: Failed to resolve overlay 'disable-wifi'
005340.490: brfs: File read: 387 bytes
005348.836: brfs: File read: /mfs/sd/overlays/gpio-shutdown.dtbo
005352.050: dterror: no symbols found
005352.061: Failed to resolve overlay 'gpio-shutdown'

Any clue why it's failing to load any overlay?

pelwell, 2021-08-16:
```
005340.099: dterror: no symbols found
005340.108: Failed to resolve overlay 'disable-wifi'
005340.490: brfs: File read: 387 bytes
005348.836: brfs: File read: /mfs/sd/overlays/gpio-shutdown.dtbo
005352.050: dterror: no symbols found
005352.061: Failed to resolve overlay 'gpio-shutdown'
```
I'm guessing you're running an upstream kernel build. The default DTB builds don't include symbols because upstream don't support overlays. You will need to include the `-@` flag in DTC_FLAGS - see https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/Makefile#L1435 and https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm64/boot/dts/broadcom/Makefile#L19 .

floion, 2021-08-16:
I'm not running an upstream kernel. It's from git://github.com/raspberrypi/linux.git
And checked the sources and it does have the change you pointed to.
Do I need to update the rpi eeprom firmware too for a newer kernel + boot firmware combination?

pelwell, 2021-08-16:
Follow our kernel-building instructions and it should just work: https://www.raspberrypi.org/documentation/computers/linux_kernel.html#building-the-kernel

floion, 2021-08-17:
I reproduced the same error on kernel `5.10.31` on 64 bits rpi3.
And the problem does not reproduce on 32 bits kernel `5.10.31` on rpi3.

pelwell, 2021-08-17:
You are clearly doing something differently to everyone else. Post your command history for the build and installation process.

floion, 2021-08-19:
@pelwell it's being built with Yocto so it's rather convoluted to place the build logs. Made some progress with this in the sense that narrowed it down to the device tree (bcm2711-rpi-4-b.dtb). Replacing the dtb from a `5.4.83` kernel build over the one in the `5.10.31` kernel, the error is not present and things work. Trying to figure out now what changes in the newer device tree might have caused this. Or are you aware of the fact that the newer kernel might require a new dtc version to compile the device tree?

pelwell, 2021-08-19:
Symbols being present is simple binary - either they are enabled in the dtc command line or not. More recent versions of dtc have added new warnings, but the fundamentals of compiling a dts file have been the same since we first started using Device Tree on Pis back in 2014.

floion, 2021-08-19:
Thanks for the reply. Progressed a bit more. Doing this change adds the required symbols in the device tree blob:

```
diff --git a/scripts/Makefile.lib b/scripts/Makefile.lib
index 43758c8..eb49edc 100644
--- a/scripts/Makefile.lib
+++ b/scripts/Makefile.lib
@@ -296,6 +296,8 @@ DTC_FLAGS += -Wnode_name_chars_strict \
        -Winterrupt_provider
 endif
 
+DTC_FLAGS += -@
+
 DTC_FLAGS += $(DTC_FLAGS_$(basetarget))
```

Still trying to see why https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm64/boot/dts/broadcom/Makefile#L19 does not do the job for me.

floion, 2021-08-19:
Maybe https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm64/boot/dts/broadcom/Makefile#L19 does not get set because it was set before? In which case we should just `+=` it?

pelwell, 2021-08-19:
I think you might have found the answer - there's no reason why we wouldn't want to set the `-@` flag if the ARCH_BCM2835 test is true, whatever other options might have been set already.

pelwell, 2021-08-19:
See commit https://github.com/raspberrypi/linux/commit/72d78ec8b6898976ea46b1c0cb9780707985cd80.

floion, 2021-08-19:
Thanks @pelwell !