raspberrypi/linux #3849: RPi4: WiFi client crashes often (brcmf_fw_crashed: Firmware has halted or crashed)

jgilles, 2020-09-11:
**Describe the bug**
WiFi client stops working and does not recover (Cypress firmware crashes).  This happens to my RPi4 almost every day and I must reboot to recover it.  After reboot, it always works but will fail again sometime later that day (duration seems random).  Raspbian is up to date with all upgrades as of 9/11/2020.

**To reproduce**
Use WiFi client mode on RPi4.

**Expected behaviour**
WiFi keeps working reliably.

**Actual behaviour**
WiFi client stops functioning entirely.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 (4GB)

* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian GNU/Linux 10

* Which firmware version (`vcgencmd version`)?
Aug 19 2020 17:38:16
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

**Logs**
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.51-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:51:40 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49856 r8192 d23872 u81920
[    0.000000] pcpu-alloc: s49856 r8192 d23872 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:84:15:CD vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
[    0.000000] Memory: 3655620K/4050944K available (10240K kernel code, 725K rwdata, 2740K rodata, 2048K init, 853K bss, 133180K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 30051 entries in 59 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x518 with crng_init=0
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000024] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000103] bcm2835: system timer (irq = 17)
[    0.000777] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000793] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000810] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000823] Switching to timer-based delay loop, resolution 18ns
[    0.001072] Console: colour dummy device 80x30
[    0.001105] printk: console [tty1] enabled
[    0.001157] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001179] pid_max: default: 32768 minimum: 301
[... 415 lines cut ...]
[ 1637.214978] brcmfmac: dongle trap info: type 0x4 @ epc 0x0007a094
                 cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
                 lr   0x0007a093 pc   0x0007a094 offset 0x25f830
                 r0   0x00000036 r1   0x00000005 r2 0x00000000 r3 0x00000001
                 r4   0x002494d0 r5   0x00000000 r6 0x00249460 r7 0x00000004
[ 1637.214989] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1637.214998] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1697.215598] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1697.215609] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1757.216630] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1757.216640] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1817.217636] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1817.217646] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1877.218660] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1877.218671] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1937.219692] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1937.219711] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 1997.220728] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 1997.220747] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[ 2057.221724] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 2057.221742] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
 --- This will continue indefinitely ---

**Additional context**
I am reproducing this daily with a TP-Link Archer A7 router (5GHz)

JamesH65, 2020-09-12:
Is it possible to test at 2,4G to see if that works OK? What sort of signal strength are you seeing - is it right on the edge? Does moving the Pi closer make the problem more or less common?

jgilles, 2020-09-14:
Looks like it is already using 2.4GHz (I thought it was using 5GHz since the LXDE WiFi panel control shows 5G and the router supports it).  Signal strength is strong.

Here is the info from wpa_cli:
```
> signal_poll
RSSI=-51
LINKSPEED=24
NOISE=9999
FREQUENCY=2417
WIDTH=40 MHz
CENTER_FRQ1=2427
```

jgilles, 2020-09-30:
Is there any other debugging I can do to help?

cbhp, 2020-10-04:
I'm having the same WIFI issues at the moment and I have to reboot the raspberry.

syslog:
```
Sep 29 18:42:59 rpi systemd[1]: Started Cleanup of Temporary Directories.
Sep 29 18:52:58 rpi kernel: [260835.693136] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Sep 29 19:17:01 rpi CRON[31988]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
```

kern.log:
```
Sep 29 06:25:03 rpi kernel: [215959.278512] Process accounting resumed
Sep 29 18:52:58 rpi kernel: [260835.693136] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Sep 29 20:19:50 rpi kernel: [266047.569262] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sep 29 20:20:03 rpi kernel: [266060.769503] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sep 30 06:25:03 rpi kernel: [302360.950984] Process accounting resumed
```

System:
Raspberry Pi 4 Model B Rev 1.2, 2 GB variant
Linux version 5.4.51-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:51:40 BST 2020
raspberrypi-firmware soc:firmware: Attached to firmware from 2020-09-02 21:13, variant start
raspberrypi-firmware soc:firmware: Firmware hash is 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43

WIFI Router: 
TP-Link Archer C7

jspuij, 2020-10-05:
It's a bug with the IEEE 802.11r fast roaming implementation in the Wifi driver. Disabling fast roaming on my mesh router fixed this issue.

cbhp, 2020-10-07:
> It's a bug with the IEEE 802.11r fast roaming implementation in the Wifi driver. Disabling fast roaming on my mesh router fixed this issue.

Indeed I have a mesh setup with several routers, but ieee80211r (Fast BSS transition) is not active. How did you come up with that? Do you have any tips how to find out the reason for the crashes in my setup?

jgilles, 2020-10-07:
I only have a single router, no mesh.  I have also tried using separate
SSIDs for 2.4GHz and 5GHz, so that the Pi would not switch between them and
it didn't help.

jspuij, 2020-10-07:
Got the answer here: https://github.com/raspberrypi/linux/issues/1342#issuecomment-681948365

jgilles, 2020-10-07:
I can't find any setting on my TP-Link Archer A7 to disable fast roaming, unfortunately.  There is only a section for "OneMesh" configuration and it doesn't let you change anything, especially if you don't have other devices...so I don't have a way to test this.

freeforall2, 2020-11-30:
Having the same issue here. Wifi is crashing on new RPi4 and the only way to recover is a reboot. As a Wi-fi AP, I am using Netgear Orbi. 

[  587.355859] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[  589.770561] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  589.771103] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[  589.771398] brcmfmac: dongle trap info: type 0x4 @ epc 0x0007a094
                 cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
                 lr   0x0007a093 pc   0x0007a094 offset 0x25f830
                 r0   0x00000036 r1   0x00000005 r2 0x00000000 r3 0x00000001
                 r4   0x002494d0 r5   0x00000000 r6 0x00249460 r7 0x00000004
[  589.771423] ieee80211 phy0: brcmf_run_escan: error (-110)
[  589.771444] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
[  649.771263] ieee80211 phy0: brcmf_run_escan: error (-110)

cbhp, 2020-11-30:
@freeforall2 Rebooting is also my only current workaround 🙁

pelwell, 2020-12-01:
What does `dmesg | grep Firmware:` show, @freeforall2?

Anyone experiencing a crash like this, please download the new clm_blob file: https://drive.google.com/file/d/1J8DdbsrZcSkDYKUPsdy2RvncttSwQdBH/view?usp=sharing

Install it with:
```
pi@raspberrypi:~$ sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob{,.bak}
pi@raspberrypi:~$ sudo cp brcmfmac43456-sdio.clm_blob /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
```
[ yes - there is a rename from 43456 to 43455 in that copy ]

freeforall2, 2020-12-01:
@pelwell here is the output :

[    0.070883] raspberrypi-firmware soc:firmware: Firmware hash is 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43
[    5.383337] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea

pelwell, 2020-12-01:
Thanks - that is the most recent firmware we have. It's interesting that the location of the firmware crash is the same as in the OP. I'll flag it up to Cypress.

freeforall2, 2020-12-01:
@pelwell Thanks ! Do you still suggest that I install the clm_blob you sent for testing ?

pelwell, 2020-12-01:
Yes please - it _might_ at least avoid the problem for you.

pelwell, 2020-12-01:
Cypress have provided another firmware to test. I suspect it's just a current build rather than one expected to solve the problem, but it's worth trying: https://drive.google.com/file/d/13u_Ipf6yUATl38HyVNiHG3H2BVxCzfkB/view?usp=sharing
You'll need to rename it when instalingl it (and reboot afterwards):
```
pi@raspberrypi:~$ sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{,.bak}
pi@raspberrypi:~$ sudo cp cyfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
```

freeforall2, 2020-12-01:
@pelwell I installed the clm_blob you provided. I am going to monitor for 24 or 48 hours. If it doesn't fix the issue, I will install the new firmware. Let me know if it doesn't make sense and I should go with a different approach. Thanks !

pelwell, 2020-12-01:
No, that sounds good - it's usually best to change one thing at a time.

pelwell, 2020-12-03:
There's now a dedicated test image Cypress would like you to run: https://drive.google.com/file/d/1ggSFL4YvPRkm0e-uYdrYWGG1TBEprtat/view?usp=sharing
Again, it will have to be renamed `bcrmfmac43455-sdio.bin` when installing it.

Please try it, and upload the logs for any failures you see.

freeforall2, 2020-12-03:
@pelwell, thanks. So here is my feedback : I only installed the clm_blob you provided and it looks like it fixed the issue (no crash since 48h while before I had at least 1 crash per day). 
Question : What do you recommend as a next step ? Should I revert to the old clm_blob and install the last bcrmfmac43455-sdio.bin you provided ? or should I keep the clm_blob and it can be considered as the fix ?

pelwell, 2020-12-03:
The firmware shouldn't crash, no matter what, so if you are willing it would be nice to install the trial firmware, revert the blob and let it crash once (or perhaps it won't), grabbing the kernel log. After that you can apt update and sudo apt install firmware-brcm80211 to restore the new blob (we released it yesterday).

freeforall2, 2020-12-04:
@pelwell before installing the trial firmware I decided to continue monitoring for an extra couple days. It actually just crashed ( with the clm_blob you provided still in place). You can find below the log (please tell me if you need more logs). What would you recommend at this stage ? should I revert the clm_blob to the original and install the trial firmware ? 
Thanks for your help !

Dec  4 21:16:05 raspberrypi kernel: [253847.841829] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Dec  4 21:16:07 raspberrypi kernel: [253850.279910] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Dec  4 21:16:07 raspberrypi kernel: [253850.280473] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Dec  4 21:16:07 raspberrypi kernel: [253850.280764] brcmfmac: dongle trap info: type 0x4 @ epc 0x0007a094
Dec  4 21:16:07 raspberrypi kernel: [253850.280764]   cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
Dec  4 21:16:07 raspberrypi kernel: [253850.280764]   lr   0x0007a093 pc   0x0007a094 offset 0x25f830
Dec  4 21:16:07 raspberrypi kernel: [253850.280764]   r0   0x00000036 r1   0x00000005 r2 0x00000000 r3 0x00000001
Dec  4 21:16:07 raspberrypi kernel: [253850.280764]   r4   0x002494d0 r5   0x00000000 r6 0x00249460 r7 0x00000004
Dec  4 21:16:07 raspberrypi kernel: [253850.280792] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  4 21:16:07 raspberrypi kernel: [253850.280813] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)

jspuij, 2020-12-04:
I can confirm that I have the same issue as well still with the new firmware

pelwell, 2020-12-04:
Run the new firmware with `brcmfmac.debug=0x100000` in cmdline.txt, and when it crashes upload the output from `dmesg`.

freeforall2, 2020-12-04:
@pelwell - thanks. I installed the trial firmware and I set brcmfmac.debug=0x100000. I will send you syslog output when it crashes.

freeforall2, 2020-12-07:
@pelwell please find below the log (crash at 05:13)- don't hesitate if you need anything else from my side. Thanks

Dec  7 05:12:48 raspberrypi kernel: [168150.183606] brcmfmac: CONSOLE: 167750.419 wlc_rrm_timer: state upd to 5
Dec  7 05:12:48 raspberrypi kernel: [168150.183621] brcmfmac: CONSOLE: 167750.419 wl0: wlc_rrm_timer: WAIT_STOP_TRAFFIC, 0 ms until measurement, waiting for PS announcement
Dec  7 05:12:48 raspberrypi kernel: [168150.183634] brcmfmac: CONSOLE: 167750.419 wlc_rrm_begin, state upd to 4
Dec  7 05:12:48 raspberrypi kernel: [168150.183646] brcmfmac: CONSOLE: 167750.419 wlc_rrm_begin: calling wlc_set_chanspec
Dec  7 05:12:48 raspberrypi kernel: [168150.183661] brcmfmac: CONSOLE: 167750.420 wlc_rrm_begin_scan: duration: 100, channel num: 1, bssid: ff:ff:ff:ff:ff:ff
Dec  7 05:12:49 raspberrypi kernel: [168150.613547] brcmfmac: CONSOLE: 167750.522 wlc_rrm_bcnreq_scancb: state: 7, status: 0
Dec  7 05:12:49 raspberrypi kernel: [168150.613562] brcmfmac: CONSOLE: 167750.522 wlc_rrm_bcnreq_scancb: scan count: 0
Dec  7 05:12:49 raspberrypi kernel: [168150.613575] brcmfmac: CONSOLE: 167750.532 wlc_rrm_meas_end: return to the home channel
Dec  7 05:12:49 raspberrypi kernel: [168150.613588] brcmfmac: CONSOLE: 167750.533 wlc_rrm_report_11k: scan results count = 0
Dec  7 05:13:07 raspberrypi kernel: [168168.633999] brcmfmac: CONSOLE: 167768.820 wl0.0: wlc_rrm_timer: WAIT_STOP_TRAFFIC, entering PS mode for off channel measurement
Dec  7 05:13:07 raspberrypi kernel: [168168.634015] brcmfmac: CONSOLE: 167768.821 wlc_rrm_timer: state upd to 5
Dec  7 05:13:07 raspberrypi kernel: [168168.634030] brcmfmac: CONSOLE: 167768.821 wl0: wlc_rrm_timer: WAIT_STOP_TRAFFIC, 0 ms until measurement, waiting for PS announcement
Dec  7 05:13:07 raspberrypi kernel: [168168.634042] brcmfmac: CONSOLE: 167768.821 wlc_rrm_begin, state upd to 4
Dec  7 05:13:07 raspberrypi kernel: [168168.634055] brcmfmac: CONSOLE: 167768.821 wlc_rrm_begin: calling wlc_set_chanspec
Dec  7 05:13:07 raspberrypi kernel: [168168.634071] brcmfmac: CONSOLE: 167768.823 wlc_rrm_begin_scan: duration: 100, channel num: 1, bssid: ff:ff:ff:ff:ff:ff
Dec  7 05:13:07 raspberrypi kernel: [168168.693940] brcmfmac: CONSOLE: 67768.883 sp+1e8 0000018b
Dec  7 05:13:07 raspberrypi kernel: [168168.693954] brcmfmac: CONSOLE: 167768.883 sp+200 0000018d
Dec  7 05:13:07 raspberrypi kernel: [168168.693967] brcmfmac: CONSOLE: 167768.883 sp+20c 00022cf7
Dec  7 05:13:07 raspberrypi kernel: [168168.693979] brcmfmac: CONSOLE: 167768.883 sp+21c 001a6291
Dec  7 05:13:07 raspberrypi kernel: [168168.693990] brcmfmac: CONSOLE: 167768.883 sp+270 00000335
Dec  7 05:13:07 raspberrypi kernel: [168168.767812] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Dec  7 05:13:09 raspberrypi kernel: [168171.183635] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Dec  7 05:13:09 raspberrypi kernel: [168171.184053] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Dec  7 05:13:09 raspberrypi kernel: [168171.184341] brcmfmac: dongle trap info: type 0x4 @ epc 0x001e85c6
Dec  7 05:13:09 raspberrypi kernel: [168171.184341]   cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
Dec  7 05:13:09 raspberrypi kernel: [168171.184341]   lr   0x001e85c7 pc   0x001e85c6 offset 0x25f830
Dec  7 05:13:09 raspberrypi kernel: [168171.184341]   r0   0x0000003f r1   0x00000005 r2 0x00000000 r3 0x00000001
Dec  7 05:13:09 raspberrypi kernel: [168171.184341]   r4   0x002494ac r5   0x00000000 r6 0x0024943c r7 0x00000004
Dec  7 05:13:09 raspberrypi kernel: [168171.184368] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:13:09 raspberrypi kernel: [168171.184387] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:14:09 raspberrypi kernel: [168231.184859] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Dec  7 05:14:09 raspberrypi kernel: [168231.185288] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Dec  7 05:14:09 raspberrypi kernel: [168231.185578] brcmfmac: dongle trap info: type 0x4 @ epc 0x001e85c6
Dec  7 05:14:09 raspberrypi kernel: [168231.185578]   cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
Dec  7 05:14:09 raspberrypi kernel: [168231.185578]   lr   0x001e85c7 pc   0x001e85c6 offset 0x25f830
Dec  7 05:14:09 raspberrypi kernel: [168231.185578]   r0   0x0000003f r1   0x00000005 r2 0x00000000 r3 0x00000001
Dec  7 05:14:09 raspberrypi kernel: [168231.185578]   r4   0x002494ac r5   0x00000000 r6 0x0024943c r7 0x00000004
Dec  7 05:14:09 raspberrypi kernel: [168231.185603] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:14:09 raspberrypi kernel: [168231.185623] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:15:09 raspberrypi kernel: [168291.186121] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:15:09 raspberrypi kernel: [168291.186147] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:16:09 raspberrypi kernel: [168351.187333] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:16:09 raspberrypi kernel: [168351.187355] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:17:01 raspberrypi CRON[9031]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Dec  7 05:17:09 raspberrypi kernel: [168411.188524] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:17:09 raspberrypi kernel: [168411.188548] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:18:09 raspberrypi kernel: [168471.189710] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:18:09 raspberrypi kernel: [168471.189733] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec  7 05:19:09 raspberrypi kernel: [168531.190900] ieee80211 phy0: brcmf_run_escan: error (-110)
Dec  7 05:19:09 raspberrypi kernel: [168531.190924] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)

pelwell, 2020-12-07:
Thanks - I've forwarded the log to Cypress.

pelwell, 2020-12-08:
In a remarkably quick turnaround there's a test firmware with a possible fix: https://drive.google.com/file/d/1YOvhSmyCmQkoF9vcsRDD5oxFkj0VGuM7/view?usp=sharing
(It will have to be renamed and installed as before).

freeforall2, 2020-12-08:
@pelwell thanks ! it was very quick ! I installed the test firmware. I will monitor for a few days and let you know the results. Just one thing : at some point, I reverted to the old clm_blob. Do you confirm that I shouldn't touch it for now. Or should I update it ? Thanks

pelwell, 2020-12-08:
I don't think it will make a difference, but since the new blob is part of the current firmware package I would prefer you to use it.

Sprout-man, 2020-12-15:
@pelwell Created an account to say thanks for posting the test firmware, and note that it's been solid for me the last several days. Prior to installation, I was the in the same boat, with a regular daily loss of wifi connectivity, but so far no issues post-update (~4 days at this point).

freeforall2, 2020-12-15:
@pelwell same here, no crash since 7 days. I can confirm that it fixed the issue. Thanks !!

pelwell, 2020-12-15:
Thanks - I'll wait a while for the clm_blob change to hit a wider audience, then release this.

robertaramar, 2020-12-21:
Is the clm_blob now released? I did an update and after that downloaded your clm_blob from Google drive and found their md5sums be identical.

pelwell, 2020-12-21:
Yes - the new clm_blobs where published at the start of the month (it was mentioned in passing here: https://github.com/raspberrypi/linux/issues/3849#issuecomment-738295473)

robertaramar, 2020-12-23:
I have installed the most recent version of the firmware-brcm80211 but I am still experiencing the above issue:
`Dec 23 11:37:23 raspi4one kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 23 11:37:23 raspi4one kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Dec 23 11:37:26 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_set_power_mgmt: error (-110)
Dec 23 11:37:28 raspi4one kernel: ieee80211 phy0: _brcmf_set_multicast_list: Setting mcast_list failed, -110
Dec 23 11:37:31 raspi4one kernel: ieee80211 phy0: _brcmf_set_multicast_list: Setting allmulti failed, -110
Dec 23 11:37:33 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_set_power_mgmt: Unable to set pm timeout, (-110)
Dec 23 11:37:36 raspi4one kernel: ieee80211 phy0: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
Dec 23 11:37:41 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:37:41 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:37:46 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:37:46 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:37:50 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:37:50 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:37:53 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:37:53 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:37:57 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:37:57 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:00 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:00 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:04 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:04 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:07 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:07 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:11 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:11 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:14 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:14 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:17 raspi4one kernel: ieee80211 phy0: brcmf_run_escan: error (-110)
Dec 23 11:38:17 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Dec 23 11:38:31 raspi4one kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Dec 23 11:38:31 raspi4one kernel: ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
`
Is there anything I can do/contribute to diagnose that issue and help fixing it. It bugs me as my Pi has to be manually rebooted every morning.
Thanks and regards,
Robert

wagnerch, 2020-12-23:
+1

64-bit, Pi4, 8GB

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.79-v8+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
...
[274647.008442] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[274649.611781] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[274649.612276] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[274649.612525] brcmfmac: dongle trap info: type 0x4 @ epc 0x0007a094
                  cpsr 0x6000019f spsr 0x600001bf sp 0x0025f888
                  lr   0x0007a093 pc   0x0007a094 offset 0x25f830
                  r0   0x00000036 r1   0x00000005 r2 0x00000000 r3 0x00000001
                  r4   0x002494d0 r5   0x00000000 r6 0x00249460 r7 0x00000004
[274649.612544] ieee80211 phy0: brcmf_run_escan: error (-110)
[274649.612559] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
```

wagnerch, 2020-12-23:
Installing test firmware from https://github.com/raspberrypi/linux/issues/3849#issuecomment-740471341 now, and enabled debug on cmdline.txt:
[    7.323646] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Dec  7 2020 20:18:26 version 7.45.444 (PRKE-TEST) (7562bb1 CY) FWID 01-6e7b6663

Will let you know if I see any problems with this test firmware.  It does appear to have a later version than the current one, even if the build date is earlier than the current.

pelwell, 2021-01-05:
There is a release candidate firmware available for testing here: https://drive.google.com/file/d/11szy6tlDFL5d7A6JdZLEwr0ZE1yQfNkE/view?usp=sharing

All being well it will be released in the next few weeks.

TamariusX, 2021-01-05:
@pelwell, thank you very much. Fixed it for one RPi4 with the first fix - no crashes ever since
Will test the release candidate right now with another one.

And yes, you all were right, Fritzbox 6660 here - Mesh was and is the culprit.

KnuckleBob, 2021-01-06:
Do I need to install the clm_blob & .bin file together?

pelwell, 2021-01-06:
No - the 3B uses the BCM43438 which uses different firmware.

Predator83, 2021-01-07:
Hi pelwell,
Thanks for digging into this! I am facing the same problem as you do and i would like to try the firmware you posted 2 days ago.

As I am really new to the Raspi, would you mind posting how to install? I know you did this earlier but when trying to follow these steps I completely messed up my Wifi. I think it was not completely clear to me which files to replace and which files to keep.

pelwell, 2021-01-07:
This is how to install that test firmware:
1. Follow this link and download the file there: https://drive.google.com/file/d/11szy6tlDFL5d7A6JdZLEwr0ZE1yQfNkE/view?usp=sharing
2.

pelwell, 2021-01-07:
[Pro Tip: - don't use Ctrl Enter when you mean Shift Enter]
Let's try that again:

1. Follow this link and download the file there: https://drive.google.com/file/d/11szy6tlDFL5d7A6JdZLEwr0ZE1yQfNkE/view?usp=sharing
It will probably end up in ~pi/Downloads and be called `brcmfmac43455-sdio.bin`.
2. In a terminal window, use `cd` to navigate to where the downloaded firmware is, e.g. `cd ~pi/Downloads` - it makes the instructions simpler.
3. `sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{,.orig}`
The funny punctuation at the end causes the filename to be repeated - once without `.orig` on the end and once with it.
4. `sudo cp brcmfmac43455-sdio.bin /lib/firmware/brcm`

You can now reboot to try the new firmware.

Reverting to the old firmware is just a matter of moving the comma at the end of step 3:
`sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{.orig,}`

Predator83, 2021-01-07:
So you basically backup the original file „brcmfmac43455-sdio.bin“ by renaming it with the ending .orig and replace it with the new one? And only this Single file?

pelwell, 2021-01-07:
Yes - that installs the new firmware file. We have also changed the clm_blob file recently, but you can get the latest version of that by running `sudo apt update` and `sudo apt install firmware-brcm80211`. Note though that this may overwrite the test firmware you just installed, so it's best to run the apt commands first (or rerun the `mv` and `cp` again afterwards).

cpmcsweeny, 2021-01-09:
> I'm having the same crashes on an RPi 3 B Rev 1.2 & WiFi goes away after some minutes or hours:
> "ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed"
> 
> Is the same chipset used on the RPi 4 & my older model?
> 
> Do I need to install the clm_blob & .bin file together?

> No - the 3B uses the BCM43438 which needs different firmware.

Just to save anybody else wasting time on this, it appears the file required for the BCM43438 on the RPi3 is brcmfmac43430-sdio.bin

I came from https://raspberrypi.stackexchange.com/questions/89803/access-point-as-wifi-router-repeater-optional-with-bridge - does anybody know if I need an update for the BCM43438 on the RPi3 (KnuckleBob suggests it has the same issue, it's not something I've experienced, but I'm doing a new build and my previous build appears to predate this issue). If so, is there an update for brcmfmac43430-sdio.bin and where would I find it?

djohnanderson, 2021-01-10:
This new firmware fixes my problems with wifi on a Raspberry Pi 4 8Gb. From what I can tell this bug has been around for more than a year and affects both Raspberry Pi 3 and 4s. How could such an obvious bug go unfixed for so long????

pelwell, 2021-01-10:
Thanks for the feedback, and that's a good question. It seems that WiFi networks are far more varied than Ethernet, and some people see problems while others never do (and I was one of the latter until I investigated an issue with Pi 400 and a particular AP that I happen to have at home).

djohnanderson, 2021-01-10:
That makes sense. For what it's worth I'm using a recent Netgear ORBI wifi 6 mesh network. Although the firmware no longer crashes on the Pi and seems to work ok I still get a bunch of these errors:

[21797.439240] ieee80211 phy0: brcmf_run_escan: error (-52)
[21797.439256] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

Taomyn, 2021-01-12:
I tried the the new firmware on my 4GB Pi4 (official PSU, headless and running OctoPrint, so no video connected) and it has made no difference, well I say no difference, the big difference seems to me that nothing is being logged by the firmware now in any of the files in /var/log. So either I'm looking in the wrong places, dmesg says nothing as well, or it's cached somewhere else. I'm fully updated as far as I can tell after running "apt update/upgrade" today.

I'm connecting to a UBNT access point at 5GHz there are a couple of other devices on the same and they don't have any problems, it's just the Pi that randomly stops getting connectivity. Luckily I simple "reconnect" on the Ubiquity console and the Pi after a few seconds reconnects until it fails again some random time later.

djohnanderson, 2021-01-12:
All my failures were logged with dmesg, so if you don't see errors there, e.g. firmware crashed, you probably have a different problem.

Taomyn, 2021-01-12:
> All my failures were logged with dmesg, so if you don't see errors there, e.g. firmware crashed, you probably have a different problem.

I was getting the messages before, that's why I came here to try the new firmware

djohnanderson, 2021-01-12:
Sounds to like you have two problems and one of them was fixed by the firmware

lyallaward, 2021-01-14:
I've been having this issue with a RPi4 (8GB) also. 

I've updated the clm_blob via 
`sudo apt update && sudo apt install firmware-brcm80211 `
and followed the instructions from @pelwell in this post https://github.com/raspberrypi/linux/issues/3849#issuecomment-756399407

Will update when I've had it running for a while.

TamariusX, 2021-01-14:
RPi 4 with 2 GB and with 4 GB - 9 days in, no wifi crash and no issues yet with the release candidate. 
RPi 3 - I can confirm that the same issue with Mesh does exist with the RPi 3b+

Provoking Mesh Handovers to a repeater can trigger more of the relevant packages and provoke the crashes - having the wifi setup done poorly help's massive in testing (not so much for the stability of course)

AlexanderNaehring, 2021-01-22:
I expect that I had the same issue. I have installed the test FW on one of my Pi 4 and will report if I still get connection drops.
Found this GH issue here in [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=278393&hilit=wifi+connection).
I also have a Mesh Wifi (wired backbone, multiple AP spread in my house, but Pi is close to one of them and not moved).
Unfortunately I have not recorded `dmesg` output from when the issue arised but will observe all my Pi closely in the next days.

lyallaward, 2021-01-24:
> I've been having this issue with a RPi4 (8GB) also.
> 
> I've updated the clm_blob via
> `sudo apt update && sudo apt install firmware-brcm80211 `
> and followed the instructions from @pelwell in this post [#3849 (comment)](https://github.com/raspberrypi/linux/issues/3849#issuecomment-756399407)
> 
> Will update when I've had it running for a while.

I've had the Pi running for 11 days since I installed the test firmware. I've been able to access it every time I've tried via the wifi so I'm thinking I might have found the issue. Previously I had a single run of about 5 days without issue. I'll also point out I don't have a multiple access point wifi network. Though my access point does offer both 2.4 and 5 GHz on the same SSID which might end up with the same mesh handoff events if a device switches frequencies.

[94 more replies]