raspberrypi/linux #3850: [5.9] intermittently semi-broken network after boot on RPi4/kernel 5.9

HiassofT, 2020-09-14:
**Describe the bug**

Sometimes I get huge ping times on ethernet, network packets going out very slowly from RPi4 (up to several seconds between tcp packets) and (almost) non-functional NFS  after boot.

Most of the time network performance is fine though (about 0.1ms ping time and linux part of NFS boot completed in about 20 seconds).

**To reproduce**
So far I've only seen the issue when netbooting the RPi4, I wouldn't rule out though that it can happen with SD card boot as well. I usually get the issue in about 1-3 out of 10 boots, sometimes it can need more than 10 reboots though.

**Expected behaviour**
Network performs normally (low ping, working NFS mounts), like with 5.4 kernel.

**Actual behaviour**
Kernel level DHCP configuration and root NFS mount sometimes take very long, NFS performance is sometimes so bad that the userspace system fails to start up.

**System**
4GB RPi4 with B1 stepping, Sep 03 boot eeprom, Sep 11 firmware, kernel 5.9-rc4 githash 4ba2756fde64c151a2bd99d6320b11b3bba0cfde

Tested with current 32bit RaspiOS (bcm2711_defconfig) and LibreELEC master (custom kernel config).

**Logs**
Working network, as a reference: http://ix.io/2xzu , ping time is about 0.1ms

One try with semi-working NFS, systemd somewhat came up http://ix.io/2xzw , ping was about 300-500ms:
```
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=1 ttl=64 time=0.494 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=2 ttl=64 time=0.442 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=3 ttl=64 time=0.360 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=4 ttl=64 time=0.464 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=5 ttl=64 time=0.425 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=6 ttl=64 time=0.344 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=7 ttl=64 time=0.404 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=8 ttl=64 time=0.345 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=9 ttl=64 time=0.425 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=10 ttl=64 time=0.387 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=11 ttl=64 time=0.330 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=12 ttl=64 time=0.435 ms
```

Another try, NFS pretty much unusable: http://ix.io/2xzx , ping up to 1024ms
```
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=1 ttl=64 time=0.333 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=2 ttl=64 time=0.348 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=3 ttl=64 time=0.267 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=4 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=5 ttl=64 time=0.312 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=6 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=7 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=8 ttl=64 time=0.293 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=9 ttl=64 time=0.334 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=10 ttl=64 time=0.317 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=11 ttl=64 time=0.342 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=12 ttl=64 time=0.336 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=13 ttl=64 time=0.322 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=14 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=15 ttl=64 time=0.291 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=16 ttl=64 time=0.337 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=17 ttl=64 time=0.250 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=18 ttl=64 time=0.349 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=19 ttl=64 time=0.321 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=20 ttl=64 time=0.270 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=21 ttl=64 time=0.382 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=22 ttl=64 time=0.305 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=23 ttl=64 time=0.239 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=24 ttl=64 time=0.409 ms
```

**Additional context**

In other tests I had seen rather tunny ping times of 1024, 2048, 3072 and 4906ms.

I had seen this issue with the previous rpi-eeprom version as well, but only on kernel 5.9. I used that very often to netboot kernel 5.4 systems (mainly LibreELEC)

pelwell, 2020-09-15:
@popcornmix Have you seen any connectivity issues with an NFS root?

popcornmix, 2020-09-15:
No I haven't. I'm still using sdcard from bootrom, then nfs for rootfs and that seems okay in my setup.
I can leave a reboot loop going on 5.9 later, but it's not an issue I've seen on numerous boots.
I think @HiassofT has network boot from bootrom.

HiassofT, 2020-09-15:
Yes, I'm using TFTP boot from the bootrom here.

So far I haven't seen the issue with SD card boot yet, but due to the intermittent nature I wouldn't dare to say that's related at this point (already got another red herring when looking into this issue)

HiassofT, 2020-09-17:
I just caught a not-too-broken-network state where LibreELEC managed to copy the squashfs root to RAM. Network and NFS are slow but I've got a running userspace.

Ping me if you'd like me to do some checks (peeking with miitool etc).

popcornmix, 2020-09-17:
I've got a reboot loop running on 5.9-rc5 with nfs booted raspbian (using sdcard for firmware). 50 reboots and counting.
I'll switch it to a LibeELEC build on 5.9 later.

Are you able to test with a different Pi4 or a different switch? Also curious if unplugging/replugging ethernet affects it.
Afraid I don't know muck about miitool diagnostics.

HiassofT, 2020-09-17:
Replugging didn't help, tried both plugging it back into the original switch (HP 1810-24G) and into another one (Netgear GS108). dmesg showed the link down/up status change but ping was still ~1000ms
```
[15437.671394] bcmgenet fd580000.ethernet eth0: Link is Down                    
[15441.726886] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off                                                                      
[15538.041544] bcmgenet fd580000.ethernet eth0: Link is Down                    
[15550.207741] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx                      
```
```
hias@camel2:~$ ping rpi4b1
PING rpi4b1.lan (192.168.1.70) 56(84) bytes of data.
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=1 ttl=64 time=1081 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=2 ttl=64 time=47.6 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=3 ttl=64 time=1047 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=4 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=5 ttl=64 time=1024 ms
64 bytes from rpi4b1.lan (192.168.1.70): icmp_seq=6 ttl=64 time=256 ms
```

I'll do test with another RPi4 and boot-tests with the Netgear switch a bit later.

popcornmix, 2020-09-17:
I assume 5.4 has always been reliable for booting? And 5.9 always unreliable?
Had you tested any kernels in between?

HiassofT, 2020-09-17:
yes, 5.4 has always been fine. 5.9 has these intermittent issues (i.e. mostly fine). haven't tested any kernels in between yet

popcornmix, 2020-09-17:
Can you try adding `genet.skip_umac_reset=1` to cmdline.txt?
That seems to be set differently between 5.4 and 5.9.

HiassofT, 2020-09-17:
I checked another RPi4 (2GB) model with the Netgear switch and got slow net on the second reboot.
```
Hardware        : BCM2711
Revision        : b03111
Serial          : 100000006a339d6f
Model           : Raspberry Pi 4 Model B Rev 1.1
```
Sep 3 bootloader, only change is BOOT_UART=1 and BOOT_ORDER=0xf241, Sep 16 fw and 5.9.0-rc5

will give umac_reset a try now.

HiassofT, 2020-09-17:
with `genet.skip_umac_reset=1` dmesg is spammed with "eth0: hw csum failure" errors, dumps and backtraces http://ix.io/2xT0
```
[    7.222757] Sending DHCP requests .
[    7.230477] eth0: hw csum failure
[    7.237285] skb len=340 headroom=144 headlen=340 tailroom=1692
[    7.237285] mac=(130,14) net=(144,20) trans=164
[    7.237285] shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
[    7.237285] csum(0x78fc ip_summed=2 complete_sw=0 valid=0 level=0)
[    7.237285] hash(0x0 sw=0 l4=0) proto=0x0800 pkttype=0 iif=0
[    7.265611] dev name=eth0 feat=0x0x0000010000004829
[    7.270501] skb headroom: 00000000: 2f 75 73 72 2f 6c 69 62 2f 6b 65 72 6e 65 6c 2d
[    7.278171] skb headroom: 00000010: 6f 76 65 72 6c 61 79 73 2f 62 61 73 65 2f 6c 69
[    7.285841] skb headroom: 00000020: 62 2f 66 69 72 6d 77 61 72 65 2f 62 72 63 6d 2f
[    7.293499] skb headroom: 00000030: 42 43 4d 32 30 37 30 32 41 30 2d 30 61 35 63 2d
[    7.301152] skb headroom: 00000040: 80 7f a8 01 80 80 00 00 78 fc 00 00 00 00 00 00
[    7.308802] skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.316455] skb headroom: 00000060: fb 49 f7 0e 81 bf d4 02 bf d7 5b fe 24 7e 31 7e
[    7.324108] skb headroom: 00000070: 07 f1 03 7f bd b7 78 3e fa 8d 3e 7e 5f 6d df 8f
[    7.331762] skb headroom: 00000080: 00 00 dc a6 32 0a 84 2f c0 4a 00 2d 38 94 08 00
[    7.339416] skb linear:   00000000: 45 c0 01 54 9e 73 00 00 40 11 55 d7 c0 a8 01 fa
[    7.347069] skb linear:   00000010: c0 a8 01 44 00 43 00 44 01 40 b3 a3 02 01 06 00
[    7.354721] skb linear:   00000020: 99 dd e9 02 00 00 00 00 00 00 00 00 c0 a8 01 44
[    7.362374] skb linear:   00000030: c0 a8 01 fa 00 00 00 00 dc a6 32 0a 84 2f 00 00
[    7.370027] skb linear:   00000040: 00 00 00 00 00 00 00 00 31 39 32 2e 31 36 38 2e
[    7.377681] skb linear:   00000050: 31 2e 32 30 00 00 00 00 00 00 00 00 00 00 00 00
[    7.385335] skb linear:   00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.392989] skb linear:   00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.400642] skb linear:   00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.408291] skb linear:   00000090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.415944] skb linear:   000000a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.423598] skb linear:   000000b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.431253] skb linear:   000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.438904] skb linear:   000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.446560] skb linear:   000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.454214] skb linear:   000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.461868] skb linear:   00000100: 00 00 00 00 00 00 00 00 63 82 53 63 35 01 02 36
[    7.469524] skb linear:   00000110: 04 c0 a8 01 fa 33 04 00 00 a8 c0 3a 04 00 00 54
[    7.477178] skb linear:   00000120: 60 3b 04 00 00 93 a8 01 04 ff ff ff 00 1c 04 c0
[    7.484830] skb linear:   00000130: a8 01 ff 03 04 c0 a8 01 fa 0f 03 6c 61 6e 0c 07
[    7.492483] skb linear:   00000140: 72 70 69 34 2d 32 67 2a 04 c0 a8 01 fa 06 04 c0
[    7.500137] skb linear:   00000150: a8 01 fa ff
[    7.504666] skb tailroom: 00000000: 16 f8 6b 61 00 00 00 00 00 00 00 00 00 00 00 00
[    7.512317] skb tailroom: 00000010: 00 00 00 00 02 00 00 00 00 00 00 00 2f 75 73 72
[    7.519971] skb tailroom: 00000020: 2f 6c 69 62 2f 6b 65 72 6e 65 6c 2d 6f 76 65 72
[    7.527635] skb tailroom: 00000030: 6c 61 79 73 2f 62 61 73 65 2f 6c 69 62 2f 66 69
[    7.535304] skb tailroom: 00000040: 72 6d 77 61 72 65 2f 62 34 33 2f 75 63 6f 64 65
[    7.542974] skb tailroom: 00000050: 32 31 5f 73 73 6c 70 6e 5f 6e 6f 62 74 2e 66 77
[    7.550633] skb tailroom: 00000060: 00 01 00 00 22 01 00 00 00 00 00 00 00 00 00 00
[    7.558287] skb tailroom: 00000070: 37 32 30 78 35 37 36 00 00 00 00 00 00 00 00 00
[    7.565940] skb tailroom: 00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.573594] skb tailroom: 00000090: 00 00 00 00 02 00 00 00 00 00 00 00 2f 75 73 72
[    7.581246] skb tailroom: 000000a0: 2f 6c 69 62 2f 6b 65 72 6e 65 6c 2d 6f 76 65 72
[    7.588898] skb tailroom: 000000b0: 6c 61 79 73 2f 62 61 73 65 2f 6c 69 62 2f 66 69
[    7.596551] skb tailroom: 000000c0: 72 6d 77 61 72 65 2f 62 34 33 2f 75 63 6f 64 65
[    7.604204] skb tailroom: 000000d0: 32 30 5f 73 73 6c 70 6e 5f 6e 6f 62 74 2e 66 77
[    7.611857] skb tailroom: 000000e0: 00 01 00 00 22 01 00 00 00 00 00 00 00 00 00 00
[    7.619511] skb tailroom: 000000f0: 31 34 34 30 78 35 37 36 00 00 00 00 00 00 00 00
[    7.627165] skb tailroom: 00000100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    7.634819] skb tailroom: 00000110: 00 00 00 00 02 00 00 00 00 00 00 00 2f 75 73 72
[    7.642473] skb tailroom: 00000120: 2f 6c 69 62 2f 6b 65 72 6e 65 6c 2d 6f 76 65 72
[    7.650127] skb tailroom: 00000130: 6c 61 79 73 2f 62 61 73 65 2f 6c 69 62 2f 66 69
[    7.657776] skb tailroom: 00000140: 72 6d 77 61 72 65 2f 62 34 33 2f 75 63 6f 64 65
[    7.665429] skb tailroom: 00000150: 31 39 5f 73 73 6c 70 6e 5f 6e 6f 62 74 2e 66 77
[    7.673084] skb tailroom: 00000160: 00 01 00 00 22 01 00 00 00 00 00 00 00 00 00 00
[    7.680739] skb tailroom: 00000170: 37 32 30 78 35 37 36 69 00 00 00 00 00 00 00 00
[    7.688390] skb tailroom: 00000180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
[    8.620794] [<c0259704>] (cpu_startup_entry) from [<c0c8a48c>] (rest_init+0xb4/0xbc)
[    8.628534] [<c0c8a3d8>] (rest_init) from [<c1200cc4>] (arch_call_rest_init+0x18/0x1c)
[    8.636442]  r5:c1250f40 r4:c18ea078
[    8.640013] [<c1200cac>] (arch_call_rest_init) from [<c1201520>] (start_kernel+0x7e0/0x820)
[    8.648358] [<c1200d40>] (start_kernel) from [<00000000>] (0x0)
[    8.654271]  r10:30c5387d r9:410fd083 r8:2eff3f00 r7:00000c42 r6:30c0387d r5:00000000
[    8.662092]  r4:c1200330
[    8.664716] eth0: hw csum failure
...
```

popcornmix, 2020-09-17:
Hmm, this is beyond my knowledge. 
drivers/net/ethernet/broadcom/genet/bcmgenet.c has changed significantly upstream for 5.9.
It may be that https://github.com/raspberrypi/linux/commit/cf55f6ceb2c0a5cce605fc171e33b0ee6640c23a is now skipping something essential.
@pelwell may know more.

HiassofT, 2020-09-17:
I checked the 5.6 kernel branch on RPiOS and that showed the same behaviour as 5.9 - intermittently slow network and splats with genet.skip_umac_reset=1

I didn't have much luck with 5.5, that wouldn't mount the nfsroot. kernel dhcp config succeeded but nfs mount failed and kernel panicked

```
[    8.121449] Sending DHCP requests ., OK
[    8.168350] IP-Config: Got DHCP answer from 192.168.1.250, my address is 192.168.1.68
[    8.179200] IP-Config: Complete:
[    8.185448]      device=eth0, hwaddr=dc:a6:32:0a:84:2f, ipaddr=192.168.1.68, mask=255.255.255.0, gw=192.168.1.250
[    8.198807]      host=rpi4-2g, domain=lan, nis-domain=(none)
[    8.207554]      bootserver=192.168.1.250, rootserver=192.168.1.20, rootpath=
[    8.207561]      nameserver0=192.168.1.250
[    8.225018]      ntpserver0=192.168.1.250
[    8.233833] uart-pl011 fe201000.serial: no DMA platform data
[   38.871478] vcc-sd: disabling
[  104.151553] VFS: Unable to mount root fs via NFS, trying floppy.
[  104.161141] VFS: Cannot open root device "nfs" or unknown-block(2,0): error -6
```

I also checked with 5.4 kernel (on LibreELEC) before and that survived about 20 reboots without any issues.

I did all these tests on the same setup (2GB RPi4, Sep 16 firmware, Netgear switch).

popcornmix, 2020-09-18:
Libreelec with 5.9 kernel and sdcard boot doing ``ping -c 1 192.168.4.1 && reboot`` in ``/storage/.config.autostart.sh`` seems stable for me - running for some dozens of reboots.

I suspect there's something pi4/switch specific, or bootrom netboot dependent that is causing this.

HiassofT, 2020-09-18:
I think I managed to narrow it down: TFTP boot from bootloader in combination with kernel 5.6 or newer seem to be the key points to trigger the issue.

The nfsroot mount issue with kernel 5.5 seems to have been caused by the rw option in nfsroot. No idea why I had it in there, with that removed raspios survived 20 reboots with TFTP/NFS without any issue.
```
[    0.000000] Linux version 5.5.19-v7l+ (hias@delle) (gcc version 8.3.0 (Debian 8.3.0-2)) #1 SMP Thu Sep 17 21:21:39 CEST 2020
```

raspios with kernel 5.6 and TFTP/NFS boot failed at the 3rd reboot and in a later test even at the very first boot
```
[    0.000000] Linux version 5.6.19-v7l+ (hias@delle) (gcc version 8.3.0 (Debian 8.3.0-2)) #3 SMP Fri Sep 18 12:43:45 CEST 2020
```

However, with the same 5.6 kernel and firmware on SD card and only root on NFS it survived 50 reboots.

I also checked kernel 5.9.0-rc4, that survived 20 SD/NFS reboots as well.

timg236, 2020-09-18:
If skipping genet.skip_umac_reset=1 makes things worse then it would seem that either the umac_reset more likely the PHY reset is not correct. 

A few months ago the bootloader was updated to reset Ethernet if network boot failed (timeout, not connected) which resolve this problem https://github.com/raspberrypi/rpi-eeprom/issues/144. Presumably, the genet/phy driver now has different init behaviour.
I think the correct fix is to reset UMAC and PHY on driver init

timg236, 2020-09-18:
The bootloader can't do this because start.elf can use TFTP to loader the kernel so either start.elf has to fully reset the PHY or the kernel has to do this.

My guess is that if the UMAC is reset without a PHY reset via MDIO then the UMAC gets upset. Given that the kernel used to reset the UMAC I think it should really reset UMAC + PHY to isolate itself from start.elf or u-boot, UEFI etc

pelwell, 2020-09-23:
I'm seeing a variety of alignment exceptions and skb BUGs netbooting with 5.9. After establishing that 5.4 is a safe starting point I'll work forwards and see where it starts to go wrong.

pelwell, 2020-09-24:
@HiassofT I've found two commits that I need to revert in order to be able to TFTP and NFS boot 5.9 reliably; those commits are ae895c49 and 9a9ba2a4. If reverting them solves your problem then we can look into what the underlying cause is.

JamesH65, 2020-09-24:
IIRC, we've had problems with offloading checksuming of small packets, we had a tweak in there somewhere to prevent it. I wonder if this bypasses the tweak.

@6by9  has a younger brain and can therefor remember more than me,

timg236, 2020-09-24:
@JamesH65 Are you thinking of Pi 3 issues which had a different network interface?

pelwell, 2020-09-24:
Are you perhaps thinking of the lan78xx and/or smsc95xx drivers? We had no end of checksum issues with them.

JamesH65, 2020-09-24:
Maybe. Told you I had an aging brain.

HiassofT, 2020-09-24:
@pelwell thanks for looking into this! I've kicked off a 5.9 build with these two commits reverted and will run my reboot loop test with that

HiassofT, 2020-09-24:
@pelwell reboot test with those two commits reverted looks good. I stopped it after 56 successful reboots and bootup times were quite consistently around 18 seconds from kernel startup to reboot service start.

Tested on RPi4 4GB B1 stepping, HP1810 switch, Sep 3 bootloader, Sep 17 firmware

pelwell, 2020-09-24:
Thanks, @HiassofT - I've pushed the reversions for now, pending a proper fix.

pelwell, 2020-09-25:
The crash seems to be caused by leftover data in the RBUF. This data will have come through the UNIMAC in its former life under control of the bootloader, but will have done so without the 64-byte headers (which the kernel now always enables). This in itself could have been problematic, but it also seems as though there might be some partial packets in there.

The bcmgenet driver _never_ flushes the RX queues - this is acknowledged by a repeated comment:
```
        /* Disable RX/TX DMA and flush TX queues */
```
but not explained. My theory is that this is to preserve a wake packet after a suspend, and for that reason the patch I've just pushed to rpi-5.9.y only resets RBUF on the initial open - not the resume. If this isn't the case then it's easy enough to make the reset unconditional.

popcornmix, 2020-09-25:
Could the flush be done by firmware when it stops using interface?

pelwell, 2020-09-25:
It probably could and should, but I don't think the driver should rely on it.

pelwell, 2020-09-28:
> Could the flush be done by firmware when it stops using interface?

There's a patch in the internal firmware repo to do this in the case that we are network booting.

HiassofT, 2020-09-29:
Thanks a lot to everyone involved in fixing this issue!

Latest 5.9 kernel and firmware look good here, reboot test survived 55+ iterations, so I think this can be closed now.

closed 2020-09-29 by HiassofT

hamishmb, 2021-02-22:
Possibly related to this?: https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1861936

Specific fixes applied were in https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1861936/comments/102.

I'm not sure if there's an open bug for this, but I couldn't find it.

juergh, 2021-02-22:
> Possibly related to this?: https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1861936
> 
> Specific fixes applied were in https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1861936/comments/102.
> 
> I'm not sure if there's an open bug for this, but I couldn't find it.

The fix for the issue you're referring to is [1] which will also be applied to all stable kernel if it hasn't been already. So it will naturally trickle down into the pi kernels.

[1] https://github.com/torvalds/linux/commit/b160c28548bc0a87cbd16d5af6d3edcfd70b8c9a

6by9, 2021-02-22:
It already has hit at least the 5.10 branch
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/net/ipv4/tcp_ipv4.c?h=linux-5.10.y
https://github.com/raspberrypi/linux/commits/rpi-5.10.y/net/ipv4
https://github.com/raspberrypi/linux/commit/981e1807748af57283a11d566b787aed6107dae9

hamishmb, 2021-02-22:
So we won't have to wait for kernel 5.10 to become the default in Raspberry Pi OS?

6by9, 2021-02-22:
Er, 5.10 is already the default for Raspberrry Pi OS.

hamishmb, 2021-02-22:
Guess I need to update my kernel then! Thanks :)

alien999999999, 2024-01-30:
So, on upstream 6.7 i have this issue somewhat frequently (i didn't in 6.1 upstream). I tftp + nfsroot boot my RPI4B (8GB) all the time; and it seems related to boot, either i have the issue or it don't

either pings of < 0.5ms or pings > 1000ms ; and it's so slow that it never gets to boot and the dhcp lease runs out.

i noticed the rw on NFSROOT option, i think i have that, so i'll try to do without, see if that does help.

alien999999999, 2024-03-24:
see https://github.com/raspberrypi/firmware/issues/1882