raspberrypi/linux #3857: modprobe: FATAL: Module lirc_rpi not found in directory /lib/modules/5.4.51-v7l+

Dolphinzzx, 2020-09-18:
**Describe the bug**
When i install lirc, then input command: sudo modprobe lirc_rpi
It will apprear following erro.
modprobe: FATAL: Module lirc_rpi not found in directory /lib/modules/5.4.51-v7l+
As shown below.
[image]

* Raspberry Pi 4B

* Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage5

* Aug 19 2020 17:39:27 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start_x)

* Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

pelwell, 2020-09-18:
lirc_rpi is no longer supported - use gpio_ir instead.

pelwell, 2021-07-19:
lirc is the user-space code, lirc_rpi was the old kernel module, gpio_ir is the kernel driver. There is a guide to using lirc with gpio_ir here: https://devkimchi.com/2020/08/12/turning-raspberry-pi-into-remote-controller/

closed 2021-07-19 by pelwell