raspberrypi/linux #3858: Cannot build kernel 5.8 (arm64) on RPI4

Pea13, 2020-09-18:
Hello,

I'm trying to build a 5.8 kernel on my RPI 4 but it fails with this error : 
`make: *** No rule to make target 'zImage'.  Stop.`

I have followed this documentation : https://www.raspberrypi.org/documentation/linux/kernel/building.md

`# apt install git bc bison flex libssl-dev make
# git clone --depth=1 --branch rpi-5.8.y https://github.com/raspberrypi/linux
# cd linux
# KERNEL=kernel7l
# make bcm2711_defconfig                                                                                                                                                                                                   
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
#
# No change to .config
#
#  make -j4 zImage modules dtbs
make: *** No rule to make target 'zImage'.  Stop. 
`

These tests are made on the latest version of RaspiOS (lite 64 bits).
Any ideas to solve this ?

Regards,

andrum99, 2020-09-18:
Change zImage to Image and it should work fine. Support for gzip compressed kernel images on Pi for arm64 only just got integrated, so you're probably running from a branch that doesn't have that option in it.

The documentation is not yet up-to-date on building for arm64, as you've discovered. See the kernel building page here with WIP updates: https://github.com/raspberrypi/documentation/blob/c666656d8782a509d7eae82ffe634cbd28b147b8/linux/kernel/building.md

Pea13, 2020-09-18:
Ho nice ! It works well.
Thank you.

closed 2020-09-18 by Pea13