raspberrypi/linux #3861: rpi4: changing default CMA reservation on aarch64 with kernels 5.8.x fails to initialize vc4-drm gpu

nkichukov, 2020-09-20:
**Describe the bug**
I have a 4GB RPI 4 that runs 64-bit kernel on GNU/Gentoo Linux. The CMA reservation/allocation seem to be set to 256MB by default. If I try to pass 'cma=<Value>M' to the kernel command line, the the video driver fails to allocate from CMA. This happens even if I define 'cma=256M' on the command line, which is the same as the default.

**To reproduce**
1.) Compile the latest 5.8.x raspberrypi kernel for 64-bit mode
2.) Boot the kernel with dtoverlay=vc4-fkms-v3d and append in the cmdline.txt file 'cma=320M'
3.) Observe the failure to start the vc4-drm gpu driver

**Expected behaviour**
vc4-drm gpu starts up properly and is able to allocate from the CMA previously reserved.

**Actual behaviour**
vc4-drm gpu fails to start. The failure is:
```
Sep 19 15:31:28 grpi4 kernel: [   14.203248] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
Sep 19 15:31:28 grpi4 kernel: [   14.203264] vc4-drm gpu: [drm]                           dumb:   4052kb BOs (1)
```

**System**

* Which model of Raspberry Pi: Pi4 4GB
* Which OS and version: GNU/Gentoo
* Which firmware version:
Jun  1 2020 13:23:40 
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)

* Which kernel version: 5.8.9-v8+ #1 SMP PREEMPT Thu Sep 17 19:56:08 CEST 2020 aarch64 GNU/Linux

**Logs**
```
Jan  1 01:00:10 grpi4 kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:32:03:AB vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=/dev/sda3 root_trim=yes rootfstype=f2fs console=tty1 rootwait genet.force_reneg=y genet.skip_umac_reset=n cma=512M
...
Jan  1 01:00:10 grpi4 kernel: [    0.000000] Memory: 2918652K/3604480K available (8126K kernel code, 970K rwdata, 3156K rodata, 2688K init, 1079K bss, 161540K reserved, 524288K cma-reserved)
...
Sep 19 15:31:28 grpi4 kernel: [   14.203248] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
Sep 19 15:31:28 grpi4 kernel: [   14.203264] vc4-drm gpu: [drm]                           dumb:   4052kb BOs (1)
```

**Additional context**
I've got HEVC HW decode through Prime on kodi, after compiling the popcornmix gbm branch with hevc hw acceleration working in 32bit LXC gentoo container. This works fine for HEVC encoded streams that are 720p, perhaps also 1080p. However, if I try to play 4K stream, then the 256M for CMA is not sufficient and I get kernel memory management failures:

```
Sep 19 15:49:57 grpi4 kernel: [  140.937727] rpivid feb10000.codec: rpivid_h265_start: (3840x2176)
Sep 19 15:49:57 grpi4 kernel: [  140.938317] cma: cma_alloc: alloc failed, req-size: 2048 pages, ret: -12
Sep 19 15:49:57 grpi4 kernel: [  140.938353] ------------[ cut here ]------------
Sep 19 15:49:57 grpi4 kernel: [  140.938379] WARNING: CPU: 2 PID: 4324 at mm/page_alloc.c:4831 __alloc_pages_nodemask+0x2e8/0x330
Sep 19 15:49:57 grpi4 kernel: [  140.938382] Modules linked in: veth xt_nat iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 ip_tables x_tables nfsd bridge stp llc binfmt_misc bfq brcmfmac sha256_generic libsha256 sha256_arm64 vc4 cfg80211 snd_soc_core v3d snd_pcm_dmaengine drm_kms_helper gpu_sched joydev evdev bcm2835_v4l2(C) bcm2835_isp(C) raspberrypi_hwmon hwmon bcm2835_codec(C) rfkill rpivid_hevc(C) brcmutil bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_mem2mem videobuf2_vmalloc snd_bcm2835(C) videobuf2_memops snd_pcm videobuf2_v4l2 vc_sm_cma(C) videobuf2_common drm snd_timer snd videodev mc drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio fixed
Sep 19 15:49:57 grpi4 kernel: [  140.938461] CPU: 2 PID: 4324 Comm: VideoPlayer Tainted: G         C        5.8.9-v8+ #1
Sep 19 15:49:57 grpi4 kernel: [  140.938464] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Sep 19 15:49:57 grpi4 kernel: [  140.938471] pstate: 20000005 (nzCv daif -PAN -UAO BTYPE=--)
Sep 19 15:49:57 grpi4 kernel: [  140.938475] pc : __alloc_pages_nodemask+0x2e8/0x330
Sep 19 15:49:57 grpi4 kernel: [  140.938484] lr : dma_direct_alloc_pages+0x29c/0x420
Sep 19 15:49:57 grpi4 kernel: [  140.938486] sp : ffffffc013673b00
Sep 19 15:49:57 grpi4 kernel: [  140.938488] x29: ffffffc013673b00 x28: 0000000000000050 
Sep 19 15:49:57 grpi4 kernel: [  140.938492] x27: 000000000000000a x26: 00000000ffffffff 
Sep 19 15:49:57 grpi4 kernel: [  140.938496] x25: ffffffc01015c534 x24: 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.938500] x23: ffffff80dd049290 x22: ffffffc010e38908 
Sep 19 15:49:57 grpi4 kernel: [  140.938503] x21: ffffffc010e38908 x20: ffffff80f6fd7c10 
Sep 19 15:49:57 grpi4 kernel: [  140.938506] x19: 0000000000000cc4 x18: 0000000000000001 
Sep 19 15:49:57 grpi4 kernel: [  140.938509] x17: 0000000000000000 x16: 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.938512] x15: ffffffffffffffff x14: ffffffc010e38908 
Sep 19 15:49:57 grpi4 kernel: [  140.938516] x13: 0000000000000000 x12: ffffffc010f2b000 
Sep 19 15:49:57 grpi4 kernel: [  140.938519] x11: ffffffc010e52000 x10: ffffffc010f2b5c8 
Sep 19 15:49:57 grpi4 kernel: [  140.938523] x9 : ffffffc01015c2d4 x8 : 0000000000000001 
Sep 19 15:49:57 grpi4 kernel: [  140.938526] x7 : 0000000000000175 x6 : ffffffc013673800 
Sep 19 15:49:57 grpi4 kernel: [  140.938529] x5 : 00000000000007ff x4 : 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.938532] x3 : 0000000000000000 x2 : 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.938535] x1 : 000000000000000b x0 : 0000000000000cc4 
Sep 19 15:49:57 grpi4 kernel: [  140.938539] Call trace:
[... 43 lines cut ...]
Sep 19 15:49:57 grpi4 kernel: [  140.939922] x17: 0000000000000000 x16: 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.939947] x15: ffffffffffffffff x14: 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.939973] x13: 0000000000000000 x12: ffffffffffffffff 
Sep 19 15:49:57 grpi4 kernel: [  140.940000] x11: 0000000000000040 x10: 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.940030] x9 : ffffffc008ad74a4 x8 : 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.940056] x7 : ffffffffffffffff x6 : 0000000000000000 
Sep 19 15:49:57 grpi4 kernel: [  140.940081] x5 : 0000000000000000 x4 : 0000000000000007 
Sep 19 15:49:57 grpi4 kernel: [  140.940108] x3 : 0000000000000000 x2 : ffffffc008ad9468 
Sep 19 15:49:57 grpi4 kernel: [  140.940134] x1 : ffffff80f5d12080 x0 : 0000000000000001 
Sep 19 15:49:57 grpi4 kernel: [  140.940160] Call trace:
Sep 19 15:49:57 grpi4 kernel: [  140.940179]  rpivid_h265_setup+0x60/0x14c0 [rpivid_hevc]
Sep 19 15:49:57 grpi4 kernel: [  140.940205]  rpivid_device_run+0xbc/0x228 [rpivid_hevc]
Sep 19 15:49:57 grpi4 kernel: [  140.940239]  v4l2_m2m_try_run+0x9c/0x108 [v4l2_mem2mem]
Sep 19 15:49:57 grpi4 kernel: [  140.940267]  v4l2_m2m_request_queue+0xe8/0x138 [v4l2_mem2mem]
Sep 19 15:49:57 grpi4 kernel: [  140.940302]  media_request_ioctl+0x1cc/0x210 [mc]
Sep 19 15:49:57 grpi4 kernel: [  140.940342]  __arm64_compat_sys_ioctl+0xa8/0x100
Sep 19 15:49:57 grpi4 kernel: [  140.940376]  el0_svc_common.constprop.2+0x98/0x1a0
Sep 19 15:49:57 grpi4 kernel: [  140.940402]  do_el0_svc_compat+0x20/0x48
Sep 19 15:49:57 grpi4 kernel: [  140.940427]  el0_sync_compat_handler+0x134/0x168
Sep 19 15:49:57 grpi4 kernel: [  140.940450]  el0_sync_compat+0x144/0x180
Sep 19 15:49:57 grpi4 kernel: [  140.940476] Code: b9420000 f94012b4 d2770000 53092400 (390572c0) 
Sep 19 15:49:57 grpi4 kernel: [  140.940506] ---[ end trace 4b0daed86467ea99 ]---
```

This has been confirmed to be happening with kernels: 5.8.2 and 5.8.9.

pelwell, 2020-09-21:
The command line parameter is working for me:
```
pi@raspberrypi:~$ dmesg | grep -i cma
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 320 MiB at 0x000000001ac00000
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 ...  cma=320M
[    0.000000] Memory: 3589732K/4050944K available (10240K kernel code, 921K rwdata, 3156K rodata, 2048K init, 870K bss, 133532K reserved, 327680K cma-reserved, 3264512K highmem)
```
Please run the same grep command with and without the cma command line parameter.

nkichukov, 2020-09-21:
Hi @pelwell 

Without passing any cma parameter, ie defaults:

```
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory: 3626388K/4050944K available (8126K kernel code, 970K rwdata, 3156K rodata, 2688K init, 1079K bss, 162412K reserved, 262144K cma-reserved)
[    6.788937] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    6.790151] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

Appending cma=320M on the command line:
```
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 320 MiB at 0x00000000e8000000
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:32:03:AB vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=/dev/sda3 root_trim=yes rootfstype=f2fs console=tty1 rootwait genet.force_reneg=y genet.skip_umac_reset=n cma=320M
[    0.000000] Memory: 3560852K/4050944K available (8126K kernel code, 970K rwdata, 3156K rodata, 2688K init, 1079K bss, 162412K reserved, 327680K cma-reserved)
[    6.941024] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    6.944468] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   15.855136] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
```

pelwell, 2020-09-21:
The kernel has created the DMA pool at an address which is too high - it has to fit within the first 1GB for our purposes.

Our Pi 4 device tree declares the CMA region like this:
```
	rmem: reserved-memory {
		#address-cells = <2>;
		#size-cells = <1>;
		ranges;
		cma: linux,cma {
			alloc-ranges = <0x0 0x00000000 0x30000000>;
			compatible = "shared-dma-pool";
			linux,cma-default;
			reusable;
			size = <0x4000000>;
		};
	};
```

I suspect that using the command line parameter bypasses that declaration.

Try using one of the cma size DT parameters instead:
```
pi@raspberrypi:~$ dtoverlay -h vc4-fkms-v3d
Name:   vc4-fkms-v3d

Info:   Enable Eric Anholt's DRM VC4 V3D driver on top of the dispmanx
        display stack.

Usage:  dtoverlay=vc4-fkms-v3d,<param>

Params: cma-256                 CMA is 256MB (needs 1GB)
        cma-192                 CMA is 192MB (needs 1GB)
        cma-128                 CMA is 128MB
        cma-96                  CMA is 96MB
        cma-64                  CMA is 64MB
        cma-size                CMA size in bytes, 4MB aligned
        cma-default             Use upstream's default value
```
320MB is not one of the preset values, so use the `cma-size` parameter instead:
```
dtoverlay=vc4-fkms-v3d,cma-size=0x14000000
```
which for me gives:
```
pi@raspberrypi:~$ dmesg | grep -i cma
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory: 3589732K/4050944K available (10240K kernel code, 921K rwdata, 3156K rodata, 2048K init, 870K bss, 133532K reserved, 327680K cma-reserved, 3264512K highmem)
```

vianpl, 2020-09-21:
If you **must** do the allocation through the kernel command line you should be able to do it through `cma=320M@0x0-0x40000000`. This makes sure CMA falls into the first GB of memory.

nkichukov, 2020-09-21:
Thanks both for the quick responses.
I do not have to use kernel command line to achieve it, so I can confirm the dtoverlay approach brought the desired result. Here is the output when I grep for '-i cma':

```
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory: 3560852K/4050944K available (8126K kernel code, 970K rwdata, 3156K rodata, 2688K init, 1079K bss, 162412K reserved, 327680K cma-reserved)
[    7.010542] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    7.011716] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```
As a final thought, is this something that I overlooked, do we need to 'fix' something in kernel or firmware so others avoid it or just improve the documentation to talk about this corner case?

pelwell, 2020-09-21:
In my view Device Tree is the preferred configuration mechanism, in which case the problem is solved already, otherwise some additional documentation is perhaps required somewhere. Fortunately you may have just provided some!

nkichukov, 2020-09-21:
Perhaps worth adding one more additional overlay cma size to the device tree file? In most cases, where people will need to decode 4K contents they would need something above the current highest 256M setting?

pelwell, 2020-09-21:
I could certainly add a few more options - 320M, 384M, 448M.

nkichukov, 2020-09-21:
That would be ideal!

pelwell, 2020-09-21:
The patch is in rpi-5.4.y now (I threw in 512M as well).

nkichukov, 2020-09-22:
Great, when can we expect to see it ported into rpi-5.8.y branch ?

I guess with this the issue can be closed as resolved.

Thanks again for the quick resolution, Phil.

pelwell, 2020-09-22:
It should be there (and rpi-5.9.y) now.

closed 2020-09-23 by nkichukov

theRealProHacker, 2024-02-01:
@joserpintuitivo you can increase the cma size by editing your `/boot/config.txt` (e.g. with `sudo nano /boot/config.txt`)
There you will likely find something like `dtoverlay=vc4-fkms-v3d` and you have to change it in one of the following ways:
- `dtoverlay=vc4-fkms-v3d,cma-size=0x14000000` (specify the size you need in bytes and hex format)
- `dtoverlay=vc4-fkms-v3d,cma-512` 
   (there are some defaults you can use, which you can see when you do `dtoverlay -h vc4-fkms-v3d`)