raspberrypi/linux #3864: 5.8.y: swap space doesn't work

Titaniumtown, 2020-09-21:
**Describe the bug**
When swapspace is created, then specified in /etc/fstab, on kernel 5.8, free -h shows swap as 0b, but with 5.4.y, it shows it as 16gb (as it should be).

**To reproduce**
Use kernel 5.8, create swap space and add it to fstab, observe swap being 0b when running free -h.

**Expected behaviour**
It should show 16gb of swap when running free -h 

**Actual behaviour**
free -h shows 0b of swap when running kernel 5.8

**System**
Model: Raspberry Pi 4 8gb
OS: Arch Linux Arm (64 bit tarball that I created)
version 6105bb6ab6db9d62a57ce2bb96c527a2cc46a717
kernel: 5.8.5.r936780.g1b7aa2df0963-1

**Additional context**
Swap works on kernel 5.4

nkichukov, 2020-09-21:
I am running 5.8.9 here and I have swap.
Swap was there also on 5.8.0 and 5.8.2 kernels.

Does your swap appear if you manually enabled it, ie:
sudo swapon <swapfile/swapdevice>

Titaniumtown, 2020-09-21:
I rolled back the kernel running on it to 5.4, but I'll test that later.