raspberrypi/linux #3865: 64-bit kernel header deb packages not available

HinTak, 2020-09-21:
Somebody mentioned they cannot get the accompanying kernel headers to build out-of-tree kernel modules against the 64-bit kernel in 2020-05-27 arm64 release. 

And indeed, only `kernel-headers-*-armhf` deb packages are in the archive, but `kernel-*arm64` are available for as recent as 2020-09-02 :

http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/

HinTak, 2020-09-21:
I just remember that the 64-bit kernel headers are, or were, in the armhf header package ( https://github.com/respeaker/seeed-voicecard/issues/191 ).

HinTak, 2020-10-28:
Argh, I see 20200902-2 and 20201022-1 headers are available in 64-bit form. If I get confirmation from a user I 'll close this.

XECDesign, 2020-10-28:
The current version is apt won't work, but once the gcc-8 built kernels make it into the 'stable' branch, everything should be okay.

Lots of discussion here - https://github.com/raspberrypi/Raspberry-Pi-OS-64bit

HinTak, 2020-10-28:
Somewhat overlapping issue - https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4 -

XECDesign, 2020-10-28:
That was the link I was meant to post, sorry.

HinTak, 2021-04-13:
gcc on current 32-bit raspbian reports `gcc (Raspbian 8.3.0-6+rpi1) 8.3.0` - I assume this can be closed (and the other one two?)

HinTak, 2021-04-13:
https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4 has closed a month ago; so this is probably obsolete.

closed 2021-04-13 by HinTak