raspberrypi/linux #3866: All ioctl commands for sd card go to timeout

davideferraretto, 2020-09-21:
WIth sdtools or mmc all cmd go to timeout.
For example if i try
`sudo ./mmc status get /dev/mmcblk0`
then i'll get
ioctl: Connection timed out
Could not read response to SEND_STATUS from /dev/mmcblk0

I have a Pi3B+ with debian 10 (buster)
kernel: Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
With dmesg I see
[  941.588296]  mmcblk0:
[ 1783.869652] sdhost-bcm2835 3f202000.mmc: __mmc_blk_ioctl_cmd: cmd error -110

Thanks.

pelwell, 2020-09-22:
From the comments on the issue on the sdtool maintainer's site (https://github.com/BertoldVdb/sdtool/issues/10), it isn't understood and also doesn't appear to be specific to Raspberry Pi.

juergh, 2021-03-05:
I believe these are different issues. The sdtool problem was introduced in 5.5 and fixed in 5.10 (BertoldVdb/sdtool#10). But the mmc tools keep fails for all kernels that I've tested 4.18 to 5.11.

juergh, 2021-03-05:
And the mmc tool fails fast, whereas sdtool spins for a couple of seconds before timing out.

Manouchehri, 2022-10-21:
`sdtool` works for me, example:
```
dave@sys:~$ sudo ./amd64-sdtool /dev/mmcblk0 status
[+] Found RCA for /dev/mmcblk0: XXXX.
[+] Card CSD: 4XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX.
[+] Write protection state: Off.
```

While `mmc` always fails:
```
dave@sys:~$ sudo mmc extcsd read  /dev/mmcblk0
ioctl: Connection timed out
Could not read EXT_CSD from /dev/mmcblk0
```

```
[1278502.805187] mmc0: cannot verify signal voltage switch
[1278503.123685] rtsx_pci_sdmmc rtsx_pci_sdmmc.0: __mmc_blk_ioctl_cmd: cmd error -110
[1278503.229154]  mmcblk0: p1 p2
```

```
dave@sys:~$ uname -a
Linux sys 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```