raspberrypi/linux #3871: mDNS domain name has changed from local to lan

mats-nk, 2020-09-28:
mDNS default configuration has changed so the Rasspberry Pi OS doesnt respond to the previous mDNS hostname `raspberrypi.local` but `raspberrypi.lan`

But when examining the processes running a `ps aux` shows a row with:

`avahi      297  0.0  0.6   5756  2920 ?        Ss   13:56   0:00 avahi-daemon: running [raspberrypi.local]`

So its a bit confusing!

ncguk, 2020-09-28:
By any chance have you recently installed Pi-Hole?

mats-nk, 2020-09-28:
A clean install of "2020-08-20-raspios-buster-armhf-lite".

Only installed updates and git

cmitu, 2020-09-29:
@mats-nk are you sure the `raspberrypi.lan` isn't set by your local router/DNS server ? Either way, this is not really a Linux kernel issue, you should use the Raspberry Pi forums or the Raspbian issue tracker.

closed 2020-09-29 by mats-nk