raspberrypi/linux #3873: Kernel build error when disabling "Enable loadable module support" with bcm2711_defconfig

julienrobin28, 2020-09-28:
Hi all

I'm working on some custom project which implies building very recent kernel versions (5.4 to 5.9), embedding required modules - without having the OS loading separated *.ko modules. So I'm using "Enable loadable module support" (which is "CONFIG_MODULES") option, setting it to disabled.

It is building and working on untouched kernel.org kernels, and also android kernel 5.4 from google, but not building using raspberrypi kernel (at least 5.4.y to 5.9.y included). This means the error has been introduced into raspberrypi/linux code somewhere
The error I get is :
```
aarch64-linux-gnu-ld: sound/soc/codecs/cs42xx8-i2c.o:(.rodata+0x0): multiple definition of `cs42xx8_of_match'; sound/soc/codecs/cs42xx8.o:(.rodata+0x658): first defined here
make: *** [Makefile:1162: vmlinux] Error 1
```

**To reproduce**
From an x86_64 computer :
```
git clone --depth=1 --branch rpi-5.9.y https://github.com/raspberrypi/linux
or even
git clone --depth=1 --branch rpi-5.4.y https://github.com/raspberrypi/linux

make -j $(nproc) ARCH=arm64 bcm2711_defconfig
make -j $(nproc) ARCH=arm64 menuconfig
(here, disable "Enable loadable module support", then save and exit)
make -j $(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-
```
**As a workaround - if no sound is needed**
As the problem is about sound stuff, I can use menuconfig, go to Device Drivers, disabling "Sound card support" (which is CONFIG_SOUND) and the kernel successfully builds.

Hoping this feedback can help you to correct some issue and improve this work!
Best regards

6by9, 2020-09-28:
It looks like we've gained a dodgy bit in the downstream patch
```
commit 0ea38b0fcd04e0cbf2e8b6b37e936445e8e17d56
Author: Matt Flax <flatmax@flatmax.org>
Date:   Wed Mar 8 20:04:13 2017 +1100

    Add support for the AudioInjector.net Octo sound card
    
    AudioInjector Octo: sample rates, regulators, reset
```
It shouldn't have
```
diff --git a/sound/soc/codecs/cs42xx8-i2c.c b/sound/soc/codecs/cs42xx8-i2c.c
index 0214e3a..a4586ea 100644
--- a/sound/soc/codecs/cs42xx8-i2c.c
+++ b/sound/soc/codecs/cs42xx8-i2c.c
@@ -45,6 +45,13 @@ static struct i2c_device_id cs42xx8_i2c_id[] = {
 };
 MODULE_DEVICE_TABLE(i2c, cs42xx8_i2c_id);
 
+const struct of_device_id cs42xx8_of_match[] = {
+       { .compatible = "cirrus,cs42448", .data = &cs42448_data, },
+       { .compatible = "cirrus,cs42888", .data = &cs42888_data, },
+       { /* sentinel */ }
+};
+MODULE_DEVICE_TABLE(of, cs42xx8_of_match);
+
 static struct i2c_driver cs42xx8_i2c_driver = {
        .driver = {
                .name = "cs42xx8",
```
as that is now in cs42xx8.c

5e4cb7b "ASoC: cs42xx8: Setup of_match_table" looks to have been the upstream patch adding the usage of cs42xx8_of_match to cs42xx8-i2c.c

julienrobin28, 2020-09-28:
Congrats, glad you figured out everything about this issue :+1: 
Needless to say that having your work for RPi4 support integrated into the real mainline kernel is really great too!

pelwell, 2020-09-29:
I've pushed a fix to the 5.4 tree - if it works for you I'll replicate it across the other branches.

The downstream patch you noticed is necessary to ensure that the correct module is loaded - without it, the kernel can load the core driver instead of the I2C shim, causing the I2C driver to never be instantiated.

julienrobin28, 2020-09-29:
Thanks,

Just tested running on RPi4 by building with "CONFIG_MODULES" option disabled : the big resulting kernel embedding every module works fine (with HDMI and audio jack). In order to get it booting I had to disable "CONFIG_GAMEPORT" and Generic sound devices (CONFIG_SND_DRIVERS) which is located in menuconfig at Device Drivers -> Sound card support -> Advanced Linux Sound Architecture -> Generic sound devices. But this is probably unrelated.

I'm going to test with standard build (with separated modules dynamically loaded by the OS) and keep you informed, just to be double sure that audio isn't broken for normal users ;) at least for RPi4

julienrobin28, 2020-09-29:
Double check completed, build and works fine with both CONFIG_MODULES enabled and disabled, using HDMI and audio jack (at least on Pi4 with using 64 bit kernel) :+1:

pelwell, 2020-09-29:
Thanks for the update - the fix has been replicated to rpi-5.8.y and rpi-5.9.y. Please close the issue if you're happy the matter has been resolved.

julienrobin28, 2020-09-29:
Perfect!
Successfully tried the rpi-5.9.y from few minutes ago, built and worked with or without CONFIG_MODULES and audio worked in both case.

During this attempt, I didn't get the USB ports working, but I guess this is something else that is being worked in rpi-5.9.y
Bye and thanks again for the work

closed 2020-09-29 by julienrobin28