raspberrypi/linux #3874: failed on  "make localmodconfig"

rpikernel, 2020-09-28:
make localmodconfig

failed:

 linux-rpi-5.4.y]$ make localmodconfig
using config: '.config'
Can't open ./arch/mips/loongson32/ls1c/Makefile at ./scripts/kconfig/streamline_config.pl line 301.
make[1]: *** [scripts/kconfig/Makefile:50: localmodconfig] Error 2
make: *** [Makefile:590: localmodconfig] Error 2

check the file:

ux-rpi-5.4.y]$ ls -l arch/mips/loongson32/ls1c/Makefile
lrwxrwxrwx 1 alarm alarm 104 Sep 27 15:40 arch/mips/loongson32/ls1c/Makefile -> '# SPDX-License-Identifier: GPL-2.0-only'$'\n''#'$'\n''# Makefile for loongson1C based machines.'$'\n''#'$'\n\n''obj-y += board.o'$'\n'

please help to solve the problem.

pelwell, 2020-09-28:
1. What platform are you compiling _on_?
2. What are you compiling _for_?
3. How did you download your source tree?

rpikernel, 2020-09-28:
1.~]$ uname -r
5.4.65-1-ARCH
generating config from above running os " zcat /proc/config.gz > .config"
2. armv7h raspberry pi 3 
3. git clone https://github.com/raspberrypi/linux.git

pelwell, 2020-09-29:
`uname` -r only says which kernel version you are using - `uname -m` or `uname -a` is more informative.

The immediate cause of the failure is that the Makefile has been converted to a symbolic link, where the content of the Makefile is used as if it was the name of the file pointed to by the link. It sounds like either the file system or the download is corrupted.

What does `git status` report?

Given how long it takes to ask you a question, your best option might be to do a fresh clone and start again. If you have enough space you can download it to a different directory in case you want to compare the two trees:
```
$ git clone https://github.com/raspberrypi/linux.git linux2
```

rpikernel, 2020-09-29:
kernel_built]$ uname -a
Linux alarmpi 5.4.65-1-ARCH #1 SMP PREEMPT Tue Sep 15 16:21:43 UTC 2020 armv7l GNU/Linux

i had downloaded zip file instead:
kernel_built]$ ls -l linux-rpi-5.4.y.zip
-rw-r--r-- 1 alarm alarm 214254678 Sep 27 14:42 linux-rpi-5.4.y.zip

this should avoid file corruption issue. Then, unzip the file,

kernel_built]$ unzip linux-rpi-5.4.y.zip

but, got the same error as before.

what was wrong?

pelwell, 2020-09-29:
I've heard of zip archives downloaded from GitHub being prone to corruption - I've no idea how - but not `git clone`.

4. Did you manage to run `git status` on the `git clone`d tree?
5. Are you extracting it into a regular ext4 filing system?

I'm downloading a tree here to repeat your test.

pelwell, 2020-09-29:
The clone and make localmodconfig worked as expected for me. I think there's something strange about your file system.

rpikernel, 2020-09-29:
trying clone again, but, the clone got stuck:
  kernel_built]$ git clone https://github.com/raspberrypi/linux.git
Cloning into 'linux'...
remote: Enumerating objects: 64, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (41/41), done.
Receiving objects:  76% (6439186/8377289), 2.15 GiB | 0 bytes/s

not be able to clone repo now.

I only be able to download zip file.

What have to do to get zip file unzip correctly?

rpikernel, 2020-09-29:
or working?

pelwell, 2020-09-29:
We advise people to clone.

You are the only person reporting this problem - it's something specific to your configuration, and this isn't a help forum.

closed 2020-09-29 by pelwell