raspberrypi/linux #3876: HDMI surround not working

cclecle, 2020-09-28:
using fresh 2020-08-20-raspios-buster-armhf-full image I can not get surround/5.1 audio to work through HDMI.
so, no pulse audio here, just alsa.

At the begining only stereo was available.
Then I put dtoverlay=vc4-fkms-v3d in config.txt, later I add  snd-bcm2835.enable_compat_alsa option in cmdline.txt.
It allow me to see a lot more audio devices.
I was able to play surround files using hw:0,1 device (wich is hardware HDMI-0) !
BUT, this device seems to be "not managed", my HT reports 7.1 input signal , even if speaker-test is set to "-c 6"... As a result, movies sound is distorded and somtimes channels are swapped.
Same with vlc / kodi: I can play 7.1 movies just fine but 5.1 have distorted sound (and i suspect subwoofer isnt working to, both in 5.1 / 7.1)

As a comparison I downloaded the last Librelec build and everything work just fine: audio output switch from 2.0 / 5.1 / 7.1 (and corresponding bit rate) juste fine.

felagund, 2021-04-05:
Now Pulse Audio is the default, is it still not working (for me, it stopped working with the move to Pulseaudio).

cclecle, 2021-04-07:
Hi Thomas, I'll try again, actually I am now using Dual boot with
OpenElec/Kodi for HT usage (5.1 ++ work with no problem) and 2nd boot
option with Ubuntu (no 5.1 at the time I installed it).
After a lot of tweak I succeeded playing 5.1 samples on Ubuntu but it was
with hard config on pulse audio / alsa and if I remember well it did not
correctly detect source configuration so output was not good if it was not
1:1 in par with source config (num of channel, sample rate etc). So I
decided to go for dual boot route..