raspberrypi/linux #3878: simple-framebuffer: probe of 0.framebuffer failed with error -12

kraj, 2020-10-01:
I am using yocto to boot rpi4/64bit with kernel 5.4 and I see this message in dmesg

[    0.000000] Linux version 5.4.64-v8 (oe-user@oe-host) (gcc version 10.2.0 (GCC)) #1 SMP PREEMPT Thu Oct 1 04:21:38 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1

```
[    0.849810] simple-framebuffer: probe of 0.framebuffer failed with error -12
 ```

Is this message normal or what am I missing?

pelwell, 2020-10-01:
It's not normal - you have some kind of configuration problem; -12 is -ENOMEM.

What do the following commands report?
```
$ cat /proc/device-tree/aliases/display0
$ dtc -I fs /proc/device-tree 2>/dev/null | grep -A10 framebuffer
```

kraj, 2020-10-01:
```
cat /proc/device-tree/aliases/display0
/chosen/framebuffer@00000000

```

and dtc dump is

```
root@raspberrypi4-64:~# dtc -I fs /proc/device-tree 2>/dev/null | grep -A10 framebuffer
                display0 = "/chosen/framebuffer@00000000";
                uart1 = "/soc/serial@7e215040";
                i2c5 = "/soc/i2c@7e205a00";
                serial0 = "/soc/serial@7e215040";
                watchdog = "/soc/watchdog@7e100000";
                i2c = "/soc/i2c@7e804000";
                i2c10 = "/soc/i2c0mux/i2c@1";
        };

        framebuffer {
                compatible = "simple-framebuffer";
                status = "disabled";
        };

        chosen {
                #address-cells = <0x01>;
                bootargs = "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:01:12:B5 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait";
                #size-cells = <0x01>;
                kaslr-seed = <0x24dcf6fe 0x49d1fb7a>;
                compatible = "simple_bus";
                ranges;
--
                framebuffer@00000000 {
                        format = "r5g6b5";
                        height = <0x00>;
                        width = <0x00>;
                        stride = <0x00>;
                        compatible = "simple-framebuffer";
                        status = "okay";
                        reg = <0x00 0x00>;
                };
        };

        phy {
                #phy-cells = <0x00>;
                compatible = "usb-nop-xceiv";
                phandle = <0x19>;
        };
```

pelwell, 2020-10-02:
I was wrong - that is normal/common behaviour if you're running headless and using either vc4-kms-v3d or vc4-fkms-v3d. The firmware doesn't create a framebuffer, and its zero dimensions are used to populate the DT framebuffer entry.

A better approach would be for the firmware to detect the absence of the framebuffer and not enable the simplefb driver.

pelwell, 2020-10-02:
> A better approach would be for the firmware to detect the absence of the framebuffer and not enable the simplefb driver.

There's now a patch in our firmware repo that does just that. It will be in future firmware releases.

kraj, 2020-10-02:
thanks @pelwell I will report back once I use the new firmware.

popcornmix, 2020-10-02:
rpi-update will get the new firmware.