raspberrypi/linux #3881: Driver for rtc-sd3078 not included

rrooggiieerr, 2020-10-03:
The driver for the RTC SD3078 is not include in the kernel for Raspberry Pi.

On Debian the kernel can be found in the linux-modules-extra package

The SD3078 is used on the PiSugar2 battery pack/HAT (not really a HAT, but ok)

pelwell, 2020-10-05:
@rrooggiieerr (you should get that keyboard looked at) - it wasn't easy finding a datasheet, and my Chinese is non-existent, but the rpi-5.4.y tree should now have the relevant driver enabled and a supporting patch to the `i2c-rtc` overlay. It all compiles and the overlay loads the correct module, but beyond that I can't test it. Can you confirm that the I2C address is 0x32?

rrooggiieerr, 2020-10-05:
Great, thanks!

Yes, It's at 0x32
```i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- 75 -- --```

How should I be able to test the new kernel, is there a testing repository that I can add to my sources.list?

pelwell, 2020-10-05:
Not for a kernel this new, no. It's not even in rpi-update yet. However, building your own kernel is actually quite easy if you follow our guide: https://www.raspberrypi.org/documentation/linux/kernel/building.md

popcornmix, 2020-10-07:
This is in latest rpi-update kernel

rrooggiieerr, 2020-10-07:
I just ran `rpi-update` and configured the RTC. Seems to work!

Thanks for the quick followup!

closed 2020-10-08 by pelwell