raspberrypi/linux #3883: CONFIG_SND_PCM_OSS - Re-enable Raspberry Pi kernel configuration option that breaks workarounds for known other kernel issue

dranch, 2020-10-05:
I'm trying to see how a somewhat recent Rpi kernel configuration change could be REVERSED.  Specifically, I'd like to see the following option be re-enabled in the stock Raspberry Pi OS kernel:

   CONFIG_SND_PCM_OSS=m

   Device Drivers --> Sound card support --> Advanced Linux Sound Architecture --> OSS PCM (digital audio) API

It's difficult to determine when this OSS /dev/dsp* sound system was disabled but I just confirmed across my personal Rpis:

   Raspbian Buster (10.4) does NOT include OSS support any of the 5.4.x kernel series
   Raspbian Stretch (9.3) has OSS support in the 4.9.x kernels but *not* in the 4.14.x or current 4.19.x kernels
   Raspbian Jessie (8.0) has OSS support in the 4.4.x and 4.9.x kernels but *not* in the current 4.14.x kernels

The reason to get this re-enabled is a known but unresolved ALSA bug with USB based sound devices detailed here including when seems like a real Raspberry Pi (0 thru 3B+) USB hardware issue not seen on the Rpi4 or x86 systems:

   https://github.com/wb2osz/direwolf/issues/194

Would it be possible to get OSS support re-enabled in the next Raspberry Pi kernel updates for all of the distro versions?

--David

P33M, 2020-10-05:
You say this is an unresolved ALSA bug on Raspberry Pi - has this issue been reported to us previously anywhere on our repositories? It seems like fixing that would be the sensible thing to do.

wb2osz, 2020-10-05:
It was reported here:  https://bugs.launchpad.net/bugs/1819560

pelwell, 2020-10-05:
It's not like us to remove kernel configuration options intentionally, but upstream changes and the defconfig system can sometimes catch us out. In this case I think the culprit is https://github.com/raspberrypi/linux/commit/eb3b705aaed9cab7efef8c905eda919b5fd902b9, which introduced a new dependency that we aren't setting. The way `make savedefconfig` works means that our attempt to enable SND_PCM_OSS just got forgotten.

I found another reference to SND_PCM_OSS and ALSA here: https://www.raspberrypi.org/forums/viewtopic.php?t=207875

All the complaints appear to be after Feb 9 2018, which is when we switched to 4.14 in our rpi-update kernels, which makes me think that the "bug" is just the loss of this option, and the inferiority of the workaround.

A patch to the defconfig to restore SND_PCM_OSS=m is in rpi-5.4.y now: https://github.com/raspberrypi/linux/commit/dcfe54515fb29a5a18efdf7dba65c08de4c97a2a. It will take a while for it to reach a kernel build you can download.

wb2osz, 2020-10-05:
I was not explicit enough in my hasty comment.
There is a timing bug in ALSA when using USB audio out.
This is disastrous with certain applications.
They turn on a transmitter, send some sound (from a software modem), then turn off the transmitter.
There is sometimes a substantial unpredictable delay before the sound starts.
The beginning of the transmission is sometimes a variable amount of silence.
The transmitter turns off before the sound is finished.

It was reported here: https://bugs.launchpad.net/bugs/1819560

The interest in using OSS is because it is a work around to ALSA which does not work correctly.
It is only a problem with earlier models.  
It does not happen with the RPi 4, probably because the USB hardware is much different.

dranch, 2020-10-05:
> A patch to the defconfig to restore SND_PCM_OSS=m is in rpi-5.4.y now: [dcfe545](https://github.com/raspberrypi/linux/commit/dcfe54515fb29a5a18efdf7dba65c08de4c97a2a). It will take a while for it to reach a kernel build you can download.

Thank you for the fix for re-enabling this work around.  Any guesstimates of when user might get this by doing a normal "apt" update?  Can we also get this fix backported to the other currently maintained Raspberry Pi kernels in Stretch and possibly Jessie (ELTS)?

Ps.  John is correct that this OSS option is only a workaround which also requires users to recompile the Direwolf application to remove ALSA support which is a big barrier to many users.  The right fix here would to try to resolve the ?USB? or ?ALSA? or ?ARM? issue that's described in https://bugs.launchpad.net/bugs/1819560.  The best we can tell, this is a Raspberry Pi specific bug impacting the Rpi 0, 0w, 1, 2, 3, 3+ specific boards.  It's NOT observed on say I2S-based sound devices on any Rpi model, not seen on the Rpi4 using the same USB sound cards, nor Raspberry Pi OS for x86.  

--David

dranch, 2020-10-05:
Curious for this root and larger issue mentioned at the bottom of my previous post, would it be more effective to create a new Github issue for that?  Maybe John's original https://bugs.launchpad.net/bugs/1819560 post is enough (even though it hasn't gone anywhere in 19 months since he posted it)?   John also did mention that other applications are impacted by this issue but other than Direwolf, I don't have a list of impacted applications.

pelwell, 2020-10-05:
Have people tried with the DWC2 driver (`dtoverlay=dwc2`) instead?

dranch, 2020-10-05:
I can't say I've tried this DWC2 option but all the web references that I can find to the Rpi with DWC2 date back to 2017 and 2018.  Is the DWC2 the officially recommended USB stack these days for Buster, Stretch, and maybe even Jessie for all hardware versions?  It doesn't seem to be as far as I can tell by checking various OS versions and Rpi hardware versions:

   sudo modprobe configs; zcat /proc/config.gz | grep -i dwc; echo "--------"; lsmod | grep -i dwc

It doesn't show up on my Rpi0, Rpi1, Rpi3 but it DOES show up on my Rpi4 (which is where Direwolf + USB audio works ok).

These Direwolf particular users are going to want to use the most stable USB+Sound solution possible.  What I don't think we want to do trade off fixes for new problems unless this DWC2 option is very stable.  If it is, I'm happy to give it a try.

pelwell, 2020-10-05:
If it's not a module then it's a built-in - is that official enough for you?

dranch, 2020-10-05:
The way I read it, only my Raspberry Pi 4 has the DWC2 kernel module loaded.  All other Raspberry Pis (0, 1, 3, etc) don't have the DWC2 kernel module installed therefore they are are only using the original USB stack.

--David

$ uname -a
Linux rpi3 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

$ sudo modprobe configs; zcat /proc/config.gz | grep -i dwc; echo "--------"; lsmod | grep -i dwc
# CONFIG_DWC_XLGMAC is not set
CONFIG_USB_DWCOTG=y
# CONFIG_USB_DWC3 is not set
CONFIG_USB_DWC2=m
# CONFIG_USB_DWC2_HOST is not set
# CONFIG_USB_DWC2_PERIPHERAL is not set
CONFIG_USB_DWC2_DUAL_ROLE=y
# CONFIG_USB_DWC2_DEBUG is not set
# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
# CONFIG_MMC_SDHCI_OF_DWCMSHC is not set
--------
$  
(DWC2 module not shown therefore not loaded)

pelwell, 2020-10-05:
> It doesn't show up on my Rpi0, Rpi1, Rpi3

> Linux rpi3 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
> CONFIG_USB_DWC2=m

Not all of the above statements can be true.

P33M, 2020-10-07:
I've compiled your testcase and can reproduce on a 3B+. From the analyzer, it looks like alsa isn't submitting *any* URBs when you're underrunning (i.e. in the dwell time between 250ms playbacks). In periods where it is submitting URBs, there's approx. 16ms of silence played before your sinewave starts, which I guess is some sort of internal buffer prepended to your data. This period is consistent. The gaps between the last playback finishing and the next starting are variable.

P33M, 2020-10-07:
There is a bug in dwc_otg. 

As you're completely draining the PCM output buffer, ALSA is subsequently not submitting any transfers to the audio OUT endpoint in the underrun condition.  A queue head (which describes an endpoint, and has a linked list of transfers for that endpoint) is removed from all periodic schedules if there are no transfers remaining for the EP. 

The queue head stores the expected next execution time point in units of USB microframes, but this value wasn't updated after the qh is added back into a queue. The stale value causes delays of up to 2.048 seconds, which is the period at which the frame counter wraps.

Running your test case with a tentative fix, it now passes reliably for all interval spacings. I have seen one odd delay of 350ms at the very start of day but cannot subsequently reproduce this, even when unplugging/replugging the audio adapter.

popcornmix, 2020-10-07:
SND_PCM_OSS is in latest rpi-update kernel.
But there will be subsequent update with @P33M's fix in later if you want to wait for that.

popcornmix, 2020-10-07:
@P33M's fix is in now in latest rpi-update kernel.

dranch, 2020-10-07:
I appreciate everyone looking at this and it seems we will potentially have both a proper ALSA fix and re-gain the OSS workaround.  While we can try the test via the bleeding edge kernels via rpi-update, any guesstimates of how long it will take until this fix is available in the mainstream kernels?  Also, any updates on backporting these fixes to Stretch and Jessie kernels?

popcornmix, 2020-10-07:
It will never be backported to Stretch or Jessie. We don't maintain old kernel trees.
Can't give a firm date for it reaching apt. Depends on a lot of activities going on. A month or two could be a best guess.

dranch, 2020-10-07:
Hmmm.. I know that at least Stretch is still being maintained but maybe only for security related issues?  Anyway.. maybe the DWC2 stack can be tried on say Stretch to see if that helps the issue there.  If you're doubtful on this point, please say so as we don't want to waste our time on something no one expects will yield any improvements.  The comments from P33M in this issue might give WB2OSZ (developer of Direwolf) an important clue if improvements can be made in Direwolf as well.

popcornmix, 2020-10-07:
Yes, jessie/stretch still get security updates from debian, but not pi specific updates (like firmware/kernel).
rpi-update on stretch should work to confirm if issue is improved.
Make a backup first is the image contains useful data just in case.

wb2osz, 2020-10-07:
Thank you for fixing this!!!

dranch, 2020-10-07:
> Yes, jessie/stretch still get security updates from debian, but not pi specific updates (like firmware/kernel).
> rpi-update on stretch should work to confirm if issue is improved.
> Make a backup first is the image contains useful data just in case.

In trying to understand your answer, I assume this DWG_OTG bug fix will be upstreamed to the main Linus kernels right?  If correct, doesn't that mean that Stretch and even Jessie might get the fixes at some later date?

popcornmix, 2020-10-07:
No. dwc_otg is a downstream module - upstream won't accept use of the FIQ in a driver.
And raspberry pi kernels are not part of debian.

So, no, jessie and stretch do not get any kernel updates.

pelwell, 2020-10-07:
dwc2 is the upstream equivalent.

dranch, 2020-10-07:
I appreciate the education here.  So it seems that this DWG_OTG module will only get fixes on the current kernel and not previous kernel families possibly around not enough resources to support backporting efforts.

In a related follow up question, is this "dwc_otg: initialise sched_frame for periodic QHs that were parked #3887" issue also in the DWG2 stack?  If not, maybe recommending the use of the DWG2 module on older OSes with older kernels might be worth a try.

pelwell, 2020-10-07:
Hence:
> Have people tried with the DWC2 driver (dtoverlay=dwc2) instead?

P33M, 2020-10-07:
> Hence:
> 
> > Have people tried with the DWC2 driver (dtoverlay=dwc2) instead?

To save our users the inconvenience of typing a line into config.txt, I did so
TL;DR: dwc2 does not suffer from this issue, as it has a completely different scheduling algorithm to the one used in dwc_otg.

pelwell, 2020-10-07:
Thank you, @P33M - you really went above and beyond on this one.

MichaIng, 2020-10-10:
As it has been mentioned that the RPi Zero suffers from the `usb_dwc` but as well, does it mean that the overlay documentation is wrong with the statement that `dwc2` is the default on RPi Zero? https://github.com/raspberrypi/firmware/blob/55c2d57e3108fc976647619943e4db2964e90e3c/boot/overlays/README#L715-L719

Probably it's hard to say as the two drivers differ too much, but from what I understand is that FIQ is the main difference between them, which is sort of an IRQ priority override, right? Is this actively used by any parts of the RPi firmware or utilities? Else could make sense to get some consistency in here going with upstream driver and in care of appearance solve bugs upstream as well instead of maintaining an own downstream driver?
At least OTG features are supported by both drivers from what I found, right? Just found Ethernet-over-USB for RPi Zero as example which is explicitly configured with the `dtoverlay=dwc2,g_ether`.

Sorry for going a bid off topic ;).

P33M, 2021-03-15:
This is fixed.

closed 2021-03-15 by P33M