raspberrypi/linux #3885: 64 bit Kernel compile issue

Dark-Sky, 2020-10-06:
I get a compile error trying to compile 64 bit for the pi3.

Compiler: aarch64-linux-gnu-gcc (GCC) 10.2.0 cross compiler
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcmrpi3_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-

```
  CC      drivers/usb/dwc2/core_intr.o
  LD [M]  drivers/staging/wlan-ng/prism2_usb.o
  CC      drivers/video/fbdev/core/bitblit.o
drivers/usb/dwc2/core_intr.c: In function 'dwc2_handle_usb_suspend_intr':
drivers/usb/dwc2/core_intr.c:545:31: error: 'struct dwc2_hsotg' has no member named 'gadget'
  545 |    usb_gadget_set_state(&hsotg->gadget, USB_STATE_NOTATTACHED);
      |                               ^~
make[3]: *** [scripts/Makefile.build:266: drivers/usb/dwc2/core_intr.o] Error 1
make[2]: *** [scripts/Makefile.build:500: drivers/usb/dwc2] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers/usb] Error 2
make[1]: *** Waiting for unfinished jobs....
```
I have compiled 32bit kernels for the pi1-pi4 and they did ok. I have not gotten to the 64bit pi4 kernel yet so I do not know how it will go.

pelwell, 2020-10-06:
Unfortunately a recently merged patch (#3151) breaks the gadget abstraction by adding references to gadget-specific state into the mainline code. I tried replacing the original `call_gadget(hsotg, suspend);` with `call_gadget(hsotg, disconnect);` (which also sets lx_state to DWC2_L3 and sets the gadget state to USB_STATE_NOTATTACHED), but that didn't recover when the device was reconnected.

Any ideas, @sn00pster and @mdevaev?

pelwell, 2020-10-06:
I've reverted c3974850cc in the meantime.

Dark-Sky, 2020-10-06:
Thanks @pelwell That fixed my issue here.

I will leave it up yo you to close this issue as it looks like other related things are going on.

mdevaev, 2020-10-06:
@pelwell Thank you for letting me know about the reverse. I've skimmed the code and **maybe** (I haven't tested it yet, I'll do it in the next few hours) that the correct way is to use `dwc2_hsotg_disconnect()`:
https://github.com/raspberrypi/linux/blob/1c64f4bc22811d2d371b271daa3fb27895a8abdd/drivers/usb/dwc2/gadget.c#L3258

As far as I can see, this function sets `DWC2_L3`, switches the state to `USB_STATE_NOTATTACHED`, but also additionally calls `call_gadget()` and kills requests to the device. This is probably how it should work. But I'm not a USB expert. 

@fizzyade (@sn00pster), why did you remove `call_gadget()` in your patch #3151, and not add it by changing `DWC2_L3` and `USB_STATE_NOTATTACHED`? Also why not `dwc2_hsotg_disconnect()`?

mdevaev, 2020-10-07:
Continued [here](https://github.com/raspberrypi/linux/pull/3151#issuecomment-704639779) to avoid offtop

hvenev, 2020-10-10:
Can you also revert this in 5.8?

pelwell, 2020-10-10:
That's done, but once 5.9 goes final I wouldn't expect rpi-5.8.y to be maintained much longer.

closed 2024-12-12 by Dark-Sky