raspberrypi/linux #3886: Enable Memory CGroup swap controller

tstevens, 2020-10-06:
In PR #3644 the resolution 472be8a85 included support for memory cgroups but did not include support for the swap controller.  I'm looking to see if that support can be added in as well.

When enabling the memory cgroup it is typical to see this snippet to add to your cmdline.txt 'cgroup_enable=memory swapaccount=1'. The current kernel does not support the swapaccount=1 portion so `docker info` for example will list `WARNING: No swap limit support`. My request would be to enable CONFIG_MEMCG_SWAP and unselect CONFIG_MEMCG_SWAP_ENABLED so the swap controller is only active when explicitly requested by 'swapaccount=1' and also only when the memory cgroup is enabled.

pelwell, 2020-10-07:
That seems very reasonable - the code involved is tiny. See 21155ec.

popcornmix, 2020-10-07:
This is in latest rpi-update kernel

tstevens, 2020-10-09:
Pulled and works as expected, thanks!

closed 2020-10-09 by tstevens