raspberrypi/linux #3893: Camera randomly disrupts i2c on other busses (RPI 3B+ & 4)

N4321D, 2020-10-08:
Recording from the camera (especially when using multiple splitter ports) 
and sampling the I2C bus (1 or others on RPI4) simultaneously randomly throws:

**errno 5** input output error or **errno 110** connection timed out.

this happens on the RPI4 every minute or so, and on the RPI3B+ less frequently (sometimes once 1h or something).

See this thread for all the specific test condition etc:
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=287015&p=1738555#p1734925

I tested in on both Buster and Stretch with different firmware versions and kernels.

Changing anything in config.txt (e.g. core_freq, arm_freq, or gpu mem etc) did not solve the issue. 

ON RPI3B+:

**raspinfo:**

```
pi@raspberrypi:~/App $ raspinfo
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2

Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
Revision	: a020d3
Serial		: 0000000070348527
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x50000
Camera          : supported=1 detected=1

Videocore information
---------------------

Sep  2 2020 21:16:35 
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30485636 2987016  26228604  11% /
devtmpfs          349404       0    349404   0% /dev
tmpfs             382684       0    382684   0% /dev/shm
tmpfs             382684    5232    377452   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             382684       0    382684   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54604    203492  22% /boot
tmpfs              76536       0     76536   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
[... 1681 lines cut ...]
#arm_freq=1500
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on,i2c_arm_baudrate=400000
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
start_x=1
gpu_mem=192
```

pelwell, 2020-10-08:
It's going to be hard to get much interest here when you have a log full of undervoltage warnings - there's a good chance that the reason using the camera affects I2C is that using it draws more power.

N4321D, 2020-10-08:
Thank you, 
You are right (although there was no undervoltage on RPI4. 
However, I ran it again on RPI3B+ with a better power supply (no undervolt warnings) and updated the output in the first post.
Unfortunately there was still the same issue.

> It's going to be hard to get much interest here when you have a log full of undervoltage warnings - there's a good chance that the reason using the camera affects I2C is that using it draws more power.

chukhanhhoang, 2021-07-08:
Hello! I have the same issue, but with multiple I2C devices on multiple channels. With the default channel 1, I enabled 3 additional I2C channels to read from and write to 4 devices simultaneously. After just a few seconds 3 channels stop working (while the (randomly) remaining one still runs) and timed-out errors appear in dmesg
```
[ 8661.510950] i2c-bcm2835 fe205800.i2c: i2c transfer timed out
[ 8661.511088] i2c-bcm2835 fe205c00.i2c: i2c transfer timed out
[ 8661.600935] i2c-bcm2835 fe804000.i2c: i2c transfer timed out
```

6by9, 2021-07-08:
@chukhanhhoang Are you running the camera at the same time, or just the different I2C buses from the Pi4?

All the I2C controllers share a common interrupt line (GIC 117 for Pi4, interrupt 21 on earlier boards), so there is a bit of cooperation required when they come to service them.

The firmware (which generally controls the camera and /dev/i2c-0) doesn't use the interrupts, so shouldn't cause any effects.

chukhanhhoang, 2021-07-08:
Thank you very much for your quick response! 
No, I am not using the camera now, just I2C devices as you said. Fortunately, my problem is solved in the latest kernel (Linux raspberrypi 5.10.46-v7l+). Now I am can read the channels (1,3,4,6) simultaneously.
Have a nice day!

6by9, 2021-07-08:
Curious as I'm not aware of any I2C changes between 5.10.17 and 5.10.46.

N4321D, 2021-07-12:
> Hello! I have the same issue, but with multiple I2C devices on multiple channels. In addition to the default channel 1, I enabled 3 additional I2C channels to read from and write to 4 devices simultaneously. After just a few seconds 3 channels stop working (while the (randomly) remaining one still runs) and timed-out errors appear in dmesg
> 
> ```
> [ 8661.510950] i2c-bcm2835 fe205800.i2c: i2c transfer timed out
> [ 8661.511088] i2c-bcm2835 fe205c00.i2c: i2c transfer timed out
> [ 8661.600935] i2c-bcm2835 fe804000.i2c: i2c transfer timed out
> ```
> 
> My setup is Raspberry Pi 4 running Raspberry Pi OS Buster (`Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux`).

Yes Linux 5.10.17-v7+  also created the same issues for me even without camera (on RPI3 B+). i2c timed out every second or so. rpi-update to 5.10.48-v7+ fixed it

6by9, 2025-01-29:
Buster and the 5.10 kernel are no longer supported. Please test with a more recent kernel.

6by9, 2025-01-29:
Closing as previous reporters were reporting it solved.

closed 2025-01-29 by 6by9