raspberrypi/linux #3896: RPi4 is low on DMA channels, channels 9 and 10 missing (?)

HiassofT, 2020-10-11:
As mentioned here https://github.com/raspberrypi/linux/pull/3895#issuecomment-706623546 RPi4 is quite low on available DMA channels. With vc4-kms active we are short by 1 DMA to also activate SPI and (duplex) I2S - each of these needs 2 DMAs but we only have 6 available DMAs and one of these is already used by mmc.

While checking DT config I noticed the downstream DT removes channels 9 and 10 compared to the upstream config - but the comment mentions channel 11 (which is in dma40)

see https://github.com/raspberrypi/linux/blob/rpi-5.9.y/arch/arm/boot/dts/bcm2711-rpi.dtsi#L100-L103
```
&dma {
	/* The VPU firmware uses DMA channel 11 for VCHIQ */
	brcm,dma-channel-mask = <0x1f5>;
};
```
vs https://github.com/raspberrypi/linux/blob/rpi-5.9.y/arch/arm/boot/dts/bcm2711.dtsi#L106
```
			brcm,dma-channel-mask = <0x07f5>;
```

Was there a particular reason for removing those 2 channels, is it possible to reclaim those 2 channels (under some circumstances), or am I missing something?

ping @pelwell

HiassofT, 2020-10-11:
forgot to post dmesgs to illustrate the issue (with bcm2835-dma debugging turned on).

bad luck: SPI got 2 DMAs, I2S then only got one and soundcard could not be instantiated http://ix.io/2Ape

better luck: I2S probed first and got 2 DMAs, SPI got only one and fell back to non-DMA mode, soundcard instantiated OK http://ix.io/2Apg

pelwell, 2020-10-12:
DMA channels do seem to be in even shorter supply on 2711 - the unintended consequence of splitting channels into 32-bit and 40-bit is to force the DT to choose one or the other, which can be restricting in the case where either would do.

The 0x01f5 is a mistake that crept in when the 40-bit channels were assigned to the ARM - we can drop the override in bcm2711-rpi.dtsi; 2598419 does so.

HiassofT, 2020-10-12:
Thanks a lot for the explanation and the DT fix!

Getting back channels 9 and 10 already helps a lot, and with the possibility to free up channel 0 I think this should be fine for most cases - otherwise we probably need to manually juggle around DMAs, eg switching SPI over to dma40.

PS: could you please push the DT fix to 5.9 as well? Did a quick test with that on 5.9 and it worked fine.

closed 2020-10-12 by HiassofT

pelwell, 2020-10-12:
I left rpi-5.9.y building before pushing the patch, but it's done now.

HiassofT, 2020-10-12:
awesome, thanks a lot!