raspberrypi/linux #3900: Add CAN_ISOTP Kconfig flag for Kernel 5.9

menschel, 2020-10-13:
Hi,

there is an upstream Kconfig flag that is valuable to catch. It is currently queued for net-next.

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/diff/net/can/Kconfig?id=2ad119d998e49003327b81edb7b854d0983b5d58

It would be great to have ISOTP CAN Protocol enabled by default on the raspberry pi, e.g. CAN_ISOTP=Y .

Thanks and Best Regards

6by9, 2020-10-14:
You're a bit premature. There are no 5.10 release candidates yet, therefore no rpi-5.10.y branch that could possibly add that config.

As it's a builtin option rather than a module it also needs to be evaluated for the amount of RAM it will consume on EVERY system, whether they care about CAN or not. At least a module only consumes some amount of storage space.

6by9, 2020-10-14:
You'd requested ```CAN_ISOTP=Y ``` which would make it built in.

We don't generally use milestones as they are an additional admin overhead to then marshal.

menschel, 2020-10-14:
Ooops, that's due to copy and paste, sorry. Please make it an M.

menschel, 2020-11-06:
I've noticed rpi-5.10.y branch is up and made a pull request.
Sorry for the noise.

closed 2020-11-06 by menschel