raspberrypi/linux #3901: Request: Enable exFAT driver as module in 5.9

gjsman, 2020-10-14:
Not a bug, but a request.

Linux 5.7 includes a new exFAT driver module from Samsung, which allows exFAT drives to natively load without the use of FUSE, and 5.8 improves that driver. I believe that for my use case and others, exFAT would be an excellent module to enable by default without the need for a custom compile. I run a Plex Server on a Pi with legally-obtained content (no torrenting) on an external SSD, so I rarely need to do writes. ExFAT gives me the ability to take that SSD with me on trips and have my full library on Windows and Mac machines natively with read/write, which is not possible with NTFS, BtrFS, or ext4; and I can have files over 4GB in size unlike FAT32.

I brought up the current need to compile-in exFAT at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=288021 , and 6by9 said that I should propose its inclusion here. The module is only (according to 6by9) 97kb in size, or ~400kb when added to all variants, thus adding very low overhead for the features it offers.

osddeitf, 2020-10-15:
Me too, same request.

pelwell, 2020-10-16:
Commit fc4e05f enables exfat support in the standard defconfigs.

osddeitf, 2020-10-16:
> Commit [fc4e05f](https://github.com/raspberrypi/linux/commit/fc4e05f2797c8e2744e7c0e037e3d7e051996049) enables exfat support in the standard defconfigs.

Thank you.

gjsman, 2020-10-16:
Thanks! Can't wait to try it when rpi-update gets a new build. :)

closed 2020-10-16 by gjsman