raspberrypi/linux #3902: 5.9 kernel breaks old dphys-swapfile generated swap files with "swapon: swapfile has holes"

trejan, 2020-10-14:
Need https://lkml.org/lkml/2020/9/4/259 to be applied. The earlier ext4 change means swapon thinks the fallocate created swap file is sparse and refuses to use it.

dphys-swapfile has a Debian specific patch to use fallocate to create the swap file instead of dd which is the recommended method. A newer dphys-swapfile has reverted this patch. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=942825 Old swap file needs to be deleted first as dphys-swapfile won't remove it even if it can't use it.

pelwell, 2020-10-14:
You mention three changes:
1. applying the kernel patch,
2. getting a newer dphys-swapfile, and
3. deleting the swap file.

Are all three measures required? If not, what are the interactions between the changes?

trejan, 2020-10-14:
It is 1 or 2+3 but preferable to just do 1 and patch the kernel.

pelwell, 2020-10-14:
Thanks. The back-port to 5.9 is now available as 862c451bf3e8.

trejan, 2020-10-14:
All good here. I've tested it and a fallocate swap file is working again. Thanks.

closed 2020-10-14 by trejan