raspberrypi/linux #3903: kernel default configuration on raspberry pi zero failed

Ailsamly, 2020-10-14:
**I am following the official guide to locally build kernel to my raspberry pi zero** https://www.raspberrypi.org/documentation/linux/kernel/building.md#default_configuration
**But when it came to preparing the default configuration by running the following commands,**
_cd linux
KERNEL=kernel
make bcmrpi_defconfig_
**it failed and showed:**
_drivers/gpu/drm/Kconfig:246: can't open file "drivers/gpu/drm/i915/Kconfig"
make[1]: *** [scripts/kconfig/Makefile:90: bcmrpi_defconfig] Error 1
make: *** [Makefile:590: bcmrpi_defconfig] Error 2_

**Does anybody know the problem and how to deal with this?
Thanks!

Leyuan**

6by9, 2020-10-15:
Did the git clone complete successfully? It sounds like you've run out of storage space for the kernel tree and the clone has aborted part way through.

Ailsamly, 2020-10-15:
> Did the git clone complete successfully? It sounds like you've run out of storage space for the kernel tree and the clone has aborted part way through.

@6by9 yes, that was done successfully

pelwell, 2020-10-15:
What does `git status` show?