raspberrypi/linux #3905: md0 array goes offline rear massive data transfer

hardwareadictos, 2020-10-15:
Hey, good morning guys.

As the title describes i have a RAID1 array mounted inside two WD 2,5" REDs and rear an hour of transfering high amounts of data i lose the array, it simply goes offline.

I have those drives plugged into two X829 boards:

Bus 002 Device 004: ID 174c:0829 ASMedia Technology Inc. 
Bus 002 Device 003: ID 174c:0829 ASMedia Technology Inc.

I cannot remount the FS rear the failure, only way is rebooting the whole server. DMESG attatched.

[md0_failure.txt](https://github.com/raspberrypi/linux/files/5383912/md0_failure.txt)

Of course, let me know about any other information requiered

Thanks in advance :D

hardwareadictos, 2020-10-21:
No one can help me with this?

pelwell, 2020-10-21:
This question is better asked in [the forums](https://www.raspberrypi.org/forums/index.php).

hardwareadictos, 2020-10-21:
> This question is better asked in [the forums](https://www.raspberrypi.org/forums/index.php).

Thanks you, will ask there also. But as dmesg is a kernel messages diagnostic log, can we also diagnose here if that could be caused by a kernel bug?

Im opened to share whatever log is needed for it to discard a kernel issue :D

Thanks in advance!

pelwell, 2020-10-21:
Unfortunately any useful information that might have been logged has been flushed out by those `rejecting I/O to offline device` messages, which appear at a ridiculous rate. That message would have to be disabled or rate-limited if you want to get more information. On the other hand, you might find some more history in /var/log/syslog or /var/log/messages.

hardwareadictos, 2020-10-21:
> Unfortunately any useful information that might have been logged has been flushed out by those `rejecting I/O to offline device` messages, which appear at a ridiculous rate. That message would have [...]

Thank you! Actually i'm doing some nextcloud syncs wich sadly makes the bus to crash often, will post those logs soon

hardwareadictos, 2020-10-23:
> Unfortunately any useful information that might have been logged has been flushed out by those `rejecting I/O to offline device` messages, which appear at a ridiculous rate. That message would have [...]

Here you have syslog and messages, not useful information under my pov... :(

[messages.txt](https://github.com/raspberrypi/linux/files/5430412/messages.txt)
[syslog.txt](https://github.com/raspberrypi/linux/files/5430413/syslog.txt)

pelwell, 2020-10-23:
I agree with your assessment: we've lost the start of the problem, unless "sd 1:0:0:0: rejecting I/O to offline device" is how it starts - but that seems unlikely.

I should have asked the question before, but this is a powered array, isn't it?

hardwareadictos, 2020-10-23:
Yes it is. Via a 4A@5V power supply. And it's not defective as I tried with another one to discard it. Could be relative to some overvoltage needed?

pelwell, 2020-10-23:
Just for diagnosing the fault there's no harm in adding `over_voltage=2`, but I have a feeling it won't make any difference. Only this week found a problem in the VL805 USB controller when talking to USB serial interfaces, but we haven't yet worked out if it could affect other devices as well.

hardwareadictos, 2020-10-23:
Let me try. Give me some days to test it. I still have large amounts of data to work with :)

hardwareadictos, 2020-10-25:
Well, with the over voltage parameter the array is not being lost anymore and i don't get those blocks messages, but i still get issues on large file transfers (concretely when i transfer lots of small files):

[image]

I can say that 2 weeks ago i wasn't having such issue, and some hardware components were updated like boot loader, kernel, etc..

Could be the cause that VL805 USB controller problem you mention @pelwell ??

hardwareadictos, 2020-10-30:
Okay, today the array got disconnected again, not a voltage issue seems....

hardwareadictos, 2020-11-25:
Hey, issue seems to still happen, not that often rear some updates, but still going on