raspberrypi/linux #3910: [Pi4] [Kernel 5.9.y] Kernel errors with vc4-hdmi audio device.

vanfanel, 2020-10-18:
Hi,

I am running latest NEXT branch kernel
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.9.0-v8+ raspberrypi/firmware#1351 SMP PREEMPT Wed Oct 14 12:50:37 BST 2020 aarch64 GNU/Linux
```
I am using the vc4-hdmi audio device, so **dtparam=audio=on** is commented in my config.txt.

Everything looks fine, except I get this everytime a program using ALSA audio quits:

```
pi@raspberrypi:~ $ dmesg
[  275.293236] BUG: scheduling while atomic: retroarch/444/0x00000002
[  275.294654] Modules linked in: hid_logitech_hidpp joydev vc4 cec drm_kms_helper v3d gpu_sched hid_logitech_dj raspberrypi_hwmon i2c_brcmstb drm drm_panel_orientation_quirks bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem snd_soc_core videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_compress videobuf2_common snd_pcm_dmaengine videodev snd_pcm mc vc_sm_cma(C) snd_timer snd syscopyarea sysfillrect sysimgblt rpivid_mem fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev
[  275.304664] Preemption disabled at:
[  275.304685] [<ffffffc008b3242c>] snd_pcm_stream_lock_irq+0x3c/0x48 [snd_pcm]
[  275.308288] CPU: 2 PID: 444 Comm: retroarch Tainted: G        WC        5.9.0-v8+ raspberrypi/firmware#1351
[  275.310145] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[  275.312021] Call trace:
[  275.313886]  dump_backtrace+0x0/0x1b0
[  275.315751]  show_stack+0x20/0x30
[  275.317547]  dump_stack+0xe0/0x14c
[  275.319301]  __schedule_bug+0xb0/0xc0
[  275.321033]  __schedule+0x75c/0x878
[  275.322705]  schedule+0x50/0x110
[  275.324300]  schedule_hrtimeout_range_clock+0x94/0x118
[  275.325881]  schedule_hrtimeout_range+0x1c/0x28
[  275.327381]  usleep_range+0x7c/0xb0
[  275.328829]  vc4_hdmi_write_infoframe+0x2f0/0x690 [vc4]
[  275.330394]  vc4_hdmi_set_audio_infoframe+0x58/0x80 [vc4]
[  275.331977]  vc4_hdmi_audio_trigger+0x3c/0x288 [vc4]
[  275.333541]  snd_soc_pcm_dai_trigger+0x5c/0x108 [snd_soc_core]
[  275.335234]  soc_pcm_trigger+0xc0/0xd8 [snd_soc_core]
[  275.336953]  snd_pcm_do_start+0x3c/0x48 [snd_pcm]
[  275.338577]  snd_pcm_action_single+0x50/0xb0 [snd_pcm]
[  275.340148]  snd_pcm_action+0x84/0xa0 [snd_pcm]
[  275.341645]  snd_pcm_action_lock_irq+0x40/0x68 [snd_pcm]
[  275.343079]  snd_pcm_common_ioctl+0xf8c/0x12a0 [snd_pcm]
[  275.344440]  snd_pcm_ioctl+0x38/0x50 [snd_pcm]
[  275.345805]  __arm64_sys_ioctl+0xb0/0xf8
[  275.347043]  el0_svc_common.constprop.0+0x7c/0x1b0
[  275.348273]  do_el0_svc+0x2c/0x98
[  275.349459]  el0_sync_handler+0x9c/0x1b8
[  275.350674]  el0_sync+0x158/0x180

```
Is this expected with the experimental 5.9.y kernel?

popcornmix, 2020-10-19:
@mripard This looks like an issue with the upstream driver. `wait_for` calls `usleep_range` and that's not allowed from atomic context of `vc4_hdmi_audio_trigger`. I guess this either needs a busy wait or moved to a worker thread.

mripard, 2020-10-21:
I'm not even sure the context should be atomic. Sending our infoframes definitely isn't, so changing from usleep to udelay would only partially fix the problem. Indeed  we would still be busy-waiting while having the IRQ disabled, and it's not really optimal. I've looked into ALSA/ASoC, and it looks like we mark our device as non-atomic and it would take a mutex instead, which sounds better.

The patch is (compile-tested only):
```
diff --git a/drivers/gpu/drm/vc4/vc4_hdmi.c b/drivers/gpu/drm/vc4/vc4_hdmi.c
index 33c1bcd24907..e80ff0c75e57 100644
--- a/drivers/gpu/drm/vc4/vc4_hdmi.c
+++ b/drivers/gpu/drm/vc4/vc4_hdmi.c
@@ -1467,6 +1467,7 @@ static int vc4_hdmi_audio_init(struct vc4_hdmi *vc4_hdmi)
 		return ret;
 	}
 
+	dai_link->nonatomic	= true;
 	dai_link->cpus		= &vc4_hdmi->audio.cpu;
 	dai_link->codecs	= &vc4_hdmi->audio.codec;
 	dai_link->platforms	= &vc4_hdmi->audio.platform;
```
I'll test it later. Do you need a particular setup to reproduce it, or is any ALSA program (like aplay) would reproduce it?

vanfanel, 2020-10-21:
@mripad  Any program will do, even speaker-test.

mripard, 2020-10-22:
It turns out that this isn't so easy to fix. I'm not sure what the proper fix would be here, so I sent a mail asking for advices:
https://lore.kernel.org/lkml/20201022095041.44jytaelnlako54w@gilmour.lan/T/#u

mripard, 2020-10-28:
I sent and merged a patch fixing this yesterday:
https://lore.kernel.org/dri-devel/20201027101558.427256-1-maxime@cerno.tech/

vanfanel, 2020-10-31:
@mripard Thanks a lot!

@popcornmix Is a new NEXT branch firmware planned soon? I think it should include this fix, the large error whenever a sound is played looks very nasty on the console.

popcornmix, 2020-10-31:
@vanfanel please test new rpi-update kernel.

vanfanel, 2020-10-31:
@popcornmix Yes, tested! No more errors, I think this can be closed :)
Thanks!

closed 2020-10-31 by vanfanel

mripard, 2020-11-02:
> @mripard Thanks a lot!
> 
> @popcornmix Is a new NEXT branch firmware planned soon? I think it should include this fix, the large error whenever a sound is played looks very nasty on the console.

If you have this error it's because you have lockdep enabled in your kernel config, which isn't great on a production kernel since it comes with a drop in performance

vanfanel, 2020-11-02:
@mripard I had the error on the 5.9 (NEXT branch) kernel. I am not building my own kernel.
@popcornmix Is lockdep enabled in the 5.9.x kernel config then?

popcornmix, 2020-11-02:
We don't have:
CONFIG_LOCKDEP
CONFIG_DEBUG_LOCKDEP
PROVE_LOCKING

enabled.