raspberrypi/linux #3911: rpi-5.4.y branch contains meaningless/obsolete commit cc03aa98561e522feb280c7a95642642ece1136b

ncopa, 2020-10-19:
TL;DR: commit cc03aa98561e522feb280c7a95642642ece1136b can and should be removed from `rpi-5.4.y` branch as it got obsoleted in upstream commit 693cd8ce3f882524a5d06f7800dd8492411877b3

When I merged the `v5.4.72` into `rpi-5.4.y` I bumped into a merge conflict in `net/bluetooth/hci_conn.c` due to commit cc03aa98561e522feb280c7a95642642ece1136b (Bluetooth: Check key sizes only when Secure Simple Pairing is enabled)

When I looked closer at that commit I realized that the logic in the commit does not make any sense at all. The commit does not change anything effectively:
```diff
diff --git a/net/bluetooth/hci_conn.c b/net/bluetooth/hci_conn.c
index 87691404d0c6..5c8a800d7d6f 100644
--- a/net/bluetooth/hci_conn.c
+++ b/net/bluetooth/hci_conn.c
@@ -1285,8 +1285,13 @@ int hci_conn_check_link_mode(struct hci_conn *conn)
                        return 0;
        }
 
-       if (hci_conn_ssp_enabled(conn) &&
-           !test_bit(HCI_CONN_ENCRYPT, &conn->flags))
+       /* If Secure Simple Pairing is not enabled, then legacy connection
+        * setup is used and no encryption or key sizes can be enforced.
+        */
+       if (!hci_conn_ssp_enabled(conn))
+               return 1;
+
+       if (!test_bit(HCI_CONN_ENCRYPT, &conn->flags))
                return 0;
 
        return 1;
```

Commit message says:
>  Fixes: d5bb334a8e17 ("Bluetooth: Align minimum encryption key size for LE and BR/EDR connections")

That commit has an additional condition before the final `return 1` which explains things:
```diff
diff --git a/net/bluetooth/hci_conn.c b/net/bluetooth/hci_conn.c
index bd4978ce8c453..3cf0764d5793f 100644
--- a/net/bluetooth/hci_conn.c
+++ b/net/bluetooth/hci_conn.c
@@ -1276,6 +1276,14 @@ int hci_conn_check_link_mode(struct hci_conn *conn)
 	    !test_bit(HCI_CONN_ENCRYPT, &conn->flags))
 		return 0;
 
+	/* The minimum encryption key size needs to be enforced by the
+	 * host stack before establishing any L2CAP connections. The
+	 * specification in theory allows a minimum of 1, but to align
+	 * BR/EDR and LE transports, a minimum of 7 is chosen.
+	 */
+	if (conn->enc_key_size < HCI_MIN_ENC_KEY_SIZE)
+		return 0;
+
 	return 1;
 }
```

However, that hunk got reverted/removed later in upstream commit 693cd8ce3f882524a5d06f7800dd8492411877b3 (Bluetooth: Fix regression with minimum encryption key size alignment), which fixes same thing as the now obsolete commit cc03aa98561e522feb280c7a95642642ece1136b.

Once the meaningless cc03aa98561e522feb280c7a95642642ece1136b is removed, v5.4.72 merges cleanly into rpi-5.4.y.

Thanks!

6by9, 2020-10-19:
Removing a commit would involve rewriting the git history, which messes up all the subsequent commit hashes. 

As and when the branch is updated 5.4.72 the conflict will be resolved, probably with a revert of the downstream cherry-pick as it is now available from upstream.
This is one of the reasons we generally avoid cherry-picking upstream commits before they've been merged into at least one of the subsystem branches, if not Linus' master branch.

pelwell, 2020-10-19:
The reversion has been pushed to rpi-5.4.y, so there will be no merge conflict.

ncopa, 2020-10-20:
> Removing a commit would involve rewriting the git history, which messes up all the subsequent commit hashes.

I had the impression that you push force sometimes, so I assumed that you rebased the patches on top of linux-5.4.y or similar. Sorry.

Thanks!

closed 2020-10-20 by ncopa