raspberrypi/linux #3919: Kernel Issues on Idle RPi4s

josefcub, 2020-10-23:
Hello!

I've got a small network of Raspberry Pi 4 4GB models, all running the 2020-08-20 64-bit beta "Lite" image, and doing various jobs from being a Kubernetes cluster to supporting the cluster and a small, private, network attached to it.   Since I started running the 64-bit image, I've noticed that my Pi4s that are -not- doing a lot (specifically my sandbox, the Pi running DNS/DHCP, and the the repository mirror and docker registry Pi) will occasionally go off the rails and produce this:

```
Oct 16 19:53:58 backup kernel: [1154789.177691] xhci_hcd 0000:01:00.0: Ring expansion failed
Oct 16 19:53:58 backup kernel: [1154789.177766] cdc_acm 1-1.3:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
Oct 16 19:53:58 backup led-control[32074]: /usr/local/bin/led-control: line 43: /dev/ttyACM0: Cannot allocate memory
Oct 16 19:53:58 backup kernel: [1154789.182799] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (-12 86)
Oct 16 19:53:58 backup kernel: [1154789.184201] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (-12 86)
Oct 16 19:53:58 backup kernel: [1154789.184332] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (-12 84)
Oct 16 19:53:59 backup kernel: [1154789.226302] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (-12 86)
Oct 16 19:53:59 backup kernel: [1154789.227686] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (-12 86)
Oct 16 19:53:59 backup kernel: [1154789.227810] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (-12 84)
Oct 16 19:53:59 backup kernel: [1154789.241425] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 18446744073709551604 (-34)
```

Those messages repeat many times per second until I reboot the affected Pi.   

The reboot does clear up the issue for anywhere from 8 to 36 hours (aside from the hundreds of megabytes of log if I haven't paid attention in a while). Is this a known issue?

Of all of the Pis in the network and cluster, the three Pis that do this with regularity are the ones that are the most idle.  The database server, the NFS server, and Kubernetes masters and workers are constantly busy, and very rarely do this (and haven't in a month or more since I stood Kubernetes up).  It's the only difference between the Pis that I can find, as they are all nearly identical (differing only in SSD size).  

All Pis are software and firmware up-to-date at this time, as I do boot directly from SSD on all of them.

Swapping the PIs around (a kubernetes worker Pi for my sandbox Pi, for instance) results in the new sandbox Pi exhibiting the same behavior eventually.

The Pis ran the 32-bit Raspbian image doing the same things without issue.

Would any of you have an idea of what's going on here?

Hardware and kernel information:

Raspberry Pi 4B, 4GB

```
$ cat /proc/version
Linux version 5.4.51-v8+ (dom@buildbot) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020

$ cat /etc/debian_version
10.6

$ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 10)
01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)

$ lsusb
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2047:03df Texas Instruments
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
```

I can provide as much information as you require, just let me know what you need.

Thank you!

P33M, 2020-10-26:
The issue is quickly reproducible with the test case because of a coincidence of factors. 
- CDC ACM defines an interrupt endpoint for serial state notifications. The default interval for this endpoint is quite large - 32ms in the case of the g_serial gadget.
- The test code opens the serial port, writes a small string to it, then closes it again. It does this in much less than a 32ms time period.
- When the port is closed, the driver removes all its URBs which causes the xHCI driver to kick the pending transfers off the transfer rings.
- Because of the speed at which this happens, the interrupt transfer almost always never gets executed by the hardware, so the xhci driver issues "Stop Endpoint / Set Transfer Ring Dequeue Pointer" to avoid trampling a TRB still "owned" by the hardware.
- A single interrupt IN transfer is enqueued at any one time, so the TR dequeue pointer specified by the driver predictably advances once per test loop iteration
- When the driver gets to the 256th TRB, it sets the TR Dequeue Pointer to the link TRB- which is a valid thing to do, but the Pi 4 controller enters a bad state. The dequeue pointer as maintained in the endpoint context doesn't get updated and is stuck like this until the device slot is deactivated.

Naively avoiding the link TRB by double-incrementing the TR dequeue pointer if the driver would otherwise stop on it does fix the interrupt endpoint, but breaks the bulk IN endpoint in some other manner.

P33M, 2020-10-27:
rpi-update firmware has a potential fix for this issue. Please test and report back.

josefcub, 2020-10-27:
After eleven hours of testing with the new rpi-update firmware, I am happy to report that:

Tue 27 Oct 2020 06:37:00 PM CDT
512 /sys/kernel/debug/usb/xhci/0000:01:00.0/devices/03/ep00/trbs
512 /sys/kernel/debug/usb/xhci/0000:01:00.0/devices/03/ep02/trbs
512 /sys/kernel/debug/usb/xhci/0000:01:00.0/devices/03/ep03/trbs
512 /sys/kernel/debug/usb/xhci/0000:01:00.0/devices/03/ep04/trbs
2048 total

everything seems to be working correctly.  Thank you very much!

closed 2020-10-27 by josefcub

rayfoss, 2020-11-18:
Raspbian armhf on first boot I wasn't getting UART output. I turned it on with `enable_uart=1`, but that floods my serial console with these messages. I'll try to set GPIO to np for 6,4. Ubuntu 20.04 server arm64 had uart on by default and did not experience the problem.
```
[  213.735018] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.811252] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.820272] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (0 ffffffff)
[  214.869327] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.879636] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.888681] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (0 ffffffff)
[  214.939392] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.949701] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  214.958741] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (0 ffffffff)
[  215.012480] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  215.022790] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  215.031837] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (0 ffffffff)
[  215.095022] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[  216.171252] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
```

pickert, 2020-12-05:
Very interesting as I've an RPi 4B 1GB that has been throwing these after the WiFi and BT suddenly failed after a raspbian version bump.. Can these be from faulty flashes or only HW damage? I always chalked it up to some "permanent" failure due to a ban firmware/rpi-update fiasco or something .. =\

[    1.187791] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 2 state (0 ffffffff)
[    1.218822] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 6 state (0 ffffffff)
[    1.252569] raspberrypi-exp-gpio soc:firmware:gpio: Failed to set GPIO 4 state (0 ffffffff)

lurch, 2020-12-05:
> the WiFi and BT suddenly failed after a raspbian version bump.. Can these be from faulty flashes or only HW damage?

Easiest way to test is to use Raspberry Pi Imager to write the latest version of Raspberry Pi OS onto a fresh SD card, and see if you still get the same errors?

pickert, 2020-12-06:
I have, many times over time.. wondered about vl805 eeprom corruption
possibly? Any thoughts?

lurch, 2020-12-06:
I've been told that EEPROM corruption is **extremely** unlikely; and even if you did have a corrupted eeprom I think you'd have a more extreme failure-mode (e.g. a complete failure to boot) than just a few errors in a logfile :wink:

pelwell, 2020-12-06:
What @lurch said. Plus, don't get confused about the various EEPROMs that are (or aren't) on a Pi 4: most Pi 4's sold so far have 2 EEPROMs - one containing the bootloader (previously kept in a file called bootcode.bin), and one containing the firmware for the VL805 USB controller. The most recent revision of the Pi 4 (and the CM4 and Pi 400) no longer have an EEPROM for the VL805 - it's firmware is part of the main EEPROM and loaded into RAM by the bootloader. There are signature checks for both firmwares, and the effect of corruption of either firmware is likely to be catastrophic.

Headcrabed, 2023-12-02:
Hello @P33M , could this quirk get upstreamed?