raspberrypi/linux #3925: brcmf_fw_complete_request kernel oops when loading firmware

jaapcrezee, 2020-10-25:
**Describe the bug**
I wanted to update the running kernel on my Raspberry PI Zero devices to 5.4.72, but I run into a kernel oops when the system loads the brcm firmware. With 5.4.42 there is no problem with the same configuration.

**To reproduce**
Just boot, 100% reproducible.

**Expected behaviour**
Working system.

**Actual behaviour**
Oops followed by reboot.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
Custom build
* Which firmware version (`vcgencmd version`)?
[    0.078601] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-04-15 11:43, variant start
[    0.079630] raspberrypi-firmware soc:firmware: Firmware hash is 82f9bb929ce2186eb1824178c1ae82902ad6275c
* Which kernel version (`uname -a`)?
[    0.000000] Linux version 5.4.72-01167-gff93994fb3f9 (jaap@jaap) (gcc version 7.5.0 (Buildroot 2020.02)) #3 SMP PREEMPT Sun Oct 25 20:05:49 CET 2020

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
This is how I normally include the kernel driver firmware into my kernel image:

demo_rpi ]$ ls -1 firmware/brcm/
brcmfmac43430-sdio.AP6212.txt
brcmfmac43430-sdio.bin
brcmfmac43430-sdio.clm_blob
brcmfmac43430-sdio.raspberrypi,3-model-b.txt
brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
brcmfmac43430-sdio.txt
brcmfmac43455-sdio.bin
brcmfmac43455-sdio.clm_blob
brcmfmac43455-sdio.txt

demo_rpi ]$ cat kernel.cfg | grep CONFIG_EXTRA_FI
CONFIG_EXTRA_FIRMWARE="brcm/brcmfmac43430-sdio.bin brcm/brcmfmac43430-sdio.txt brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt brcm/brcmfmac43430-sdio.clm_blob brcm/brcmfmac43455-sdio.bin brcm/brcmfmac43455-sdio.txt brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt regulatory.db regulatory.db.p7s"
CONFIG_EXTRA_FIRMWARE_DIR="../firmware"

[log.txt](https://github.com/raspberrypi/linux/files/5435774/log.txt)

jaapcrezee, 2020-10-25:
It seems to be triggered by this commit: https://github.com/raspberrypi/linux/commit/c01471d3f0667f7ca719c7f207502513bdd04577
If I revert that, the oops disappears.

Although this triggers another problem which is just a bit less dramatic
```
[    3.578247] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    3.596743] ieee80211 phy0: brcmf_c_process_clm_blob: clmload (7222 byte file) failed (-52)
[    3.617025] ieee80211 phy0: brcmf_c_preinit_dcmds: download CLM blob file failed, -5
[    3.636781] ieee80211 phy0: brcmf_bus_started: failed: -5
[    3.648745] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-5
[    3.669389] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed
```

jaapcrezee, 2020-10-25:
@pelwell You seem to be the author of c01471d; can you shed your light on it?

pelwell, 2020-10-25:
Where did you get your firmware and clm_blob? The 43430 firmware that we provide has an embedded clm_blob.

jaapcrezee, 2020-10-26:
From here: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/brcm

But IMHO the kernel should not crash, even with bogus firmware (txt) data. Especially because the kernel is crashing on a call to `strnstr`.

pelwell, 2020-10-26:
I agree it shouldn't crash - I was hoping to be able to reproduce the crash (which I think is _after_ the call to `strnstr`) and verify the fix.

jaapcrezee, 2020-10-27:
If you have a patch somewhere, I can also test it for you @pelwell ?

pelwell, 2020-10-27:
There's no clm_blob in that page you linked to.

jaapcrezee, 2020-10-27:
Oh sorry you are right, I got the clm from somewhere different; probably because of kernel log messages. I did not document that, only that I committed it to my own archive on 2020-05-29. It has the following hash:
```
2c4037b6289ea08b79ba3139860ee66f565e8e614ea3e39eda82b7aae1ae155c  brcmfmac43430-sdio.clm_blob
```
I probably took it from here ( https://github.com/RPi-Distro/firmware-nonfree/ ) but not really sure anymore, cannot directly find after quickly searching for it.

But maybe I should ask first where to get the proper firmware .bin files from when using the kernel source from github.com/raspberrypi/linux ?

pelwell, 2020-10-27:
Either of the 43430 firmwares you've found - the one in the kernel.org repo or the one from RPi-Distro/firmware-nonfree - should be OK, but the latter is preferred. But neither need a clm_blob.

1. Can you upload the clm_blob you have somewhere I can download it? Or email it to me - phil@raspberrypi.com. Once I can reproduce the crash it shouldn't take long to fix the bug.

2. Delete or rename the blob and see if it stops crashing and starts working.

jaapcrezee, 2020-10-27:
I think we can leave out 1. Because:

I changed my kernel config to (excluding clm):
```
CONFIG_EXTRA_FIRMWARE="brcm/brcmfmac43430-sdio.bin brcm/brcmfmac43430-sdio.txt brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt brcm/brcmfmac43455-sdio.bin brcm/brcmfmac43455-sdio.txt brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt regulatory.db regulatory.db.p7s"
CONFIG_EXTRA_FIRMWARE_DIR="../firmware"
```
And it still crashes:
```
[    3.365163] 8<--- cut here ---
[    3.368297] Unable to handle kernel paging request at virtual address c0b18c98
[    3.375632] pgd = (ptrval)
[    3.378379] [c0b18c98] *pgd=00a0841e(bad)
[    3.382460] Internal error: Oops: 80d [#1] PREEMPT SMP ARM
[    3.388038] CPU: 0 PID: 74 Comm: kworker/0:2 Not tainted 5.4.72-01167-gff93994fb3f9 #2
[    3.396074] Hardware name: BCM2835
[    3.399560] Workqueue: events request_firmware_work_func
[    3.404967] PC is at brcmf_fw_complete_request+0x28c/0x7f4
[    3.410543] LR is at strnstr+0x5c/0x78
[    3.414347] pc : [<c05b6d94>]    lr : [<c09b7a0c>]    psr: 00000113
[    3.420705] sp : d7813e58  ip : 00000000  fp : c0f06248
[    3.426005] r10: d789f700  r9 : c0b18b20  r8 : 00000366
[    3.431307] r7 : d789fd40  r6 : 00000004  r5 : d789f800  r4 : 00000001
[    3.437930] r3 : c0b9f7e4  r2 : 00000041  r1 : 00000023  r0 : c0b18c98
```

I could remove all those bins and txts again and start working my way up from rpi/firmware-nonfree;

First I will try keeping only 43430-sdio.bin without the regulatory.db files; maybe the problem is triggered by those.

Here are my firmware files:
https://github.com/jaapcrezee/firmware/raw/main/fw.tar.gz

jaapcrezee, 2020-10-27:
Without clm & regulatory files it seems to work

```
[    2.997406] Freeing unused kernel memory: 1024K
[    3.009373] Run /preboot.sh as init process
[    3.023788] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    3.048124] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    3.060333] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.073216] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.090719] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.101563]  mmcblk0: p1 p2 < p5 p6 > p3 p4
[    3.112582] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
1970-01-01 00:00:03 - info    - [    3.134175] usb 1-1: new high-speed USB device number 2 using dwc_otg
Changing the name of the logger from '' to 'preboot'.
1970-01-01 00:00:03 - preboot - i[    3.148427] Indeed it is in host mode hprt0 = 00001101
nfo    - Preboot version 33
1970-01-01 00:00:03 - preboot - info    - Shared library version 20
[    3.267728] mmc1: new high speed SDIO card at address 0001
[    3.307379] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    3.364719] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
[    3.377748] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
1970-01-01 00:00:03 - preboot - info    - Preboot OS detected Linux
1970-01-01 00:00:03 - preboot - info    - Mounting procfs, sysfs and populating /dev based on /sys...
[    3.491093] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[    3.509110] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    3.524202] usb 1-1: Product: USB 2.0 Hub
[    3.537178] hub 1-1:1.0: USB hub found
```

pelwell, 2020-10-28:
I can't reproduce the seg fault - using your clm_blob crashes the 43430 firmware for me. Since you have a fix (don't load the unneeded clm_blob) I'm going to close this.

closed 2020-10-28 by pelwell

jaapcrezee, 2020-10-28:
This is not a fix but a workaround. I will investigate further.

pelwell, 2020-10-28:
Yes and no - if there's no valid reason for doing the thing that causes the issue then it's less clear cut - "Doctor! Doctor! It hurts when i go like this..."

I've read the code again and it's not obvious where the bug is. But if you can come up with a clear set of steps to apply to a stock image and demonstrate the crash then feel free to reopen.