raspberrypi/linux #3932: PianoDACPlus sound card register fails on 5.4.y kernel

allocom, 2020-10-27:
**Describe the bug**
on  dmesg gets below failure report.
[    6.820321] snd-allo-piano-dac-plus soc:sound: ASoC: failed to init link PianoDACPlus: -517
[    6.820393] snd-allo-piano-dac-plus soc:sound: snd_soc_register_card() failed: -517

This card was working 4.19.y kernel without any issue.

@pelwell  Please suggest to resolve the issue.

pelwell, 2020-10-27:
Those errors are actually more like unavoidable warnings, so they should be take with a pinch of salt. Are you a saying that the card doesn't work on 5.4?

allocom, 2020-10-27:
card not working with 5.4.y , card registrations failing.

pelwell, 2020-10-27:
Comment out the `dtoverlay` line in config.txt and reboot, then run the following:
```
$ sudo udevadm monitor -p > udev.txt &
$ sudo dtoverlay allo-piano-dac-plus-pcm512x-audio
```
You can then halt the udevadm using `fg` and Ctrl-C and post the content of udev.txt.

Also post the output of `lsmod`.

allocom, 2020-10-27:
[udev.txt](https://github.com/raspberrypi/linux/files/5445236/udev.txt)
[lsmod_info.txt](https://github.com/raspberrypi/linux/files/5445237/lsmod_info.txt)

pelwell, 2020-10-27:
With the codec drivers modified to probe without hardware, I see different errors:
```
[    6.940927] snd-allo-piano-dac-plus soc:sound: ASoC: Neither/both codec name/of_node are set for PianoDACPlus
[    6.940945] snd-allo-piano-dac-plus soc:sound: ASoC: failed to init link PianoDACPlus: -22
[    6.940962] snd-allo-piano-dac-plus soc:sound: snd_soc_register_card() failed: -22
[    6.941009] snd-allo-piano-dac-plus: probe of soc:sound failed with error -22
```

pelwell, 2020-10-27:
With that fixed, I get:
```
[    7.936008] snd-allo-piano-dac-plus soc:sound: control 2:0:0:Digital Playback Volume:0 is already present
[    7.936028] pcm512x 1-004d: ASoC: Failed to add Digital Playback Volume: -16
[    7.936117] snd-allo-piano-dac-plus soc:sound: ASoC: failed to instantiate card -16
[    7.936364] snd-allo-piano-dac-plus soc:sound: snd_soc_register_card() failed: -16
```
I think you've got some work to do to fix what looks like a name clash (-16 is -EBUSY).

pelwell, 2020-11-12:
This should now be resolved by https://github.com/raspberrypi/linux/commit/95d7686066c3261e9104970a87f279afc49a9d64.