raspberrypi/linux #3941: poweroff/shutdown on a Pi 400 can stall

pelwell, 2020-11-03:
The most recent firmware release (https://github.com/raspberrypi/firmware/commit/0c3ecac52e4ef4dde82e6261b917fd77a884be37) prevents Pi 400 from powering off reliably - there's at least a 50% chance that it will stall with the display still active. Pi 400 bypasses the usual watchdog-based poweroff mechanism and uses one of the GPIOs on the expander to trigger a full power down. Control of the expander is via mailbox calls, and in the failure case the first of two mailbox methods (the GET from the GET/SET pair) appears not to be completing.

Although suspicion first fell on the recent `tryboot` changes, either in the firmware or in the kernel, these appear not to be the cause. `sudo SKIP_KERNEL=1 rpi-update` exonerated the firmware changes, and choosing a few strategic commits to test turned my attention to the 408 upstream commits. A few hours of bisecting later (which took two attempts due to the probabilistic nature of the failure) pointed the finger of suspicion at 0cc67945ea59 - which makes sense as it is a mailbox-related change.

timg236, 2020-11-16:
This fixes shutdown but after rpi-update Pi400 won't power on. It seems that a reboot is happening instead of the GPIO expander shutdown.

pelwell, 2020-11-16:
It works for me.

timg236, 2020-11-17:
My bad. Works for me after a clean rpiupdate. It must have been a slightly out of date local kernel build