raspberrypi/linux #3942: Kernel aarch64 5.10-rc2 pi4 8G keyboard / mouse issue

Dark-Sky, 2020-11-03:
I have no mouse with aarch64 5.10-rc2 on the pi4 8G but all seems good with the pi4 4G board. I have also noticed this while compiling and booting the mainline 5.10-rc1 kernel. I tried on the usb 3.0 and usb 2.0 ports with no luck. The usb MSD boot works ok on both boards.

popcornmix, 2020-11-03:
Did this work on 5.10-rc1?

Dark-Sky, 2020-11-03:
> Did this work on 5.10-rc1?

I believe so but I will double check. I save all kernels I push.

pelwell, 2020-11-03:
It's working for me (mouse and keyboard) on the current rpi-5.10.y ToT (which is 5.10-rc2).

pelwell, 2020-11-03:
Before you ask, it is an 8GB board, and it's running the v8 kernel:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.0-rc2-v8+ (phil@buildbot) (aarch64-unknown-linux-gnueabi-gcc (crosstool-NG crosstool-ng-1.22.0-115-g785dbce - aarch64-linux-gnueabi) 5.2.1 20151005, GNU ld (crosstool-NG crosstool-ng-1.22.0-115-g785dbce - aarch64-linux-gnueabi) 2.26.20160125) #760 SMP PREEMPT Tue Nov 3 17:12:34 GMT 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000100000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000100000-0x0000000036ffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
```

Dark-Sky, 2020-11-03:
I checked and the mouse and keyboard is not working on 5.10-rc1 either on my pi4 8G. I guess I never checked. Is this another one of these board revision pre-production/production things again like the last time when you fixed it?

```
Hardware	: BCM2835
Revision	: d03114
Serial		: xxxxxxxxxxxxxxxxx
Model		: Raspberry Pi 4 Model B Rev 1.4
```

Dark-Sky, 2020-11-03:
```
[    3.181956] sd 0:0:0:0: [sda] Attached SCSI disk
[    3.262431] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[    3.394578] usb 1-1: device descriptor read/64, error -71
[    3.630547] usb 1-1: device descriptor read/64, error -71
[    3.738672] usb usb1-port1: attempt power cycle
[    4.390356] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[    4.999942] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.206513] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.414348] usb 1-1: device not accepting address 4, error -22
[    5.542353] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[    5.544110] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.750474] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.958347] usb 1-1: device not accepting address 5, error -22
[    5.960332] usb usb1-port1: unable to enumerate USB device
```

pelwell, 2020-11-03:
You must be running an upstream kernel - I have:
```
Hardware        : BCM2711
Revision        : d03114
Serial          : 100000005a8b4668
Model           : Raspberry Pi 4 Model B Rev 1.4
```
Have you tried without the disk attached?

Dark-Sky, 2020-11-03:
I left out my serial # number on purpose but if it was upstream I would have to have uboot. Going to install the kernel on a sdcard but may have to wipe out the uboot before hand.

pelwell, 2020-11-03:
It was the `BCM2835` I noticed.

vianpl, 2020-11-03:
USB upstream with v5.10-rc2 on RPi4b+8GB is OK for me.

@Dark-Sky FYI you don't need u-boot to run the upstream kernel. It's just a matter of making sure you update your kernel command line so it points to the right devices.

Dark-Sky, 2020-11-03:
I found the problem here it is the *.elf *.dat files. I had the latest raspberrypi-bootloader / raspberrypi-bootloader-x files installed and when I down graded from (20201029-1 => 20201016-1) My keyboard and mouse now works on 5.10-rc2. I still have that BCM2835 thing though.

```
[ray@pi4 ~]$ sudo pacman -S raspberrypi-bootloader raspberrypi-bootloader-x
warning: downgrading package raspberrypi-bootloader (20201029-1 => 20201016-1)
warning: downgrading package raspberrypi-bootloader-x (20201029-1 => 20201016-1)
```

pelwell, 2020-11-04:
Don't worry about the BCM2835 - it's a feature of the v8 kernel and I was comparing with the wrong Pi.

I don't understand why temporarily reverting the firmware would have helped. I did just manage to break USB on a Pi 400 by using an dtb that was a mixture of 5.4 and 5.10, so it is worth making sure that you do install the .dtbs when you build the kernel.

Are you happy to close the issue?

Dark-Sky, 2020-11-04:
> You must be running an upstream kernel - I have:
> 
> ```
> Hardware        : BCM2711
> Revision        : d03114
> Serial          : 100000005a8b4668
> Model           : Raspberry Pi 4 Model B Rev 1.4
> ```
> 
> Have you tried without the disk attached?

I tried this today booting off the sdcard and not the usb MSD drive and it boots perfect with my keyboard and mouse with the latest raspberrypi-bootloader / raspberrypi-bootloader-x files from the firmware git.

You can close it if you want but I still can not boot from a usb MSD with the latest raspberrypi-bootloader / raspberrypi-bootloader-x and have a keyboard and mouse and I do if I downgrade to the previous version.

Dark-Sky, 2020-11-06:
@pelwell 

I finally got to the bottom of this issue. I switched to the manjaro unstable branch and upgraded all of the packages and now I have mouse and keyboard. I am thinking the upgraded libinput / libevdev fixed the issue.

closed 2020-11-06 by Dark-Sky

Dark-Sky, 2020-11-08:
@pelwell 

I hate to be a pain with this but this issue is till her with the 5.10-rc2 kernel. I had forgotten I had installed the 5.9 kernel when I was testing. I have done some more testing and here is a summary:

Using the latest *.dat / *.elf and the 5.10-rc2 and it's bcm2711-rpi-4-b.dtb
pi4 8G booting on sdcard works fine with the keyboard and mouse
pi4 4G booting on sdcard or usb drive works fine with the keyboard and mouse
pi4 8G booting on usb drive boots but no keyboard and mouse. Errors (Assuming keyboard & mouse here):

```
[    3.149543] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[    3.277762] usb 1-1: device descriptor read/64, error -71
[    3.513708] usb 1-1: device descriptor read/64, error -71
[    3.621836] usb usb1-port1: attempt power cycle
[    4.273507] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[    4.883056] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.089651] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.297500] usb 1-1: device not accepting address 4, error -22
[    5.425505] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[    5.427230] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.633627] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 1.
[    5.841500] usb 1-1: device not accepting address 5, error -22
[    5.843456] usb usb1-port1: unable to enumerate USB device
```
There are 2 scenarios where I can get the pi4 8G to boot from the usb drive on the 5.10-rc2 kernel:

1. Using the 20201016 *.dat / *.elf and the 5.10-rc2 kernel and it's bcm2711-rpi-4-b.dtb
2. Using the latest *.dat / *.elf and the 5.10-rc2 kernel and using the 5.9.3 kernel bcm2711-rpi-4-b.dtb

Pic with info using scenario #2 above:

[image]

When looking at the diff between the kernel 5.9.3 and 5.10-rc2 kernel bcm2711-rpi-4-b.dts the differences seem to be involving VC4 and resetting the usb. I am no programmer but the resetting stands out as it is not in the 5.9.3 bcm2711-rpi-4-b.dts and the mouse and keyboard works ok.

A note on the kernel:

I am the one that does the raspberry pi kernels for manjaro-arm. It uses your bcm2711_defconfig as a base and I change nothing involving it. There are some more modules added that are requested in the forums like for anbox for android emulation, file systems and encryption. So I do not believe the the extra addition of some modules has anything to do with this issue as it gets fixed using the 2 above hacks mentioned and they are all fixes involving the boot partition.

Dark-Sky, 2020-11-12:
@pelwell In case you are wondering your commit 6905ca481ce7618f489cce6ad67b679c3918bc65 today fixed this issue.