raspberrypi/linux #3944: Unstable framerate of SPI camera with ondemand governor since 5.4

BigMuscle85, 2020-11-05:
We have Raspberry Pi 3B+ with a camera connected via SPI interface.

kernel 4.19: When "ondemand" governor is enabled, CPU switches between 2 steps: 600 MHz and 1400 MHz. This can influence SPI clock, so the core clock is fixed at 400 MHz in /boot/config.txt (core_freq=400). Everything works correctly.

kernel 5.x: With "ondemand" governor, CPU frequency change is more smooth. It is changed in 100 MHz steps. But even with fixed core clock, the SPI camera frame rate is unstable and desynchronization occurs very often. It can be minimized a bit with "conservative" governor, but it does not fix it. The problem appears in 5.4 and 5.9 kernels.

When governor is set to "powersave" or "performance", so the CPU frequency is fixed, the problem does not appear. But this way is not effective (fixed 600 MHz is too low for our application...)

"vcgencmd measure_clock arm" shows that the CPU frequency is scaled as expected.
"vcgencmd measure_clock core" shows "frequency(1)=400000000" all the time.

pelwell, 2020-11-05:
What is the processor utilisation when streaming from the camera? How are you driving the camera - a user-space application via spidev or a kernel driver? Either way will rely on the scheduling responsiveness of the ARM cores. For an application that demands low latency and high throughput you are going to have to enable the performance governor. Counter-intuitively, keeping one of the cores non-idle with a busy loop of some kind (in a shell, "while true; do true; done &") should have the same effect on the ARM clock of enabling the performance governor, but without needing to make a system configuration change.

BigMuscle85, 2020-11-05:
CPU usage is between 10% - 20%. Camera is controlled via kernel module which emulates V4L2 video device and sends image data to user.

Currently, we use workaround that fixes CPU frequency at 900 MHz which is sufficient for most cases. But allowing automatic frequency scaling would be definitely better.
`/opt/vc/bin/vcmailbox 0x00038002 12 8 3 900000000 1`

pelwell, 2020-11-05:
What do _you_ think is causing the unstable framerate, and what do you think could be done about it?

BigMuscle85, 2020-11-06:
I actually do not understand why this happen. But it is obvious that the problem is not caused by different CPU frequency but rather the change itself. Maybe the problem is not visible with 4.19 kernel because there is only one change from 600 to 1400 MHz but there are much more changes (in 100 MHz steps) in 5.x kernels. Could there be some problem that negatively influences the latency during the CPU frequency change?

Is there some possibility to set the frequency step to see how it behaves with the different steps (e.g. 200 or 400 MHz)?

BigMuscle85, 2020-11-06:
I made an experiment with "conservative" governor, freq_step settings changed to 100 so it changes between min and max value only. Core frequency fixed at 400 MHz.

/sys/devices/system/cpu/cpufreq/conservative/freq_step = 100
/sys/devices/system/cpu/cpufreq/conservative/ignore_nice_load = 0
/sys/devices/system/cpu/cpufreq/conservative/down_threshold = 20
/sys/devices/system/cpu/cpufreq/conservative/up_threshold = 80
/sys/devices/system/cpu/cpufreq/conservative/sampling_rate = 20000
/sys/devices/system/cpu/cpufreq/conservative/sampling_down_factor = 1

Executing the command `watch -n 0.1 "vcgencmd measure_clock arm && vcgencmd measure_clock core"` to see how the CPU frequency is scaled. Kernels tested 4.19.127-v7+ and 5.9.2-v7+. The CPU frequency now switches between 600 MHz and 1400 MHz in the same manner with both kernels. But SPI camera framerate is stable with 4.19 and very unstable with 5.9.

popcornmix, 2020-11-06:
> Also cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_transition_latency: 4.19 returns 355000, but 5.9 returns 4294967295. Is that correct?
4294967295 is -1 or:
```
/*
 * Maximum transition latency is in nanoseconds - if it's unknown,
 * CPUFREQ_ETERNAL shall be used.
 */

#define CPUFREQ_ETERNAL			(-1)
```
so that isn't supicious

popcornmix, 2020-11-06:
What about switching to userspace governor and manually writing frequencies to:
/sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed

Is framerate unstable when a specific frequency is set? Or when changing? Or not at all?

BigMuscle85, 2020-11-11:
It seems that increasing /sys/devices/system/cpu/cpufreq/conservative/sampling_rate to 200000 reduces the problem. It does not appear so frequently then but still appears.

I also did some testing:
- problem does not appear when changing governor to "userspace" and single frequency change via "scaling_setspeed"
- contant change of frequency via "scaling_setspeed" in while loop:
 -- 5.9: problem appears almost immediately (also when changing between 600 and 800 MHz only)
 -- 4.19: only slight absolutely irrelevant FPS jumping (8.8 -> 8.7)

```
sudo sh -c "echo userspace > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor"
while :
do
  sudo sh -c "echo 600000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_setspeed"
  sudo sh -c "echo 1400000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_setspeed"
done
```

BigMuscle85, 2021-04-22:
Is there something new? Currently on kernel 5.10 and it is still broken. 
Is there something I could check or debug?

/var/log/kern.log is randomly being filled with the following warnings that come from Lepton Kernel module
```
Apr 22 20:56:34 pi kernel: [   27.607020] lepton:VSYNC 1935 miss!
Apr 22 20:56:34 pi kernel: [   27.610853] lepton:Lost frame sync!
Apr 22 20:56:35 pi kernel: [   27.967907] lepton:VSYNC 1973 miss!
Apr 22 20:56:35 pi kernel: [   27.981150] lepton:Lost frame sync!
Apr 22 20:57:01 pi kernel: [   55.651814] lepton:VSYNC 4888 miss!
Apr 22 20:57:01 pi kernel: [   55.654409] lepton:Lost frame sync!
Apr 22 20:57:02 pi kernel: [   56.307114] lepton:VSYNC 4957 miss!
Apr 22 20:57:02 pi kernel: [   56.320369] lepton:Lost frame sync!
Apr 22 21:00:48 pi kernel: [  282.574377] lepton:VSYNC 28782 miss!
Apr 22 21:00:48 pi kernel: [  282.576286] lepton:Lost frame sync!
Apr 22 21:01:03 pi kernel: [  297.418272] lepton:VSYNC 30345 miss!
Apr 22 21:01:03 pi kernel: [  297.422051] lepton:Lost frame sync!
Apr 22 21:01:05 pi kernel: [  299.374675] lepton:VSYNC 30551 miss!
Apr 22 21:01:05 pi kernel: [  299.376479] lepton:Lost frame sync!
Apr 22 21:01:33 pi kernel: [  327.533435] lepton:VSYNC 33516 miss!
Apr 22 21:01:33 pi kernel: [  327.538655] lepton:Lost frame sync!
Apr 22 21:01:39 pi kernel: [  333.022739] lepton:VSYNC 34094 miss!
```

When kernel downgraded to 4.19, every starts working correctly.

mi-hol, 2021-04-23:
Wondering if the changes to "kernel 5.x: With "ondemand" governor" could be the common root cause for this bug and the degraded network performance reported in  #4133.
Do these issues sound related?

BigMuscle85, 2021-04-24:
I'm not aware of degraded network performance. In our systems, we use network for SSH connections only. The problem started after upgrading from 4.19 to 5.4 and still appears in 5.10. Currently we tried to build one system with RPi4 and the problem seems to be much stronger than on RPi3.

Previously (RPi3 + kernel 4.19), similar problem could appear (but not so often) when spi0 thread was running with normal priority while having high CPU load. The solution was to set spi0 thread's priority to real-time. Now, with kernel 5.x, it appears very often regardless the set priority.

The kernel module works in the way that it waits for IRQ interrupt (called VSYNC) on GPIO pin 17 and then it reads packet from SPI. 
- "VSYNC xxx miss!" warning means that another IRQ interrupt comes before previous SPI transaction has been completed.
- "Lost frame sync!" means that packet counter does not match thus some packets have been lost.

BigMuscle85, 2022-02-17:
I switch to kernel 5.15.23 and this problem still appears.