raspberrypi/linux #3945: Enable wireguard on 5.9.y kernel

gongsearch, 2020-11-05:
Hey together,

could you please consider enabling wireguard in 5.9.y?

This would be great.

Thanks
Sven

andrum99, 2020-11-05:
Per https://github.com/raspberrypi/linux/pull/3557, it's already enabled on arm64 builds, but perhaps not on the 5.9 branch.

Neustradamus, 2020-11-09:
Before the end of this year, WireGuard will be in the next LTS "5.10.x" branch.

pelwell, 2020-11-10:
Wireguard is enabled for arm64 kernels on rpi-5.9.y and rpi-5.10.y.

MichaIng, 2020-11-23:
I opened a PR to add it for armhf as well, hope I did that right: #3967

AlphaJack, 2021-01-27:
@pelwell Why not on armv7?

pelwell, 2021-01-27:
As stated in #3555 and #3557:
```
I see you've enabled it on 64-bit only, which ought to make any overhead in size or performance less noticeable.
```
All 64-bit capable platforms are guaranteed to have at least 1GB of RAM and a faster processor.

MichaIng, 2021-01-27:
See my PR which got already merged, i.e. the current 5.10 branch contains the WireGuard module for ARMv7/armhf.

Not sure what the performance and CPU argument should be, since the alternative OpenVPN is much more inefficient than WireGuard and there is really zero problem to run WireGuard even on RPi Zero.

AlphaJack, 2021-01-28:
Indeed, using the DKMS module on a similar device with 100Mb/s ethernet gives me 95-85Mb/s through Wireguard, absolutely without increasing CPU or RAM usage.

WertiaCoffee, 2021-01-28:
Same issue here. Concerned about enabling wireguard in 5.9y

pelwell, 2021-01-28:
Work on 5.9 stopped months ago - 5.10 is an LTS kernel and will be released to the stable branch any day now.

PR #3967 (2020-11-23): configs: Enable WireGuard kernel module on armhf configs
Closes: #3945