raspberrypi/linux #3948: Kernel panic after upgrade to 5.4.72+ #1356

pklosko, 2020-11-07:
Hello,
I run PiAware with RTL-SDR dongle on RPi Zero. Second service running on RPi is OpenVPN as a client to connect to my VPS. Clear instlation by Raspberry Pi Imager (windows), new SD card, WiFi connection, IPv4 addres, 3A power source.

After upgrade (apt-get .... ) from 5.4.51+ to 5.4.72+ kernel (and also firmware-realtek, firmware-bcm ) ... every cca 3 hours kernel panic is raised - see pics at https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=290368#p1755451

I also tried upgrading to 5.4.74+ [unofficial, intalled by rpi-update]. The result is the same

This problems can be solved only by downgrade back to 5.4.51

**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

popcornmix, 2020-11-11:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

pklosko, 2020-11-14:
Hi, 
I've tried the procedure you suggested.
It looks like the last "stable" kernel is 5.4.59 (5.4.59+ #1336 Wed Aug 19 15:57:25 BST 2020 armv6l GNU/Linux). 

The results, logs, documentation of the procedure etc.  is here: https://github.com/pklosko/RPiZeroW-KernelPanic-3948

popcornmix, 2020-11-14:
Are you saying:
https://github.com/Hexxeh/rpi-firmware/commit/0db18b19afb84f78cdd8c708d54c597c67d03441 is stable and https://github.com/Hexxeh/rpi-firmware/commit/caf7070cd6cece7e810e6f2661fc65899c58e297 is unstable? (as that doesn't seem likely).
I'm not seeing all the commit hashes from rpi-firmware in your "rpi-update" log.

pklosko, 2020-11-14:
No,
https://github.com/Hexxeh/rpi-firmware/commit/b1ab7b7113e548a81523a6b5f6a13178cb885587 = UNSTABLE

https://github.com/Hexxeh/rpi-firmware/commit/0db18b19afb84f78cdd8c708d54c597c67d03441  = STABLE

(I didn't test https://github.com/Hexxeh/rpi-firmware/commit/caf7070cd6cece7e810e6f2661fc65899c58e297)

popcornmix, 2020-11-16:
Could you run the two or three more tests to confirm the exact version that caused the error?

pklosko, 2020-11-17:
I don't know what other tests I should do.
Do you mean 3 untested commits
https://github.com/Hexxeh/rpi-firmware/commit/e70a90efc09925adbc84e9dded5526ee155e00e0, https://github.com/Hexxeh/rpi-firmware/commit/17de03e4e2f270694b371a777cac65c70b69a57c and https://github.com/Hexxeh/rpi-firmware/commit/caf7070cd6cece7e810e6f2661fc65899c58e297 ?

popcornmix, 2020-11-17:
Yes.

pklosko, 2020-11-23:
Hi,
Now I`m testing https://github.com/Hexxeh/rpi-firmware/commit/e70a90efc09925adbc84e9dded5526ee155e00e0 ... looks stable
Current uptime [CET] : 23:55:49 up  1:40,  1 user,  load average: 0.73, 0.75, 0.78
P.

pklosko, 2020-12-16:
I'm testing 5.10 (https://github.com/Hexxeh/rpi-firmware/commit/e69c146b47f017176c0a119b6dfa80f187fa9c87) 
Dec 15, 23:24:15 up 10 min,  1 user,  load average: 1.06, 0.89, 0.56
5:00:01 - KERNEL PANIC

pklosko, 2021-02-14:
it was a HW error in the end, issue can be closed

closed 2021-02-14 by pklosko