raspberrypi/linux #3951: raspberry pi 4 create a bad sectors on usb3.0 ext ssd/hdd

evgeserg, 2020-11-09:
On raspberry pi 4 8gb mount a different times 1 ssd(ext4, later ntfs) and wd passport hdd(ntfs) on usb 3.0 ports. After a while i have a problem with read and write a files on ext drives. In dmesg i have next rows:
_[15498.351480] sd 0:0:0:0: [sda] tag#10 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[15498.351533] sd 0:0:0:0: [sda] tag#10 Sense Key : 0x4 [current]
[15498.351562] sd 0:0:0:0: [sda] tag#10 ASC=0x44 ASCQ=0x0
[15498.351587] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x28 28 00 00 1e b2 00 00 01 00 00
[15498.351614] blk_update_request: critical target error, dev sda, sector 2011648 op 0x0:(READ) flags 0x80700 phys_seg 32 prio class 0
[15498.352495] sd 0:0:0:0: [sda] tag#11 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[15498.352514] sd 0:0:0:0: [sda] tag#11 Sense Key : 0x4 [current]
[15498.352532] sd 0:0:0:0: [sda] tag#11 ASC=0x44 ASCQ=0x0
[15498.352550] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x28 28 00 00 1e b3 00 00 01 00 00
[15498.352569] blk_update_request: critical target error, dev sda, sector 2011904 op 0x0:(READ) flags 0x80700 phys_seg 32 prio class 0
[15528.476632] sd 0:0:0:0: [sda] tag#8 uas_eh_abort_handler 0 uas-tag 1 inflight: IN
[15528.476651] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x28 28 00 00 1e b2 00 00 00 08 00
[15528.516610] scsi host0: uas_eh_device_reset_handler start
[15528.667699] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd_

OS i have first ubuntu, next and now raspbyan, soft in OS is transmission, samba, minidlna. The bug fix fot stack was implemented:
_BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1_

The problem has been observed for 3 months. The drives was fixed with victoria 5.28

Please, help with the problem.

Best regards, Evgeniy.

lurch, 2020-11-13:
As it says at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
"Spinning hard-disk drives nearly always require a powered USB hub. Even if it appears to work you are likely to encounter intermittent failures without a powered USB HUB"

There was a similar issue recently (which I now can't find) where a user was getting filesystem-corruption on their HDD, which went away when they used a powered USB hub.

Alternatively, maybe you have a problem with a USB3 adaptor which doesn't support UAS properly? https://www.raspberrypi.org/forums/viewtopic.php?t=245931

evgeserg, 2020-11-13:
Many thanks, Andrew!

Now i'll wait for next errors in dmegs, after change ext. box usb3.0. If
errors still happens, i'll try solution, your describe.

Best regards, Evgeniy.

spcharc, 2020-12-01:
I had also frequently encountered similar IO errors on my Pi3B. Solved by using another charger to power my HDD. Since then my Pi has been up for months without any IO errors.

evgeserg, 2020-12-01:
Many thanks for answers.

For now i change my ext HDD box.

evgeserg, 2020-12-01:
To solve the problem, I changed the external HDM box. I have not seen such errors in the last 14 days.
Thanks a lot for the answers!

closed 2020-12-01 by evgeserg