raspberrypi/linux #3953: vc4-hdmi device: "no controls" in alsamixer until sound is player

vanfanel, 2020-11-10:
I have been using the fantastic vc4-hdmi audio device for months now, since it came to 5.4/5.9 series of Pi kernels, in Rasberry Pi OS.

However, I noticed that ALSAMIXER claims that "there are no controls for this device" until I play a sound through ALSA. Any program does: even simply running speakertest is enough.
Just after running the program, ALSAMIXER allows volume adjustment.

The bad thing about this is that ALSA volume adjustments can not be aplied on startup by the corresponding service, so volume is always at 100% on system boot, and can't be adjusted until a sound is played.

 More information:
-Running latest 5.10.0-rc3 kernel in 64bit mode on Pi4.
-Running latest Raspberry Pi OS 64bit. No X server, no PulseAudio, nothing that can interfere with the issue.
-My `/usr/share/alsa/cards/vc4-hdmi.conf` is set like this:

```
# Configuration for the VC4-HDMI sound card using software IEC958
# subframe conversion

<confdir:pcm/hdmi.conf>
vc4-hdmi.pcm.hdmi.0 {
	@args [ CARD AES0 AES1 AES2 AES3 ]
	@args.CARD {
		type string
	}
	@args.AES0 {
		type integer
	}
	@args.AES1 {
		type integer
	}
	@args.AES2 {
		type integer
	}
	@args.AES3 {
		type integer
	}
	type iec958
	slave {
		format IEC958_SUBFRAME_LE
		pcm {
			type hooks
			slave.pcm {
				type hw
				card $CARD
				device 0
			}
			hooks.0 {
				type ctl_elems
				hook_args [
				{
					name "IEC958 Playback Default"
					optional true
					lock true
					preserve true
					value [ $AES0 $AES1 $AES2 $AES3 ]
				}
				]
			}
		}
	}
	status [ $AES0 $AES1 $AES2 $AES3 ]
}

# default with plug
vc4-hdmi.pcm.default {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type plug
	slave.pcm {
		type softvol
		slave.pcm {
			@func concat
			strings [ "hdmi:" $CARD ]
		}
		control {
			name "PCM Playback Volume"
			card $CARD
		}
	}
}
```
...As instructed here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&p=1636828#p1636828

So, any ideas on what could be going on?

popcornmix, 2020-11-10:
Moved this to linux repo as I suspect the issue would lie there.

andrum99, 2020-11-10:
I'm seeing the same thing on 5.9.3-v7l+ ##1364 installed via rpi-update. I'm runnign KMS on a 32-bit OS on a Pi 4B.

popcornmix, 2020-11-10:
I can reproduce this. I suspect it's an ordering issue of calls in the hdmi_vc4 driver (but no guesses currently as to which calls...)

6by9, 2020-11-10:
What controls are you looking at? Volume control?

I suspect they are implemented by the alsalib plugins, in which case you'll need a way to make alsalib load the plugin, not change the kernel driver. The hardware has no volume control ability.

popcornmix, 2020-11-10:
Boot up with vc4 driver. Run alsamixer. It days no volume control. Run "aplay norah.wav". Run alsamixer. I can control volume now.
That state sounds like a kernel driver issue.

andrum99, 2020-11-10:
> The hardware has no volume control ability.

The recommended ALSA config, which we're using, provides a software volume control.

6by9, 2020-11-10:
> The recommended ALSA config, which we're using, provides a software volume control.

Exactly. The hardware has no capability. There is nothing to load in the kernel.

If there's a software volume control plugin, then it's a matter of getting that userspace library loaded, and presumably that's not happening until it is used.

vanfanel, 2021-01-31:
Any news on this? This bug hit me today again :(
Something else I can try?

vanfanel, 2021-11-03:
@popcornmix This is still happening. Is this a bug or not? I am trying to close old issues.

6by9, 2021-11-03:
It appears to be documented that softvol will not be available immediately
https://alsa.opensrc.org/Softvol

> Note:The new volume control won't appear imidiately! Only after the first usage of the newly defined device (e.g. with speaker-test), should amixer sget Softmaster display the new control. Mixers that were already started before the first usage (like KMix) have to be restarted to adopt the changes. If the new control is still not there, try restarting ALSA or your PC.

It's not something I'm investigating further.

popcornmix, 2021-11-03:
Thanks @6by9. Sounds like this is behaving as expected @vanfanel so I'm not sure there is anything to fix.

vanfanel, 2021-11-03:
@6by9 @popcornmix  Thanks for the information. Nothing to fix, as you guys say.

closed 2021-11-03 by vanfanel

rtomasa, 2022-09-11:
@6by9 @popcornmix I know that this is an old thread, but I'm now experiencing this issue with the new September Raspberry OS release. Now alsamixer does not provide the PCM control for the vc4-hdmi card. The previous version of Raspberry OS worked fine and provided the volume control via alsamixer. I'm using a Pi3+. Any recent change or config that I should be aware to make it work?