raspberrypi/linux #3959: kmssink crashes on Pi3B+ running 5.4.72-v7+ (vc4_plane_mode_set)

gohai, 2020-11-12:
**Describe the bug**
When attempting to playback the test H264 video with GStreamer outside of X using the kmssink, I am getting repeatedly getting the following crash (although it does work at times). This is running the Full KMS stack.

```
[  157.132120] 8<--- cut here ---
[  157.132204] Unable to handle kernel NULL pointer dereference at virtual address 00000064
[  157.132317] pgd = e8cdcaa1
[  157.132358] [00000064] *pgd=36d93835, *pte=00000000, *ppte=00000000
[  157.132455] Internal error: Oops: 17 [#1] SMP ARM
[  157.132514] Modules linked in: rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc joydev vc4 cec drm_kms_helper drm brcmfmac brcmutil drm_panel_orientation_quirks snd_soc_core sha256_generic snd_compress libsha256 snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_hwmon cfg80211 rfkill bcm2835_codec(C) bcm2835_v4l2(C) i2c_bcm2835 v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer videodev snd vc_sm_cma(C) mc uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[  157.133208] CPU: 2 PID: 691 Comm: v4l2h264dec0:sr Tainted: G         C        5.4.72-v7+ #1356
[  157.133311] Hardware name: BCM2835
[  157.133440] PC is at vc4_plane_mode_set+0x9d8/0x13e0 [vc4]
[  157.133551] LR is at vc4_plane_mode_set+0xbc0/0x13e0 [vc4]
[  157.133614] pc : [<7f445748>]    lr : [<7f445930>]    psr: 20000013
[  157.133695] sp : b24ffa98  ip : b24ffa98  fp : b24ffb34
[  157.133755] r10: b53e00c0  r9 : ffff0000  r8 : 00000001
[  157.133813] r7 : 000003c0  r6 : b53e0104  r5 : 00000001  r4 : b8285000
[  157.133889] r3 : 00000060  r2 : 00000002  r1 : 00000780  r0 : b8285000
[  157.133962] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  157.134040] Control: 10c5383d  Table: 324a406a  DAC: 00000055
[  157.134114] Process v4l2h264dec0:sr (pid: 691, stack limit = 0x15237604)
[  157.134194] Stack: (0xb24ffa98 to 0xb2500000)
[  157.134251] fa80:                                                       b24ffb00 00000001
[  157.134343] faa0: 00000000 b24ffab0 00000000 00000000 00000000 00000002 00000002 00000000
[  157.134436] fac0: b8285000 00000001 b8258440 00000008 b53e00c0 00000000 7f453ef0 00000000
[  157.134527] fae0: 00000780 00000000 00000003 00000000 b5042000 00000000 00000000 00000000
[  157.134614] fb00: 00000000 80e04f88 805a140c b8285000 b51b44c0 b831e240 b8285000 7f2eb8f8
[  157.134714] fb20: 7f44ffb0 00000003 b24ffb8c b24ffb38 7f4462b0 7f444d7c b51b44c0 00000002
[  157.134812] fb40: 00000002 00000002 b24f7884 b24f7880 b5042000 b51b44c0 b24ffbb4 b24ffb68
[  157.134904] fb60: 7f391f14 00000003 b51b44c0 b831e240 b8285000 7f2eb8f8 7f44ffb0 00000003
[  157.134994] fb80: b24ffbb4 b24ffb90 7f2e06d0 7f44627c 00000000 b51b44c0 b8258800 00000001
[  157.135085] fba0: 00000000 00000001 b24ffbd4 b24ffbb8 7f2e082c 7f2e05ec b51b44c0 b5042000
[  157.135185] fbc0: b51b44c0 00000001 b24ffc14 b24ffbd8 7f433664 7f2e07e4 b82585d8 b8258800
[  157.135288] fbe0: 00000005 80e04f88 b24ffc14 b5e20840 b5e20840 b51b44c0 00000001 b24f750c
[  157.135389] fc00: ffffffff b24f7508 b24ffca4 b24ffc18 7f38be60 7f43342c b24ffc3c b24ffc28
[  157.135478] fc20: 7f393298 7f39269c b5042000 b5e20840 b24ffc54 b24ffc40 7f2e3970 7f38b2fc
[  157.135569] fc40: b8258800 b5e20840 b6cdc200 07800000 7f42b680 7f3926cc b24ffc70 7f38b2fc
[  157.135660] fc60: b53e00d8 b53e00c0 b24ffc84 b24ffc78 805ad9a4 80e04f88 b24ffc9c b8285000
[  157.135754] fc80: b51b44c0 b8258800 b5e20840 00000000 b53e00c0 00000000 b24ffcc4 b24ffca8
[  157.135842] fca0: 7f38c0a4 7f38b89c b8285000 b51b44c0 b831e240 b5e20840 b24ffcf4 b24ffcc8
[  157.135928] fcc0: 7f2e179c 7f38c090 00000000 7f2e16b4 b53e00c0 b5e20840 00000000 00000000
[  157.136015] fce0: 07800000 04380000 b24ffd44 b24ffcf8 7f395244 7f2e16c0 00000000 00000780
[  157.136096] fd00: 00000438 00000000 00000000 07800000 04380000 b24ffd84 7f38af5c 04380000
[  157.136178] fd20: b53e00c0 b831e270 b5e20840 b831e240 00000000 00000000 b24ffdd4 b24ffd48
[  157.136260] fd40: 7f395a30 7f39514c 00000000 00000780 00000438 00000000 00000000 07800000
[  157.136344] fd60: 04380000 b24ffd84 00000000 07800000 00000000 00000000 00000438 00000780
[  157.136440] fd80: 805ca4a8 b50e9e80 0000099e 0000001c 00000000 00000000 b8258628 b8258908
[  157.136538] fda0: 00000100 80e04f88 00000002 00000000 b8258800 00000002 7f395910 b24ffe34
[  157.136627] fdc0: b52a7f00 c03064b7 b24ffe04 b24ffdd8 7f37a988 7f39591c 00000000 80e04f88
[  157.136721] fde0: 00000030 7f3aa8ac 00000030 b24ffe34 000000b7 b52a7f00 b24ffee4 b24ffe08
[  157.136819] fe00: 7f37abb4 7f37a8cc 00000001 7f3b8258 00000000 b24ffe38 76d75980 7f395910
[  157.136906] fe20: b953e540 00000030 b24ffe34 7477e5d4 00000000 0000004b 0000004a 000000c0
[  157.137005] fe40: 00000000 00000000 00000000 00000780 00000438 00000000 00000000 04380000
[  157.137103] fe60: 07800000 cf589d00 00000000 7fffffff 76d75980 000000f0 b24fff4c b24ffe88
[  157.141084] fe80: 801b7fec 801b5098 b5ea03c0 0046b000 00000054 00000040 b24fff24 b24ffea8
[  157.145200] fea0: 802c3170 802ccc10 00000000 80116138 b5f76200 80e04f88 b24ffefc 7477e5d4
[  157.149300] fec0: b5e6a3d8 c03064b7 7477e5d4 b953e540 00000008 00000036 b24fff6c b24ffee8
[  157.152935] fee0: 8031df80 7f37a9d8 b50e71ac bb29a090 00000000 80e04f88 b24fff24 b24fffb0
[  157.156569] ff00: 0046b290 00000017 b50e9e80 b5f76200 b5f76240 8092e134 b24fff74 80e04f88
[  157.160766] ff20: 8092e134 00000000 00000081 8032a93c 00026ee8 7477e5d4 c03064b7 80e04f88
[  157.164670] ff40: b24fff5c b953e541 00000000 c03064b7 7477e5d4 b953e540 00000008 00000036
[  157.168777] ff60: b24fff94 b24fff70 8031e414 8031dbbc 00026ee8 7477e5d4 c03064b7 00000036
[  157.172736] ff80: 801011c4 b24fe000 b24fffa4 b24fff98 8031e450 8031e3b4 00000000 b24fffa8
[  157.176736] ffa0: 80101000 8031e444 00026ee8 7477e5d4 00000008 c03064b7 7477e5d4 04380000
[  157.180362] ffc0: 00026ee8 7477e5d4 c03064b7 00000036 00000780 00000438 00000000 76efd660
[  157.184013] ffe0: 75ce908c 7477e5b4 75ccf88c 76bbd51c 80000010 00000008 00000000 00000000
[  157.187641] Backtrace:
[  157.191297] [<7f444d70>] (vc4_plane_mode_set [vc4]) from [<7f4462b0>] (vc4_plane_atomic_check+0x40/0x1a4 [vc4])
[  157.194994]  r10:00000003 r9:7f44ffb0 r8:7f2eb8f8 r7:b8285000 r6:b831e240 r5:b51b44c0
[... 31 lines cut ...]
[  157.317448]  r10:00000036 r9:00000008 r8:b953e540 r7:7477e5d4 r6:c03064b7 r5:00000000
[  157.321427]  r4:b953e541
[  157.325375] [<8031e3a8>] (ksys_ioctl) from [<8031e450>] (sys_ioctl+0x18/0x1c)
[  157.329361]  r9:b24fe000 r8:801011c4 r7:00000036 r6:c03064b7 r5:7477e5d4 r4:00026ee8
[  157.333360] [<8031e438>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[  157.337354] Exception stack(0xb24fffa8 to 0xb24ffff0)
[  157.341344] ffa0:                   00026ee8 7477e5d4 00000008 c03064b7 7477e5d4 04380000
[  157.345375] ffc0: 00026ee8 7477e5d4 c03064b7 00000036 00000780 00000438 00000000 76efd660
[  157.349413] ffe0: 75ce908c 7477e5b4 75ccf88c 76bbd51c
[  157.353425] Code: e594a008 e5922050 e5933010 e0833282 (e5930004)
[  157.357495] ---[ end trace 79bf510738c5f9b3 ]---
````
Add a clear and concise description of what you think the bug is.

**To reproduce**
Enable full KMS, have the Pi boot to the console, run:

```gst-launch-1.0 -e filesrc location=/opt/vc/src/hello_pi/hello_video/test.h264 ! h264parse ! v4l2h264dec capture-io-mode=4 ! kmssink```

**System**

* Which model of Raspberry Pi? Pi 3B+
* Which OS and version (`cat /etc/rpi-issue`)? `Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage`
* Which firmware version (`vcgencmd version`)? `version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)`
* Which kernel version (`uname -a`)? `5.4.72-v7+`

gohai, 2020-11-13:
One thing to note: it appears like this crash is occurring much less often after setting `cma=256` and `gpu_mem=128` in `/boot/config.txt`. (I thought the latter is no longer relevant with fKMS, but the `/dev/video*` nodes don't show up, and the `v4l2h264dec` isn't available when I set it to 16M.)

6by9, 2020-11-13:
gpu_mem is required for the codecs, but not for the rendering if using KMS or FKMS.
Setting it to 16M switches to the cutdown firmware which disables a load of firmware services, including the codecs.

gohai, 2020-11-13:
@6by9 That makes sense, thanks. Quick observation: with KMS, and gpu_mem 32M it also doesn't quite work (`bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2`), but it does with 64M.

The original issue still stands, though.

gohai, 2020-11-14:
I might be mistaken (don't have kernel symbols), but I believe the null-pointer dereference happens in the (inlined) `vc4_plane_calc_load`, which does:

```
	crtc_state = drm_atomic_get_existing_crtc_state(state->state,
							state->crtc);
	vrefresh = drm_mode_vrefresh(&crtc_state->adjusted_mode);
```

There is another caller of `drm_atomic_get_existing_crtc_state` in the same file, which does check for `null` before using it, and I am also seeing the same with in some other drivers.

6by9, 2020-11-14:
Can you retest on the 5.9 or 5.10 kernel please? ```sudo BRANCH=next rpi-update``` will get you the latest 5.10. Normal warnings about not doing it on a critical Pi.

5.4 got an early version of the Pi4 KMS support, and it wasn't perfect. Then again this hasn't changed.

A null deref to get to an offset of 0x00000064 seems unlikely to be a the offset to ```adjusted_mode``` in a ```struct drm_crtc_state``` I get to an offset of around 32 (0x20), not 0x64.

You can disable the load_tracker via debugfs, but I'm not sure it disables that bit (it controls ```vc4->load_tracker_enabled``` and not ```vc4->load_tracker_available```)

gohai, 2020-11-14:
Thank you @6by9.

Here some initial impressions with 5.10. With the default `gpu_mem`, the repro GStreamer pipeline wouldn't run:

```
[   36.646934] bcm2835_mmal_vchiq: error -1 queuing message
[   36.646967] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -1 (Not enough GPU mem?)
```

After setting `gpu_mem=128` playback seems to work fine largely, and I haven't seen the specific `vc4_plane_mode_set` null-pointer dereference so far. I did see some other backtraces, though. (Those look similar to ones that I've also seen on 5.4.)

___

(a) This BUG happened when I was trying to exit out `gst-launch-1.0` with Ctrl+C (no relevant message prior to that in dmesg, this is `5.10.0-rc3-v7+`):

```
[   82.323229] ------------[ cut here ]------------
[   82.328455] kernel BUG at kernel/dma/mapping.c:208!
[   82.333613] Internal error: Oops - BUG: 0 [#1] SMP ARM
[   82.338798] Modules linked in: rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc vc4 cec joydev drm_kms_helper brcmfmac brcmutil drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sha256_generic sysimgblt fb_sys_fops backlight raspberrypi_hwmon cfg80211 rfkill i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm snd_timer vc_sm_cma(C) videodev snd mc uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[   82.356851] CPU: 2 PID: 702 Comm: gst-launch-1.0 Tainted: G         C        5.10.0-rc3-v7+ #1370
[   82.362997] Hardware name: BCM2835
[   82.369100] PC is at dma_unmap_sg_attrs+0x2c/0x6c
[   82.375268] LR is at vb2_dc_dmabuf_ops_detach+0x6c/0x70 [videobuf2_dma_contig]
[   82.381486] pc : [<8019ca48>]    lr : [<7f1ba4fc>]    psr: a0000013
[   82.387697] sp : 86a55c58  ip : 80a0150c  fp : 86a55c74
[   82.393906] r10: 86a55e64  r9 : 7f469d9c  r8 : 8529b660
[   82.400080] r7 : 85388bec  r6 : 85388800  r5 : 869b16c0  r4 : 869b1940
[   82.406268] r3 : c0c0c0c0  r2 : 40800000  r1 : 00000000  r0 : 8165ee10
[   82.412454] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   82.417133] Control: 10c5383d  Table: 069e006a  DAC: 00000055
[   82.420292] Process gst-launch-1.0 (pid: 702, stack limit = 0xf3776c9f)
[   82.423461] Stack: (0x86a55c58 to 0x86a56000)
[   82.426612] 5c40:                                                       86a55c7c 86a55c68
[   82.429786] 5c60: 809e25f8 869b1940 86a55c94 86a55c78 7f1ba4fc 8019ca28 00000020 86a55c88
[   82.432984] 5c80: 869b16c0 85b40840 86a55cac 86a55c98 8072d878 7f1ba49c 869b16c0 85b40840
[   82.436189] 5ca0: 86a55cc4 86a55cb0 7f4634f4 8072d790 836b4200 836b4318 86a55ce4 86a55cc8
[   82.439419] 5cc0: 7f4822c4 7f4634c4 809e25f8 836b4200 836b4318 836b4200 86a55cfc 86a55ce8
[   82.442669] 5ce0: 7f349158 7f482288 836b4200 836b4318 86a55d34 86a55d00 7f349d10 7f34910c
[   82.445943] 5d00: 80656dd4 80656500 86a55d2c 858a9688 858a9600 858a968c 858a9600 8529b660
[   82.449230] 5d20: 7f469d9c 86a55e64 86a55d44 86a55d38 7f451ab8 7f349bd4 86a55d64 86a55d48
[   82.452533] 5d40: 7f3053fc 7f451a80 858a9618 858a9600 858a960c 858a9600 86a55d7c 86a55d68
[   82.455868] 5d60: 7f468dc4 7f30538c 858a960c 858a9618 86a55d94 86a55d80 7f46e114 7f468d98
[   82.459227] 5d80: 858a9674 80e05008 86a55da4 86a55d98 7f46e13c 7f46e0a0 86a55dec 86a55da8
[   82.462609] 5da0: 7f469d70 7f46e124 80a0d1b0 80a0d1b0 80a0d1b0 ffffe000 801e001b 86969258
[   82.466008] 5dc0: 86a55de4 e6389c11 8034ee1c 80e05008 85388800 8529b600 00000000 00000000
[   82.469430] 5de0: 86a55dfc 86a55df0 7f469db8 7f469c14 86a55e34 86a55e00 7f454438 7f469da8
[   82.472881] 5e00: 00000000 e6389c11 7f469d9c 7f489b2c 80e05008 7f469d9c 000000af 8529b600
[   82.476355] 5e20: 86a55e64 00000004 86a55f14 86a55e38 7f454664 7f45438c 0000e200 00000001
[   82.479859] 5e40: 7f498498 b776a3bc 834bce40 00000004 c00464af 86a55e64 7e863e94 00000004
[   82.483385] 5e60: 00000000 000000cd 40000013 868cca80 b776a3bc 80e76100 86a55ebc 86a55e88
[   82.486926] 5e80: 8019718c 802271b4 ffffffff 00000000 814e3010 868cca80 86969258 000e0003
[   82.490500] 5ea0: 86a55ed4 86a55eb0 8010bc44 8010cb0c 80e05008 00000000 80100204 00000004
[   82.494108] 5ec0: fffffe30 8035184c 86a55ef4 86a55ed8 8035184c 80195340 c00464af 80e05008
[   82.497746] 5ee0: c00464af e6389c11 86a55f04 c00464af 80e05008 834bce41 834bce40 00000000
[   82.501418] 5f00: 86a54000 00000019 86a55fa4 86a55f18 80344dc8 7f454488 86a55f4c 86a55f28
[   82.505118] 5f20: 8010bc44 8010cb0c 80e05008 7e863e94 c00464af 00000000 80100204 86a54000
[   82.508847] 5f40: 86a55f84 86a55f50 8010cb8c 8010bc20 ffffffff 00000000 00000000 80d8fbd0
[   82.512583] 5f60: 801000a8 c00464af 00000000 80100204 86a54000 e6389c11 86a55fac 73f01c60
[   82.516348] 5f80: 7e863e94 c00464af 00000036 80100204 86a54000 00000036 00000000 86a55fa8
[   82.520123] 5fa0: 80100040 80344c00 73f01c60 7e863e94 00000019 c00464af 7e863e94 00000000
[   82.523930] 5fc0: 73f01c60 7e863e94 c00464af 00000036 75d61cec 0119a1d4 011b5958 011a3a8c
[   82.527739] 5fe0: 75d5808c 7e863e74 75d3e88c 76c2c51c 80000010 00000019 00000000 00000000
[   82.531548] Backtrace:
[   82.535361] [<8019ca1c>] (dma_unmap_sg_attrs) from [<7f1ba4fc>] (vb2_dc_dmabuf_ops_detach+0x6c/0x70 [videobuf2_dma_contig])
[   82.539294]  r4:869b1940
[... cut ...]
[  170.207418] [<80344bf4>] (sys_ioctl) from [<80100040>] (ret_fast_syscall+0x0/0x28)
[  170.207427] Exception stack(0x82f47fa8 to 0x82f47ff0)
[  170.207451] 7fa0:                   73f01c60 7eea8ea4 00000019 c00464af 7eea8ea4 00000000
[  170.207466] 7fc0: 73f01c60 7eea8ea4 c00464af 00000036 75d5dcec 00be91d4 00c03af8 00c031d4
[  170.207477] 7fe0: 75d5408c 7eea8e84 75d3a88c 76c2851c
[  170.207494]  r10:00000036 r9:82f46000 r8:80100204 r7:00000036 r6:c00464af r5:7eea8ea4
[  170.207502]  r4:73f01c60
[  170.207572] ---[ end trace 47e1f087469c3c71 ]---
```

gohai, 2020-12-11:
Hi @6by9: do you have any updates on this issue - anything that you'd like me to try?

My best guess at what's happening is that something is corrupting memory when one is exiting out a `v4l2h264dec`/`kmssink` GStreamer pipeline on the Pi 3. I've seen a whole slew of random crashes - some immediately after quitting out, afters when shutting down the machine afterwards. (I am attaching a random one about a bad page in `sshd` below just as an illustration.)

With the Pi 4, by the way, this doesn't happen for me.

6by9, 2020-12-12:
The development focus is on the 5.10 kernel and a large number of updates/improvements have been made. I don't really want to go back and try to fix things up on 5.4.

Please test with the current 5.10 kernel - you can use "sudo BRANCH=next rpi-update" to get the latest build.

gohai, 2020-12-12:
Thank you for your message, @6by9. My [earlier comment](https://github.com/raspberrypi/linux/issues/3959#issuecomment-727248750) had a some of the various crashes I got on 5.10 after playing video.

gohai, 2020-12-19:
@6by9 I just tested again with the latest 5.10 kernel, and was able to make it crash by exiting a GStreamer pipeline. Oops below. Wishing you happy holidays, most importantly.

````
[  312.647431] kernel BUG at kernel/dma/mapping.c:208!
[  312.651999] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  312.656592] Modules linked in: hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc vc4 cec drm_kms_helper drm brcmfmac brcmutil drm_panel_orientation_quirks snd_soc_core sha256_generic snd_compress snd_pcm_dmaengine syscopyarea cfg80211 sysfillrect sysimgblt fb_sys_fops backlight rfkill raspberrypi_hwmon i2c_bcm2835 snd_bcm2835(C) bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem snd_pcm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc snd_timer videobuf2_memops videobuf2_v4l2 videobuf2_common snd videodev vc_sm_cma(C) mc uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[  312.672403] CPU: 2 PID: 891 Comm: python3 Tainted: G         C        5.10.0-v7+ #1380
[  312.677820] Hardware name: BCM2835
[  312.683181] PC is at dma_unmap_sg_attrs+0x2c/0x6c
[  312.688551] LR is at vb2_dc_dmabuf_ops_detach+0x6c/0x70 [videobuf2_dma_contig]
[  312.693992] pc : [<8019cbe4>]    lr : [<7f2814fc>]    psr: a0000013
[  312.699412] sp : 83a91c58  ip : 80a01508  fp : 83a91c74
[  312.704684] r10: 83a91e64  r9 : 7f3bed84  r8 : 826eea60
[  312.707567] r7 : 83a70bf0  r6 : 83a70800  r5 : 8f4f8680  r4 : 8f4f8c40
[  312.710445] r3 : c0c0c0c0  r2 : 40800000  r1 : 00000000  r0 : 8165ce10
[  312.713253] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  312.716153] Control: 10c5383d  Table: 02ba006a  DAC: 00000055
[  312.718976] Process python3 (pid: 891, stack limit = 0xe2c9bafd)
[  312.721899] Stack: (0x83a91c58 to 0x83a92000)
[  312.724801] 1c40:                                                       83a91c7c 83a91c68
[  312.727736] 1c60: 809e493c 8f4f8c40 83a91c94 83a91c78 7f2814fc 8019cbc4 00000020 83a91c88
[  312.730651] 1c80: 8f4f8680 856b1900 83a91cac 83a91c98 8072f2d4 7f28149c 8f4f8680 856b1900
[  312.733520] 1ca0: 83a91cc4 83a91cb0 7f3b84e0 8072f1ec 82859800 82859918 83a91ce4 83a91cc8
[  312.736527] 1cc0: 7f3d72ac 7f3b84b0 809e493c 82859800 82859918 82859800 83a91cfc 83a91ce8
[  312.739550] 1ce0: 7f461158 7f3d7270 82859800 82859918 83a91d34 83a91d00 7f461d18 7f46110c
[  312.742576] 1d00: 80658630 80657d5c 83a91d2c 855db688 855db600 855db68c 855db600 826eea60
[  312.745621] 1d20: 7f3bed84 83a91e64 83a91d44 83a91d38 7f3a6ab0 7f461bdc 83a91d64 83a91d48
[  312.748635] 1d40: 7f4393fc 7f3a6a78 855db618 855db600 855db60c 855db600 83a91d7c 83a91d68
[  312.751637] 1d60: 7f3bddb0 7f43938c 855db60c 855db618 83a91d94 83a91d80 7f3c30fc 7f3bdd84
[  312.754732] 1d80: 855db674 80e05008 83a91da4 83a91d98 7f3c3124 7f3c3088 83a91dec 83a91da8
[  312.757811] 1da0: 7f3bed58 7f3c310c baaf3a00 baaf3e64 83a91de4 83a91dc0 802272f0 802284d0
[  312.760912] 1dc0: 83a91df4 b582051d 8010bbe8 80e05008 83a70800 826eea00 00000000 00000000
[  312.764016] 1de0: 83a91dfc 83a91df0 7f3beda0 7f3bebfc 83a91e34 83a91e00 7f3a9430 7f3bed90
[  312.767107] 1e00: 00000000 b582051d 7f3bed84 7f3deb2c 80e05008 7f3bed84 000000af 826eea00
[  312.770292] 1e20: 83a91e64 00000004 83a91f14 83a91e38 7f3a9658 7f3a9384 0000e200 00000001
[  312.773488] 1e40: 7f3ed498 00000001 8579c0c0 00000004 c00464af 83a91e64 7eb22b0c 00000004
[  312.776730] 1e60: 00000000 000000cd 83a91e9c 83a91e78 802e67e8 802f1c00 00000001 00000100
[  312.779909] 1e80: 814c6000 00000255 80e05008 00e76000 83a91f2c 83a91ea0 802e7ca4 802e64c8
[  312.783096] 1ea0: 8010bbe8 8010caac 80e05008 00000000 0000080f 80113f58 ffffe000 00e762f4
[  312.786333] 1ec0: 83a91f04 8035215c 83a91ef4 83a91ed8 8035215c 801953f0 c00464af 80e05008
[  312.789580] 1ee0: c00464af b582051d 83a91f04 c00464af 80e05008 8579c0c1 8579c0c0 00000000
[  312.792866] 1f00: 83a90000 0000001d 83a91fa4 83a91f18 8034563c 7f3a9480 00000800 809ea688
[  312.796167] 1f20: 83a91f74 83a91f30 809ea688 7eb22b0c 80e05008 00000000 ffffffff 10c5383d
[  312.799457] 1f40: 10c5387d 00000008 83a91f8c 80e0abe0 0000080f 809ea52c 00e762f4 83a91fb0
[  312.802854] 1f60: 00e762f4 00e93010 83a91fac 83a91f78 80113f58 b582051d 80100d9c 74301e20
[  312.806259] 1f80: 7eb22b0c c00464af 00000036 80100204 83a90000 00000036 00000000 83a91fa8
[  312.809701] 1fa0: 80100040 80345474 74301e20 7eb22b0c 0000001d c00464af 7eb22b0c 00000000
[  312.813104] 1fc0: 74301e20 7eb22b0c c00464af 00000036 75656cec 00e762f4 00e93010 00eaa954
[  312.816536] 1fe0: 7564d08c 7eb22aec 7563388c 76d1951c 80000010 0000001d 00000000 00000000
[  312.820048] Backtrace:
[  312.823530] [<8019cbb8>] (dma_unmap_sg_attrs) from [<7f2814fc>] (vb2_dc_dmabuf_ops_detach+0x6c/0x70 [videobuf2_dma_contig])
[  312.827136]  r4:8f4f8c40
[  312.830681] [<7f281490>] (vb2_dc_dmabuf_ops_detach [videobuf2_dma_contig]) from [<8072f2d4>] (dma_buf_detach+0xf4/0x11c)
[  312.834316]  r5:856b1900 r4:8f4f8680
[  312.838249] [<8072f1e0>] (dma_buf_detach) from [<7f3b84e0>] (drm_prime_gem_destroy+0x3c/0x48 [drm])
[  312.842034]  r5:856b1900 r4:8f4f8680
[  312.845998] [<7f3b84a4>] (drm_prime_gem_destroy [drm]) from [<7f3d72ac>] (drm_gem_cma_free_object+0x48/0x8c [drm])
[  312.849807]  r5:82859918 r4:82859800
[  312.853786] [<7f3d7264>] (drm_gem_cma_free_object [drm]) from [<7f461158>] (vc4_bo_destroy+0x58/0x5c [vc4])
[  312.857647]  r4:82859800
[  312.861566] [<7f461100>] (vc4_bo_destroy [vc4]) from [<7f461d18>] (vc4_free_object+0x148/0x238 [vc4])
[  312.865336]  r5:82859918 r4:82859800
[... cut ...]
[  312.952920] Exception stack(0x83a91fa8 to 0x83a91ff0)
[  312.956863] 1fa0:                   74301e20 7eb22b0c 0000001d c00464af 7eb22b0c 00000000
[  312.960871] 1fc0: 74301e20 7eb22b0c c00464af 00000036 75656cec 00e762f4 00e93010 00eaa954
[  312.964812] 1fe0: 7564d08c 7eb22aec 7563388c 76d1951c
[  312.968808]  r10:00000036 r9:83a90000 r8:80100204 r7:00000036 r6:c00464af r5:7eb22b0c
[  312.972831]  r4:74301e20
[  312.976829] Code: e35c0000 0a000002 e3530002 9a000008 (e7f001f2)
[  312.980697] ---[ end trace 5002afb4955b73f4 ]---
````