raspberrypi/linux #3960: [5.10] enable feature: F2FS compression

pasman1, 2020-11-15:
Hi.

There are plans to polish f2fs compression for production use:

https://lore.kernel.org/lkml/20201015225508.GA2465821@google.com/

I think it's worth testing this feature. 
Initially only lz4 compression method may be enabled, to check a kernel size.

popcornmix, 2020-11-16:
Can you build a kernel with this enabled and confirm it works?
CONFIG_F2FS_FS_COMPRESSION
https://www.raspberrypi.org/documentation/linux/kernel/building.md

We'd also be intersted in kernel size (of /boot/kernel7l.img) with and without the change and the free memory ("free" after boot) with and without.

pasman1, 2020-11-16:
This take some time - I have 2-core laptop only.

pelwell, 2020-11-16:
We can wait.

pasman1, 2020-11-17:
First compilation of 5.10-rc3 32bit. Compressed kernel is bigger about 92kb 
for all compression methods enabled.

pasman1, 2020-12-22:
Ok. I do not need this feature.

closed 2020-12-22 by pasman1