raspberrypi/linux #3962: Can't use USB soundcards simultaneously in Raspberry Pi 4B. Works on Pi 3.

biasha2, 2020-11-17:
**Describe the bug**

Multiple USB  sound cards (even two) can't play simultaneously in Raspberry Pi 4B. Works fine in RPI3.
When trying to play sound second attempt fails with "Not enough bandwidth".
Bandwidth is not that high. Seems it is not properly calculated/checked by USB driver.

**To reproduce**

Playing 10secs one by one - OK
```
pi@raspberrypi:~ $ aplay -d 10 -c 2 -t raw -D hw:v15 -f S24_3LE -r 44100 /dev/zero
Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 44100 Hz, Stereo
pi@raspberrypi:~ $ aplay -d 10 -c 2 -t raw -D hw:v15_1 -f S24_3LE -r 44100 /dev/zero
Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 44100 Hz, Stereo
```
Playing 10secs simultaneously - FAIL
```
pi@raspberrypi:~ $ aplay -d 10 -c 2 -t raw -D hw:v15 -f S24_3LE -r 44100 /dev/zero & aplay -d 10 -c 2 -t raw -D hw:v15_1 -f S24_3LE -r 44100 /dev/zero
[1] 691
Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 44100 Hz, Stereo
Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 44100 Hz, Stereo
aplay: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S24_3LE
SUBFORMAT:  STD
SAMPLE_BITS: 24
FRAME_BITS: 48
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (125011 125012)
PERIOD_SIZE: 5513
PERIOD_BYTES: 33078
PERIODS: (3 4)
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 132300
TICK_TIME: 0
pi@raspberrypi:~ $
[1]+  Done                    aplay -d 10 -c 2 -t raw -D hw:v15 -f S24_3LE -r 44100 /dev/zero

pi@raspberrypi:~ $ sudo journalctl -e
Nov 17 02:37:37 raspberrypi kernel: usb 1-1.2: Not enough bandwidth for new device state.
Nov 17 02:37:37 raspberrypi kernel: usb 1-1.2: Not enough bandwidth for altsetting 1
Nov 17 02:37:37 raspberrypi kernel: usb 1-1.2: 1:1: usb_set_interface failed (-28)
```

**Expected behaviour**

Both soundcards work. Just like it work **on Raspberry 3**, look:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux

pi@raspberrypi:~ $ aplay -d 10 -c 2 -t raw -D hw:v15 -f S24_3LE -r 44100 /dev/zero & aplay -d 10 -c 2 -t raw -D hw:v15_1 -f S24_3LE -r 44100 /dev/zero & sudo usbtop
Bus ID 0 (All USB buses)        To device       From device
  Device ID 1 :                 0.00 kb/s       0.00 kb/s
  Device ID 2 :                 0.00 kb/s       0.00 kb/s
  Device ID 3 :                 8.52 kb/s       5.42 kb/s
  Device ID 6 :                 291.17 kb/s     30.30 kb/s
  Device ID 8 :                 291.17 kb/s     30.30 kb/s
Bus ID 1 (USB bus number 1)     To device       From device
  Device ID 1 :                 0.00 kb/s       0.00 kb/s
  Device ID 2 :                 0.00 kb/s       0.00 kb/s
  Device ID 3 :                 8.52 kb/s       5.42 kb/s
  Device ID 6 :                 291.17 kb/s     30.30 kb/s
  Device ID 8 :                 291.17 kb/s     30.30 kb/s
^Cpi@raspberrypi:~ $
[1]-  Done                    aplay -d 10 -c 2 -t raw -D hw:v15 -f S24_3LE -r 44100 /dev/zero
[2]+  Done                    aplay -d 10 -c 2 -t raw -D hw:v15_1 -f S24_3LE -r 44100 /dev/zero
```
[... 696 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

```

P33M, 2020-11-25:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=273027

There is a limit of a maximum of 1023 bytes of periodic full-speed data per frame on Pi 4. Your two audio devices have a single Isochronous endpoint alternate setting of 588 bytes (and frame interval of one frame), so you can't use both at once.

A workaround is to use the USB-C connector's OTG functionality in xHCI host mode. `rpi-update` firmware has a recent bugfix that allows the use of the config.txt option `otg_mode=1` which will swap the dwc2 controller for the xHCI host.

You will need a USB-C hub, preferably one with power-delivery, that acts as an in-line breakout. One like this, for example: https://www.amazon.co.uk/UGREEN-Aluminum-Pass-Through-Thunderbolt-Compatible/dp/B077FSBGQK/

Alternatively, power the Pi 4 through GPIO pins and use a USB-C to USB Type-A adapter, then plug a regular USB2 hub into the adapter.

biasha2, 2020-11-30:
Thank you for the explanation.

So,
1. Is this a hardware limitation of the VIA Lab's VL805 controller and can't be resolved?
Or
2. Is this a firmware errata that should be resolved in the EEPROM code? Should I raise this issue on the https://github.com/raspberrypi/rpi-eeprom instead? Sounds to me like Raspberry team should forward it to VIA.
Or
3. Should it be resolved by software calculation of the bandwidth? I've tried XHCI_SW_BW_CHECKING quirk, but it just freezes.

The proposed workaround does not really work for me:
When I use USB2 hub it puts 12M limit on the hub, and I receive bandwidth error on 3rd soundcard connected.
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 41, If 0, Class=Hub, Driver=hub/4p, 12M
If I use USB3 hub (not the one you've linked though) then lsusb does not see it, not sure why. No errors from the journalctl.
Plus it's hard for me to power through GPIO, because I want hat.

**For now I transfer sound via ethernet to the Pi3. And Pi3 can easily handle 5 soundcards that I have.** With any hub for 5th...
But this looks extremely lame. I've obtained Pi4 mainly because of a superior USB support described at
https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md
"_General limitations (not Pi 4)_", "_Devices with known issues (not Pi 4)_" it says...
But in reality its grangma does job much better.

It's just odd and sad :(
You should describe such limitations in red on the first page...

P33M, 2020-11-30:
The workaround isn't working because your hub is incorrectly detected as a full-speed hub (not high speed).
Have you updated the firmware+kernel? What is the output of `vcgencmd version` after running `rpi-update`?

biasha2, 2020-12-02:
The workaround indeed works after rpi-update (tested with usb3 hub). However as I said, powering though GPIO is hard for me because of the hat. And seems that 3 ampere usb data y-cables don't exist. Thank you for the workaround anyway.

But is it going to be addressed in a proper way somewhen? What is the answer to my question above - #1, #2 or #3?
It's kinda weird to detour main usb controller to make usb devices to work.

P33M, 2020-12-04:
Some time ago we raised this issue with Via and this resulted in a firmware (0138a1) that moved the FS periodic bandwidth cap from 792 bytes/frame to 1024 bytes/frame. There is likely to be an underlying hardware limitation due to the way FS period transactions are scheduled. 

The controller doesn't support the XHCI CHECK_BANDWIDTH command.

Darry77, 2021-02-22:
> The workaround indeed works after rpi-update (tested with usb3 hub). However as I said, powering though GPIO is hard for me because of the hat. And seems that 3 ampere usb data y-cables don't exist. [...]

I use a USB 3.1 hub with USB PD (power delivery supports) over USB-C that successfully powers the Pi 4 . 

I The USB 3.1 hub has 3 USB A ports and is powered by a 30W USB PD power supply . I believe the USB PD power supply is only actually capable of providing 3A when outputting 5V (I doubt this USB 3.1 hub can actually run on anything other than 5V ). If my assumptions are correct, the the USB 3.1 HUB and Pi 4 are sharing 5V at a max of 3A and seem to be working fine, power-wise . Ironically the USB 3.1 hub has a Via chipset .

The next part of my USB setup is an externally powered USB 2.0 multi-TT (transaction translator hub), which, AFAIK, is important if one wants to run multiple full bandwidth USB 1.1 devices over USB 2.0. This multi-TT USB 2.0 is connected to the USB 3.1 hub that powers the Pi4 .

Finally, I have 3 USB 1.1 "CM106 like" devices (probably actually CM6206 variants) and a Fiio E10K connected to the multi-TT USB 2.0 hub . 

I have yet to test the behavior of this setup with simultaneous 4-way USB 1.1 traffic but, so far, I am not getting any USB bandwidth errors or apparent bandwidth related audio glitches .

PSU : RAVPower RP-PC144
USB 3.1 hub with USB PD : Ankmax P531H
USB 2.0 multi-TT hub : Plugable USB2-HUB-AG7
USB-C cable between Pi 4 and USB 3.1 hub : Ugreen cable rated for 5V/3A, 9V/3A,12V/3A, 15V/3A, Max. 20V/3A(60W)

vschi, 2021-11-05:
I have the same issue.
It would be great if there would be a build in solution instead of adding more hardware than needed in order to overcome this limitation. To overcome this issue I have ordered already 5 different USB soundcards as I thought the problem is caused by them.

I would also like to run 4 USB soundcards on a Raspberry Pi 4B and play continuously mp3 music.
Strangly I'm sometimes able to run music on 2 USB soundcards. But it depends on the mood of the Raspberry or the order of play command.

Or does anybody have another solution for this behavior?

JokerJinlee123, 2021-12-09:
I have tried to use three 2.0 USB Hubs with one sound card plugs in the first port of each hub. The remaining USB port of Raspberry Pi 4 is connected with one sound card using direct plug in method. Turns out it unexpectedly work well and perfect. I able to control each sound card and play simultaneously even though the "dmesg" still showed the error "usb_set_interface failed (-28)". Not sure how does it work, I assume that the USB hub actually have use certain amount of bandwidth so each port will not shared the bandwidth.

mildmojo, 2022-09-21:
I'm getting this same failure on a brand new Pi 4B 8GB:
- RPi OS 11 (bullseye)
- Kernel 5.15.61-v8+
- Bootloader 507b2360eb46af23c05844b289dc5ae4ecfc3cca (from `vcgencmd bootloader_version`)
- VL805 firmware 000138a1 (from `sudo rpi-eeprom-update`)

After connecting 2 or 3 USB sound devices, connecting an additional sound device triggers a stream of "Not enough bandwidth for new device state" in `dmesg` logs. 

Things I've tried after reading and re-reading this github issue, the [forum thread about this issue](https://forums.raspberrypi.com/viewtopic.php?p=1726669) from 2020, and the [forum thread about the firmware update](https://forums.raspberrypi.com/viewtopic.php?t=273027) from 2020:

- Verified that I have VL805 firmware `000138a1` with `sudo rpi-eeprom-update`
- Tried multiple multi-TT USB hubs with Realtek and VIA chipsets (MTT verified by checking `lsusb -v` and finding `bDeviceProtocol 2 TT per port`)
- Tried every combination of connecting sound devices; all on one MTT hub, one each on 3 MTT hubs and one plugged directly, devices split between two or more hubs, using the Pi's USB2/USB3 ports, etc.

I'm developing an audio gadget, so the USB-C port is in OTG mode and I'm unable to use it as an additional host mode port.

I don't fully understand the issue here; for some people the firmware update back in 2020 seemed to solve their problems. I don't think [biasha2 got an answer](https://github.com/raspberrypi/linux/issues/3962#issuecomment-735810030) about whether it's is a hardware issue or a firmware issue.

So, is this solvable, or is it a permanent USB limitation on all Pi 4s? What _is_ the limitation? I thought it was that each transaction translator has a chunk of bandwidth to split between full-speed devices, and the onboard single-TT VL805 quickly exhausts itself after one or two audio devices. I thought a multi-TT hub was supposed to wrap each device in its own partitioned bandwidth pool. That doesn't seem to be happening, or maybe there's a bug in the VL805's bandwidth prediction that's not taking multi-TT hubs into account? Would a CM4 in a carrier board with a multi-TT hub chip behave any better?

Here are my `lsusb` outputs when I have 3 audio devices attached to one USB3 multi-TT hub (VIA VL817 chipset) connected to a USB3 port on the Pi, after the system's already started spewing `usb 1-1.1.4: Not enough bandwidth` errors into `dmesg` output:

<details><summary><code>lsusb -t</code></summary>
<pre>
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 37, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 38, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 38, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 1: Dev 38, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 38, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 39, If 4, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 39, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 39, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 39, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 39, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 4: Dev 40, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 4: Dev 40, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 4: Dev 40, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 4: Dev 40, If 3, Class=Human Interface Device, Driver=usbhid, 12M
</pre>
</details>

<details><summary><code>lsusb -v</code></summary>
<pre>
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            5.15
  iManufacturer           3 Linux 5.15.61-v8+ dwc2_hsotg
  iProduct                2 DWC OTG Controller
  iSerial                 1 fe980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
[... 1849 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
</pre>
</details>

I had such high hopes for the "new" Pi 4's supposed USB upgrades, but this has been a bad experience so far. :grimacing:

Darry77, 2022-09-21:
> I'm getting this same failure on a brand new Pi 4B 8GB:
> 
>     * RPi OS 11 (bullseye)
> 
>     * Kernel 5.15.61-v8+
> 
>     * Bootloader 507b2360eb46af23c05844b289dc5ae4ecfc3cca (from `vcgencmd bootloader_version`)
> 
>     * VL805 firmware 000138a1 (from `sudo rpi-eeprom-update`)
> 
> 
> After connecting 2 or 3 USB sound devices, connecting an additional sound device triggers a stream of "Not enough bandwidth for new device state" in `dmesg` logs.
> 
> Things I've tried after reading and re-reading this github issue, the [forum thread about this issue](https://forums.raspberrypi.com/viewtopic.php?p=1726669) from 2020, and the [forum thread about the firmware update](https://forums.raspberrypi.com/viewtopic.php?t=273027) from 2020:
> 
>     * Verified that I have VL805 firmware `000138a1` with `sudo rpi-eeprom-update`
> 
>     * Tried multiple multi-TT USB hubs with Realtek and VIA chipsets (MTT verified by checking `lsusb -v` and finding `bDeviceProtocol 2 TT per port`)
> 
>     * Tried every combination of connecting sound devices; all on one MTT hub, one each on 3 MTT hubs and one plugged directly, devices split between two or more hubs, using the Pi's USB2/USB3 ports, etc.
> 
> 
> I'm developing an audio gadget, so the USB-C port is in OTG mode and I'm unable to use it as an additional host mode port.
> 
> I don't fully understand the issue here; for some people the firmware update back in 2020 seemed to solve their problems. I don't think [biasha2 got an answer](https://github.com/raspberrypi/linux/issues/3962#issuecomment-735810030) about whether it's is a hardware issue or a firmware issue.
> 
> So, is this solvable, or is it a permanent USB limitation on all Pi 4s? What _is_ the limitation? I thought it was that each transaction translator has a chunk of bandwidth to split between full-speed devices, and the onboard single-TT VL805 quickly exhausts itself after one or two audio devices. I thought a multi-TT hub was supposed to wrap each device in its own partitioned bandwidth pool. That doesn't seem to be happening, or maybe there's a bug in the VL805's bandwidth prediction that's not taking multi-TT hubs into account? Would a CM4 in a carrier board with a multi-TT hub chip behave any better?
> 
> Here are my `lsusb` outputs when I have 3 audio devices attached to one USB3 multi-TT hub (VIA VL817 chipset) connected to a USB3 port on the Pi, after the system's already started spewing `usb 1-1.1.4: Not enough bandwidth` errors into `dmesg` output:
> `lsusb -t`
> `lsusb -v`
> 
> I had such high hopes for the "new" Pi 4's supposed USB upgrades, but this has been a bad experience so far. 😬

I had the same issue you do and tried pretty much the same things you did before giving up on the VL805 and USB 1.x full speed devices .

One possible workaround might be to use USB 2.0 high speed (or faster) devices . I presume that you are using USB 1.x full speed ones. 

As for the CM carrier board, AFAICR, it does not have a discrete USB controller (VL805) and neither does the compute module itself, so the only USB controller is the one integrated into the Pi SOC . However, the CM carrier does have a PCIE slot, so you might be able to use a different USB controller on a PCIE add-in card, which might work better. See https://pipci.jeffgeerling.com/#usb-cards for possible options .