raspberrypi/linux #3963: Kernel 5.9.8 bcm2835_mmal_vchiq error

Dark-Sky, 2020-11-18:
Using RPi pi4 8G

I compiled and installed the latest 5.9.8 kernel and have fbturbo disabled so I can have This as my Renderer:

```
GL_RENDERER   = V3D 4.2
GL_VERSION    = 2.1 Mesa 20.2.1
GL_VENDOR     = Broadcom
```

If I enable fbturbo and get llvmpipe as a renderer there is no issue. I also compiled the new 5.4 and 5.10-rc kernels and there is no issue.

With fbturbo disabled I get this:

```
[    4.819169] ------------[ cut here ]------------
[    4.819986] bcm2835_mmal_vchiq: error -1 queuing message
[    4.820516] WARNING: CPU: 2 PID: 196 at drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c:2469 vchiq_release_internal+0x118/0x1dc
[    4.823402] Modules linked in: crct10dif_ce(+) gpu_sched rfkill cec bcm2835_isp(C+) drm_kms_helper bcm2835_v4l2(C+) bcm2835_codec(C+) raspberrypi_hwmon bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig snd_soc_core videobuf2_memops videobuf2_v4l2 videobuf2_common snd_compress videodev vc_sm_cma(C) snd_bcm2835(C) snd_pcm_dmaengine mc snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio sch_fq_codel crypto_user drm drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 dwc2
[    4.829171] bcm2835-isp bcm2835-isp: bcm2835_isp_get_supported_fmts: get_param ret 4294967295.
[    4.831207] CPU: 2 PID: 196 Comm: systemd-udevd Tainted: G         C        5.9.8-1-MANJARO-ARM #1
[    4.832693] bcm2835-isp bcm2835-isp: Unregister from media controller
[    4.833794] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[    4.833797] pstate: 80000005 (Nzcv daif -PAN -UAO BTYPE=--)
[    4.833810] pc : vchiq_release_internal+0x118/0x1dc
[    4.833812] lr : vchiq_release_internal+0x9c/0x1dc
[    4.833814] sp : ffff80001240b720
[    4.833815] x29: ffff80001240b720 x28: 0000000000000003 
[    4.833818] x27: ffff8000101359e0 x26: ffff800008e8a240 
[    4.833820] x25: 000000000000001f x24: ffff80001240b778 
[    4.833823] x23: ffff0001f3793bbc x22: ffff0001f32a1e00 
[    4.833825] x21: ffff80001135c2b0 x20: ffff0001f3793b88 
[    4.846892] x19: ffff0001f4a9d070 x18: 0000000000000030 
[    4.847822] x17: ffff800011ffaad8 x16: ffff800011ffaadc 
[    4.848735] x15: ffff0001f32a2338 x14: ffffffffffffffff 
[    4.849639] x13: ffff80009240b777 x12: ffff80001240b77f 
[    4.850526] x11: 0000000000000020 x10: 0000000000000001 
[    4.851405] x9 : ffff800010aecbb4 x8 : ffff80001240b6f0 
[    4.852268] x7 : 00000000ff000000 x6 : 0000000000000001 
[    4.853113] x5 : 0000000000000001 x4 : 0000000000000001 
[    4.853934] x3 : 0000000000000000 x2 : 00000000000000ff 
[    4.854717] x1 : 0000000000000000 x0 : 0000000000000000 
[    4.855488] Call trace:
[    4.856247]  vchiq_release_internal+0x118/0x1dc
[    4.856991]  vchiq_release_service_internal+0x24/0x30
[    4.857744]  vchiq_open_service_internal+0xe0/0x12c
[    4.858494]  vchiq_open_service+0x84/0x110
[    4.859224]  vchiq_mmal_init+0x118/0x1e0 [bcm2835_mmal_vchiq]
[    4.859637] random: crng init done
[    4.859973]  bcm2835_codec_create+0x5c/0x450 [bcm2835_codec]
[    4.859980]  bcm2835_codec_probe+0xb8/0x2c4 [bcm2835_codec]
[    4.860730] random: 7 urandom warning(s) missed due to ratelimiting
[    4.861456]  platform_drv_probe+0x60/0xb4
[    4.861459]  really_probe+0xec/0x490
[    4.861462]  driver_probe_device+0x64/0xcc
[    4.861464]  device_driver_attach+0xcc/0xd4
[    4.861467]  __driver_attach+0x90/0x130
[    4.861470]  bus_for_each_dev+0x7c/0xd0
[    4.861472]  driver_attach+0x30/0x3c
[    4.861474]  bus_add_driver+0x114/0x200
[    4.861476]  driver_register+0x84/0x140
[    4.861478]  __platform_driver_register+0x54/0x60
[    4.861482]  bcm2835_v4l2_codec_driver_init+0x30/0x1000 [bcm2835_codec]
[    4.861486]  do_one_initcall+0x54/0x2c0
[    4.861491]  do_init_module+0x60/0x29c
[    4.864618] bcm2835-isp bcm2835-isp: Unregistering node output[0] device node /dev/video0
[    4.865191]  load_module+0x21b8/0x2894
[    4.865193]  __do_sys_finit_module+0xdc/0xf4
[    4.865196]  __arm64_sys_finit_module+0x2c/0x3c
[    4.865200]  el0_svc_common.constprop.0+0x7c/0x1c0
[    4.865202]  do_el0_svc+0x30/0xa0
[    4.865205]  el0_sync_handler+0x9c/0x1d4
[    4.865207]  el0_sync+0x158/0x180
[    4.865212] ---[ end trace 607c87dd57303e37 ]---
[    4.865879] (efault): Unregistering node (null)[0] device node /dev/video0
[    4.866573] ------------[ cut here ]------------
[    4.867253] (efault): Unregistering node (null)[0] device node /dev/video0
[    4.867894] WARNING: CPU: 2 PID: 196 at drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c:2470 vchiq_release_internal+0x150/0x1dc
[    4.868465] (efault): Unregistering node (null)[0] device node /dev/video0
[    4.869052] Modules linked in: crct10dif_ce gpu_sched rfkill cec bcm2835_isp(C+) drm_kms_helper bcm2835_v4l2(C+) bcm2835_codec(C+) raspberrypi_hwmon bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig snd_soc_core videobuf2_memops videobuf2_v4l2 videobuf2_common snd_compress videodev vc_sm_cma(C) snd_bcm2835(C) snd_pcm_dmaengine mc snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio sch_fq_codel crypto_user drm drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 dwc2
[    4.869656] bcm2835_mmal_vchiq: error -1 queuing message
[    4.870452] CPU: 2 PID: 196 Comm: systemd-udevd Tainted: G        WC        5.9.8-1-MANJARO-ARM #1
[    4.871478] bcm2835_mmal_vchiq: mmal-vchiq: VCHIQ close failed
[    4.871994] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[    4.886837] pstate: 80000005 (Nzcv daif -PAN -UAO BTYPE=--)
[    4.887647] pc : vchiq_release_internal+0x150/0x1dc
[    4.888457] lr : vchiq_release_internal+0x9c/0x1dc
[    4.889253] sp : ffff80001240b720
[    4.890006] x29: ffff80001240b720 x28: 0000000000000003 
[    4.890785] x27: ffff8000101359e0 x26: ffff800008e8a240 
[    4.891577] x25: 000000000000001f x24: ffff80001240b778 
[    4.892376] x23: ffff0001f3793bbc x22: ffff0001f32a1e00 
[    4.893129] x21: ffff80001135c2b0 x20: ffff0001f3793b88 
[    4.893896] x19: ffff0001f4a9d070 x18: 0000000000000030 
[    4.894691] x17: ffff800011ffaad8 x16: ffff800011ffaadc 
[    4.895484] x15: ffff0001f32a2338 x14: ffffffffffffffff 
[    4.896245] x13: ffff80009240b777 x12: ffff80001240b77f 
[    4.897043] x11: 0000000000000020 x10: 0000000000000001 
[    4.897851] x9 : ffff800010aecbb4 x8 : ffff80001240b6f0 
[    4.898663] x7 : 00000000ff000000 x6 : 0000000000000001 
[    4.899439] x5 : 0000000000000001 x4 : 0000000000000001 
[    4.900229] x3 : 0000000000000000 x2 : 00000000000000ff 
[    4.901004] x1 : 0000000000000000 x0 : 0000000000000000 
[    4.901790] Call trace:
[    4.902597]  vchiq_release_internal+0x150/0x1dc
[    4.903360]  vchiq_release_service_internal+0x24/0x30
[    4.904201]  vchiq_open_service_internal+0xe0/0x12c
[    4.905151]  vchiq_open_service+0x84/0x110
[    4.905879]  vchiq_mmal_init+0x118/0x1e0 [bcm2835_mmal_vchiq]
[    4.906641]  bcm2835_codec_create+0x5c/0x450 [bcm2835_codec]
[    4.907414]  bcm2835_codec_probe+0xb8/0x2c4 [bcm2835_codec]
[    4.908186]  platform_drv_probe+0x60/0xb4
[    4.908926]  really_probe+0xec/0x490
[    4.909653]  driver_probe_device+0x64/0xcc
[    4.910409]  device_driver_attach+0xcc/0xd4
[    4.911118]  __driver_attach+0x90/0x130
[    4.911802]  bus_for_each_dev+0x7c/0xd0
[    4.912506]  driver_attach+0x30/0x3c
[    4.913176]  bus_add_driver+0x114/0x200
[    4.913783]  driver_register+0x84/0x140
[    4.914355]  __platform_driver_register+0x54/0x60
[    4.914913]  bcm2835_v4l2_codec_driver_init+0x30/0x1000 [bcm2835_codec]
[    4.915475]  do_one_initcall+0x54/0x2c0
[    4.915991]  do_init_module+0x60/0x29c
[    4.916486]  load_module+0x21b8/0x2894
[    4.916969]  __do_sys_finit_module+0xdc/0xf4
[    4.917448]  __arm64_sys_finit_module+0x2c/0x3c
[    4.917922]  el0_svc_common.constprop.0+0x7c/0x1c0
[    4.918390]  do_el0_svc+0x30/0xa0
[    4.918866]  el0_sync_handler+0x9c/0x1d4
[    4.919335]  el0_sync+0x158/0x180
[    4.919800] ---[ end trace 607c87dd57303e38 ]---
[    4.920480] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    4.928665] vc4-drm gpu: bound fe600000.firmwarekms (ops vc4_fkms_ops [vc4])
[    4.929448] checking generic (3e6fa000 3fc000) vs hw (0 ffffffffffffffff)
[    4.929452] fb0: switching to vc4drmfb from simple
```

6by9, 2020-11-20:
What value is gpu_mem set to on your systems? I'm guessing that you've dropped it down to 16, and hence the codec components aren't present in the firmware.
I know 5.10 has an issue under those circumstances (or at least did last time I tried), but 5.9 was OK. That's on the list of things to look into.

Dark-Sky, 2020-11-20:
> What value is gpu_mem set to on your systems? I'm guessing that you've dropped it down to 16, and hence the codec components aren't present in the firmware.

64

Dark-Sky, 2020-11-20:
Looks like I forgot to mention up above that this is a 64bit compile of 5.9.8.

6by9, 2020-11-20:
I think something has gone wrong in a recent rebase. I was running 5.9.6, but have just rebased my tree to 5.9.8 and I just saw a backtrace too.

Dark-Sky, 2020-11-23:
@6by9 
I compiled the 5.9.10 kernel pushed today and looks like the backtrace is gone.

closed 2024-12-12 by Dark-Sky