raspberrypi/linux #3964: ttyUSB not detected after upgrade to 5.4.66+

ensarkarabudak, 2020-11-18:
I’ve run into an issue with the Quectel **BG96** module. I do not have anything on ttyUSB3 and 2.
It appears as though some changes in 5.4.66 have affected how USB2-USB3 is treated? 

[ChangeLog-5.4.66](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.4.66):

>     Change rules for EC21, EC25, BG96 and EG95 to match vendor specific
>     serial functions only, to prevent binding to class functions. Require
>     2 endpoints on ff/ff/ff functions, avoiding the 3 endpoint QMI/RMNET
>     network functions.

**System**
Raspbery Pi 4 Model B or 3B+

```
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4
```

```
pi@raspberrypi:~ $ vcgencmd version
Nov 17 2020 13:40:08 
Copyright (c) 2012 Broadcom
version 20c430d6cf4343ed95afba07c4735b7c587a4c83 (clean) (release) (start)
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.77-v7l+ #1371 SMP Tue Nov 17 13:35:27 GMT 2020 armv7l GNU/Linux
```

**Logs**

```
pi@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 2c7c:0296 Quectel Wireless Solutions Co., Ltd. BG96 CAT-M1/NB-IoT modem
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
```
pi@raspberrypi:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 7, If 0, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 7, If 1, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 7, If 2, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 7, If 3, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 7, If 4, Class=Vendor Specific Class, Driver=qmi_wwan, 480M

```

```
pi@raspberrypi:~ $ dmesg
[  757.680431] usb 1-1.1: USB disconnect, device number 5
[  757.681345] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[  757.681477] option 1-1.1:1.0: device disconnected
[  757.687264] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[  757.687410] option 1-1.1:1.1: device disconnected
[  757.688625] qmi_wwan 1-1.1:1.4 wwan0: unregister 'qmi_wwan' usb-0000:01:00.0-1.1, WWAN/QMI device
[  760.586253] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[  760.720216] usb 1-1.1: New USB device found, idVendor=2c7c, idProduct=0296, bcdDevice= 0.00
[  760.720235] usb 1-1.1: New USB device strings: Mfr=3, Product=2, SerialNumber=4
[  760.720252] usb 1-1.1: Product: Qualcomm CDMA Technologies MSM
[  760.720267] usb 1-1.1: Manufacturer: Qualcomm, Incorporated
[  760.720283] usb 1-1.1: SerialNumber: 5af3af3d
[  760.731565] option 1-1.1:1.0: GSM modem (1-port) converter detected
[  760.731933] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB0
[  760.732472] option 1-1.1:1.1: GSM modem (1-port) converter detected
[  760.732799] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB1
[  760.736519] qmi_wwan 1-1.1:1.4: cdc-wdm0: USB WDM device
[  760.737401] qmi_wwan 1-1.1:1.4 wwan0: register 'qmi_wwan' at usb-0000:01:00.0-1.1, WWAN/QMI device, 16:9c:93:b3:12:dc
[ 1694.517102] usb 1-1.1: USB disconnect, device number 6
[ 1694.517990] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[ 1694.518123] option 1-1.1:1.0: device disconnected
[ 1694.519010] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[ 1694.519136] option 1-1.1:1.1: device disconnected
[ 1694.520294] qmi_wwan 1-1.1:1.4 wwan0: unregister 'qmi_wwan' usb-0000:01:00.0-1.1, WWAN/QMI device
[ 1699.471305] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
[ 1699.605396] usb 1-1.1: New USB device found, idVendor=2c7c, idProduct=0296, bcdDevice= 0.00
[ 1699.605416] usb 1-1.1: New USB device strings: Mfr=3, Product=2, SerialNumber=4
[ 1699.605433] usb 1-1.1: Product: Qualcomm CDMA Technologies MSM
[ 1699.605449] usb 1-1.1: Manufacturer: Qualcomm, Incorporated
[ 1699.605464] usb 1-1.1: SerialNumber: 5af3af3d
[ 1699.624867] option 1-1.1:1.0: GSM modem (1-port) converter detected
[ 1699.625233] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB0
[ 1699.625780] option 1-1.1:1.1: GSM modem (1-port) converter detected
[ 1699.626116] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB1
[ 1699.629701] qmi_wwan 1-1.1:1.4: cdc-wdm0: USB WDM device
[ 1699.630746] qmi_wwan 1-1.1:1.4 wwan0: register 'qmi_wwan' at usb-0000:01:00.0-1.1, WWAN/QMI device, 16:9c:93:b3:12:dc
```

```
pi@raspberrypi:~ $ dmesg | grep tty
[  757.681345] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[  757.687264] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[  760.731933] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB0
[  760.732799] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB1
[ 1694.517990] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[ 1694.519010] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[ 1699.625233] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB0
[ 1699.626116] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB1
```

**Additional context**
I'm solving the problem by downgrading the kernel to **5.4.65**.

The outputs after downgrading the kernel to **5.4.65**:

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.65-v7l+ #1341 SMP Tue Sep 15 13:13:32 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2c7c:0296 Quectel Wireless Solutions Co., Ltd. BG96 CAT-M1/NB-IoT modem
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
pi@raspberrypi:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 3, If 2, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 3, If 3, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 1: Dev 3, If 4, Class=Vendor Specific Class, Driver=qmi_wwan, 480M

```
```
pi@raspberrypi:~ $ dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:49:FB:39 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=3afdcacc-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.001069] printk: console [tty1] enabled
[    1.178126] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 29, base_baud = 0) is a PL011 rev2
[   48.799825] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB0
[   48.800271] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB1
[   48.800707] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB2
[   48.801194] usb 1-1.1: GSM modem (1-port) converter now attached to ttyUSB3

```

dbaba, 2020-11-30:
I have the same issue with the [v1.20201022 firmware](https://github.com/raspberrypi/firmware/releases/tag/1.20201022).

It worked before in [v1.20200819](https://github.com/raspberrypi/firmware/tree/1.20200819) release but no longer works in the following releases.

There's a [change related to Quectel USB stuff](https://github.com/torvalds/linux/commit/2bb70f0a4b238323e4e2f392fc3ddeb5b7208c9e#diff-47fc76895f34c1942f7bba54e15ca520f6aa6d7ed9d3b7a33b24f4b4231a3c93) in the [option.c change logs](https://github.com/torvalds/linux/commits/master/drivers/usb/serial/option.c). But unfortunately I don't have any idea of the relevance between the issue and the change.

ensarkarabudak, 2020-12-01:
any ideas?
@bmork @jhovold

jhovold, 2020-12-01:
On Mon, Nov 30, 2020 at 10:21:06PM -0800, Ensar wrote:

Please report this to linux-usb@vger.kernel.org and include the output
of "lsusb -v" (and/or the more compact output of "usb-devices") for the
device in question. It doesn't hurt CCing me and Bjørn.

bmork, 2020-12-01:
Well, I found this now: https://lists.freedesktop.org/archives/libqmi-devel/2017-October/002509.html

And I see the problem: This device has two ff/ff/ff functions with 3 endpoints, where one of those is RMNET and the other is serial.   This device ID should obviously not have been included in that patch.  Sorry about that.  I'll send you a partial revert shortly, @jhovold 

There is also this ff/fe/ff function, which AFAIK is unique to this device? Weird

ensarkarabudak, 2020-12-10:
Does anyone know what firmware version this fix([USB: serial: option: fix Quectel BG96 matching](https://github.com/torvalds/linux/commit/c98fff7332dbd6e028969f8c2bda3d7bc7a024d8#diff-47fc76895f34c1942f7bba54e15ca520f6aa6d7ed9d3b7a33b24f4b4231a3c93)) will be available on?

jhovold, 2020-12-10:
It has been backported to all stable trees and should show up in the
next release.

ensarkarabudak, 2020-12-25:
This issue fixed in 5.10+ versions. 
Thanks, @jhovold @bmork

closed 2020-12-25 by ensarkarabudak