raspberrypi/linux #3971: Raspberry Pi 4 Modell 4GB USB Ports suspended with Raspberry Pi OS 32

fopinski, 2020-11-26:
**Describe the bug**
Raspberry Pi 4 Modell 4GB USB Ports suspended with RPI OS 32.
Ports can't be activated.

**To reproduce**
Install latest image of Raspberry Pi OS 32 (2020-08-20-raspios-buster-armhf) with Raspberry Imager.

**Expected behaviour**
USB hardware attached to the ports should get recognized.

**Actual behaviour**
USB hardware attached to the ports are not recognized as ports are suspended.

**System**
Raspberry Pi 4 Modell B 4GB 

```
cat /etc/rpi-issue
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4

vcgencmd version
Oct 22 2020 13:59:27
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

uname -a
Linux raspberrypi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux

pi@raspberrypi:~ $ sudo raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4

Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
Revision        : c03112
Serial          : 10000000de9fee10
Model           : Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Oct 22 2020 13:59:27
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30359292 2994148  26081056  11% /
devtmpfs         1827796       0   1827796   0% /dev
tmpfs            1959892       0   1959892   0% /dev/shm
tmpfs            1959892    8616   1951276   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1959892       0   1959892   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   55171    202925  22% /boot
[... 3083 lines cut ...]
[    2.178754] bcmgenet fd580000.ethernet eth0: Link is Down
[    7.560506] urandom_read: 3 callbacks suppressed
[    7.560509] random: wpa_supplicant: uninitialized urandom read (32 bytes read)
[    7.586052] random: dhcpcd: uninitialized urandom read (120 bytes read)
[    8.267127] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   61.655669] random: crng init done
[   61.655674] random: 1 urandom warning(s) missed due to ratelimiting
[  386.287102] usb 1-1.3: new high-speed USB device number 4 using xhci_hcd
[  386.419588] usb 1-1.3: New USB device found, idVendor=04e8, idProduct=1d23, bcdDevice= 1.10
[  386.419593] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  386.419597] usb 1-1.3: Product: Mighty Drive    
[  386.419601] usb 1-1.3: Manufacturer: Samsung 
[  386.419605] usb 1-1.3: SerialNumber: 07731091A1A9
[  386.420947] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[  386.421061] scsi host0: usb-storage 1-1.3:1.0
[  387.447782] scsi 0:0:0:0: Direct-Access     Samsung  Mighty Drive     PMAP PQ: 0 ANSI: 0 CCS
[  388.500670] sd 0:0:0:0: [sda] 4029440 512-byte logical blocks: (2.06 GB/1.92 GiB)
[  388.500797] sd 0:0:0:0: [sda] Write Protect is off
[  388.500802] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
[  388.500925] sd 0:0:0:0: [sda] No Caching mode page found
[  388.506967] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  388.558397]  sda: sda1
[  388.559092] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  443.102100] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

fopinski, 2020-11-27:
The only thing that helps running the 32 bit version is using arm_64bit=1 in /boot/config.txt

fopinski, 2020-12-04:
I installed 5.10 to get the referenced driver but still no USB working:
`Linux raspberrypi 5.10.0-rc6-v7l+ #1 SMP Fri Dec 4 20:52:04 CET 2020 armv7l GNU/Linux`

[raspinfo2.txt](https://github.com/raspberrypi/linux/files/5645431/raspinfo2.txt)

fopinski, 2025-02-26:
It was a hardware defect

closed 2025-02-26 by fopinski

pelwell, 2025-02-26:
Thanks for taking the trouble to report back.

fopinski, 2025-02-26:
It took me a while to get the board replaced but now it's all fine... Should have done that earlier