raspberrypi/linux #3976: boot failure & xhci_hcd error rpi 4B 8GB

dbrown2k, 2020-11-28:
Failure to boot after installing PREEMPT-RT KERNEL rpi-4.19.y-rt, on raspberry pi 4B 8GB, using USB SSD

Installed the [2020-08-20-raspios-buster-armhf.img] then followed instructions [https://gnipsel.com/linuxcnc/uspace/rpi4-rt.html] on reboot gives errors;

xhci_hcd 0000:01:00.0: can't setup: -110
xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110

and fails to boot further, unable to extract logs as boot halts, because the USB fails to start SSD cannot boot.

pelwell, 2020-11-29:
The -rt branches are hosted but not supported. I'll leave this issue open in case somebody else can offer some insight, but we won't be looking into the problem (which will be that the patches required to support running the VL805 XHCI controller without an EEPROM are only found in 5.4 and later).

Fincer, 2020-12-18:
I had non-functioning RPi 4 8GB USB ports (`aarch64`) with exactly same error messages until I fixed a BCM2711 device tree file and replaced several firmware files. Please check out my detailed reply on Arch Linux ARM forums, as well as pierrec's one:

[Arch Linux ARM - No USB detected on Raspberry Pi 4 with aarch64](https://archlinuxarm.org/forum/viewtopic.php?f=65&t=14734&p=65334#p65334)

The fix itself is quite trivial. It requires a proper [BCM2711 device tree file](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/bcm2711-rpi-4-b.dts) with `#include "bcm283x-rpi-usb-peripheral.dtsi"` line which is apparently missing in some official Raspberry Pi kernel archive files as far as I have seen. Additionally, you need correct firmware files as I've explained in my forum post.

I have not tested yet if the fix works for boot process but at least it fixes `xhci_hcd`/USB driver loading issue for Raspberry Pi 4 8GB + `aarch64`/ARM64.

rxdu, 2021-08-14:
> I had non-functioning RPi 4B 8GB USB ports (`aarch64`) with exactly same error messages until I fixed a BCM2711 device tree file and replaced several firmware files. Please check out my detailed reply on Arch Linux ARM forums, as well as pierrec's one:
> 
> [Arch Linux ARM - No USB detected on Raspberry Pi 4 with aarch64](https://archlinuxarm.org/forum/viewtopic.php?f=65&t=14734&p=65334#p65334)
> 
> The fix itself is quite trivial. It requires a proper [BCM2711 device tree file](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/bcm2711-rpi-4-b.dts) with `#include "bcm283x-rpi-usb-peripheral.dtsi"` line which is apparently missing in some official Raspberry Pi kernel archive files as far as I have seen. Additionally, you need correct firmware files as I've explained in my forum post.
> 
> I have not tested yet if the fix works for boot process but at least it fixes `xhci_hcd`/USB driver loading issue for Raspberry Pi 4 8GB + `aarch64`/ARM64.

Thanks so much for the information you provided. This also solved my USB problem with a 32-bit build of kernel 4.19.y-rt on RPi 4B 8GB.