raspberrypi/linux #3979: cec-client is broken with full KMS

plasticassius, 2020-11-29:
**Describe the bug**
When full KMS is enabled, cec-client doesn't function.

**To reproduce**
gpu_mem=128
dtparam=audio=on

[pi4]
dtoverlay=vc4-kms-v3d-pi4
max_framebuffers=2

**Expected behaviour**
$echo "pow 0" | sudo cec-client -s -d 1
opening a connection to the CEC adapter...
power status: standby

**Actual behaviour**
I believe all commands are broken. Here iraspinfo.logs one example:
$echo "pow 0" | sudo cec-client -s -d 1
opening a connection to the CEC adapter...
power status: unknown
Segmentation fault

**System**
[raspinfo.log](https://github.com/raspberrypi/linux/files/5612570/raspinfo.log)
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2\0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2

Linux Desktop 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
Revision	: c03112
Serial		: 10000000195824ae
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Oct 22 2020 13:59:27 
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       14593520 5286872   8681944  38% /
devtmpfs         1801484       0   1801484   0% /dev
tmpfs            1933580    8276   1925304   1% /dev/shm
tmpfs            1933580   17260   1916320   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1933580       0   1933580   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   55827    202269  22% /boot
tmpfs             386716       8    386708   1% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

[... 558 lines cut ...]
[   14.583526] Bluetooth: Core ver 2.22
[   14.583568] NET: Registered protocol family 31
[   14.583574] Bluetooth: HCI device and connection manager initialized
[   14.583593] Bluetooth: HCI socket layer initialized
[   14.583602] Bluetooth: L2CAP socket layer initialized
[   14.583616] Bluetooth: SCO socket layer initialized
[   14.589279] Bluetooth: HCI UART driver ver 2.3
[   14.589288] Bluetooth: HCI UART protocol H4 registered
[   14.589324] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.589477] Bluetooth: HCI UART protocol Broadcom registered
[   14.745988] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.745995] Bluetooth: BNEP filters: protocol multicast
[   14.746007] Bluetooth: BNEP socket layer initialized
[   22.079999] broken atomic modeset userspace detected, disabling atomic
[   28.397934] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.398589] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.399432] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.472748] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.473732] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.477091] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.477483] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   28.664239] Bluetooth: RFCOMM TTY layer initialized
[   28.664256] Bluetooth: RFCOMM socket layer initialized
[   28.664354] Bluetooth: RFCOMM ver 1.11
[   28.732326] fuse: init (API version 7.31)

popcornmix, 2020-11-29:
Actual behaviour
```
$echo "pow 0" | sudo cec-client -s -d 1
opening a connection to the CEC adapter...
power status: unknown
Segmentation fault
```
sounds like an issue for libcec.
You haven't said what the version of libcec or its origin is, but for kms support you'll probably need to build the latest version from source and enable the linux kernel api. e.g.
```
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 
```

plasticassius, 2020-11-29:
Yes, I am using the libcec and cec-client that came with the buster raspbian distribution. I'll look into building it myself.

closed 2020-11-29 by plasticassius

pelwell, 2020-11-29:
By the way, `dtoverlay=vc4-kms-v3d` is the preferred setting now - the `-pi4` is added automatically when appropriate.

6by9, 2020-11-29:
And you really need to be using 5.9 or 5.10 for full functionality on Pi4.
5.4 is known to have issues.

plasticassius, 2020-11-29:
> And you really need to be using 5.9 or 5.10 for full functionality on Pi4. 5.4 is known to have issues.

Thanks for the tips on libcec and kernel versions.

I thought I wouldn't need full functionality, mainly I'd like to get less tearing and frame loss. And, it does seem to work better than fake KMS even with 5.4. However, it's still not perfect, so perhaps I'll try the newer kernels, I presume I can get them with rpi-update.

I'm using chromium-browser to watch Netflix and Youtube. It works quite well, but still has some tearing and stuttering.

HiassofT, 2020-11-29:
FYI: instead of using the pulse-eight libcec library+utils you can also use cec-ctl from v4l-utils - that is the native linux tool, developed by the same people as the linux kernel CEC stack.

plasticassius, 2020-11-30:
> you can also use cec-ctl from v4l-utils

Thanks, works like a charm.