raspberrypi/linux #3981: RPi 4B, external HDD randomly freezes

githubharald, 2020-12-02:
**Describe the bug**

1. Problem description:
I'm having random problems with my external harddisk (HDD) connected to my Raspberry Pi 4B. When accessing the HDD with samba, or by doing a git push, the operation sometimes freezes on my computer or smartphone or smart-tv. When I then have a look at the HDD, I can see the white status LED blinking without stopping. I can ssh the RPi, but when I want to access the HDD (e.g. cd or ls on it), the ssh session also freezes. I also can not reboot using the reboot command. But as soon as I unplug the HDD, everything works fine again. After rebooting things usually work fine for quite some time again. Also keeping the HDD active (writing a file each minute) does not help, but it seems like the problem is not power-supply-related, because there are not more spin-ups needed because the HDD is never in standby mode.

2. What I tried so far:
At the beginning I directly attached the HDD to the USB3 port. It worked, but sometimes it made “click” noises, which might be caused by too little power. So I connected it to the USB2 port. Worked fine until after 1 month I got the issues described above, that samba was hanging. This happened multiple times, from different client devices. I then plugged a powered USB2 hub in between, and the problems disappeared for 2 month. I though it was solved, but this week I had the issue 2 times, first time when doing a git push, second time when I wanted to backup files from my computer to the samba share. I also added a cron job that writes a timestamp-file each minute to avoid the HDD to go to standby mode. But also with this it happened again after 5-6 days.

3. Hardware:

    Hard disk connected to powered USB 2.0 Hub, Hub connected to RPi USB2 port (now switched to USB3 port)
    External hard drive: TOSHIBA MQ01UBD100, 1TB, 3 partitions, all formatted as ext4
    Powered USB 2.0 Hub: Belkin 7 port hub (lsusb shows 3 hubs, nut sure which one it is)
    The part of the lsusb output showing something hub related:
        Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
        Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
        Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
    RPi: model 4B, 4GB RAM, Linux raspberrypi 5.4.72-v7l+, apt-get update and upgrade done just a few days ago
    Power supply: the “official” one
    Connected to network via LAN port

**To reproduce**
Details about my setup see above. Basically, it seems to be somehow random, so there is no "do action X to cause issue Y".

**Expected behaviour**
Run 24/7 without at some point having processes hang which access the HDD. 

**Actual behaviour**
Again, see above. Processes (like samba, git, or just some Python script writing to the HDD) accessing HDD hang. 

**System**

Output of raspinfo: [raspinfo.txt](https://github.com/raspberrypi/linux/files/5628135/raspinfo.txt)

* Which model of Raspberry Pi? Raspberry Pi 4 Model B Rev 1.2
* Which OS and version: Raspberry Pi reference 2020-08-20
* Which firmware version: Oct 22 2020 13:59:27 version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)
* Which kernel version: Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

**Logs**
dmesg only showing the OOM errors due to the hanging scripts which should keep the HDD active. Therefore the interesting information in the log seems to be already gone. 

Here are some parts of /var/log/messages which might have something to do with the problem: 

````
Nov 30 21:11:57 raspberrypi kernel: [433824.234047] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Nov 30 21:11:57 raspberrypi kernel: [433824.234060] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Nov 30 21:11:57 raspberrypi kernel: [433824.334485] usb 1-1.1.4.3: reset high-speed USB device number 5 using xhci_hcd
Nov 30 21:12:04 raspberrypi kernel: [433831.514247] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Nov 30 21:12:04 raspberrypi kernel: [433831.514261] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Nov 30 21:12:04 raspberrypi kernel: [433831.614585] usb 1-1.1.4.3: reset high-speed USB device number 5 using xhci_hcd
Nov 30 21:12:04 raspberrypi udisksd[393]: Error performing housekeeping for drive /org/freedesktop/UDisks2/drives/TOSHIBA_MQ01UBD100_1736TJIQT: Error updating SMART data: sk_disk_smart_read_data: Input/output error (udisks-error-quark, 0)
Nov 30 21:12:35 raspberrypi kernel: [433862.154573] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Nov 30 21:12:35 raspberrypi kernel: [433862.154586] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Nov 30 21:12:35 raspberrypi kernel: [433862.254858] usb 1-1.1.4.3: reset high-speed USB device number 5 using xhci_hcd
Nov 30 21:15:08 raspberrypi kernel: [434015.096131] python3 D 0 29096 29095 0x00000000
Nov 30 21:15:08 raspberrypi kernel: [434015.096151] Backtrace:
Nov 30 21:15:08 raspberrypi kernel: [434015.096184] [<c0a9ddb8>] (__schedule) from [<c0a9e634>] (schedule+0x5c/0xd0)
Nov 30 21:15:08 raspberrypi kernel: [434015.096205] r10:c1204ff0 r9:00000000 r8:00000002 r7:c0a9f07c r6:cc059ba0 r5:cc058000
Nov 30 21:15:08 raspberrypi kernel: [434015.096218] r4:d63a6ac0 
````

 And some time later, we get OOM errors because of all the processes that should write to the file on the disk but hang: 
````
     Dec 1 18:46:36 raspberrypi kernel: [511504.291042] systemd-journal invoked oom-killer: gfp_mask=0x400dc2(GFP_KERNEL_ACCOUNT|__GFP_HIGHMEM|__GFP_ZERO), order=0, oom_score_adj=0
    Dec 1 18:46:36 raspberrypi kernel: [511504.291052] CPU: 2 PID: 1821 Comm: systemd-journal Tainted: G C 5.4.72-v7l+ #1356
    Dec 1 18:46:36 raspberrypi kernel: [511504.291055] Hardware name: BCM2711
    Dec 1 18:46:36 raspberrypi kernel: [511504.291059] Backtrace:
    Dec 1 18:46:36 raspberrypi kernel: [511504.291072] [<c020e1c8>] (dump_backtrace) from [<c020e544>] (show_stack+0x20/0x24)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291078] r7:ffffffff r6:00000000 r5:60000113 r4:c12a1130
    Dec 1 18:46:36 raspberrypi kernel: [511504.291086] [<c020e524>] (show_stack) from [<c0a84cb0>] (dump_stack+0xd4/0x118)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291094] [<c0a84bdc>] (dump_stack) from [<c039250c>] (dump_header+0x64/0x1f8)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291099] r10:00400dc2 r9:ca54a000 r8:00000000 r7:00000558 r6:c0ddc7ac r5:d2aabd00
    Dec 1 18:46:36 raspberrypi kernel: [511504.291103] r4:ca54bdd8 r3:00000000
    Dec 1 18:46:36 raspberrypi kernel: [511504.291109] [<c03924a8>] (dump_header) from [<c03918a4>] (oom_kill_process+0x17c/0x188)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291113] r7:00000558 r6:c0ddc7ac r5:ca54bdd8 r4:d2aabd00
    Dec 1 18:46:36 raspberrypi kernel: [511504.291119] [<c0391728>] (oom_kill_process) from [<c039230c>] (out_of_memory+0x244/0x344)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291123] r7:c1204ff0 r6:c12083c0 r5:ca54bdd8 r4:d2aabd00
    Dec 1 18:46:36 raspberrypi kernel: [511504.291131] [<c03920c8>] (out_of_memory) from [<c03e1154>] (__alloc_pages_nodemask+0xbfc/0x1230)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291135] r6:c12a3550 r5:00001ca9 r4:00000000
    Dec 1 18:46:36 raspberrypi kernel: [511504.291142] [<c03e0558>] (__alloc_pages_nodemask) from [<c03c4260>] (__do_fault+0xb0/0x170)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291147] r10:00000055 r9:ca54be80 r8:00000000 r7:00000055 r6:00000000 r5:d5a303a8
    Dec 1 18:46:36 raspberrypi kernel: [511504.291150] r4:ca54be80
    Dec 1 18:46:36 raspberrypi kernel: [511504.291156] [<c03c41b0>] (__do_fault) from [<c03c816c>] (handle_mm_fault+0x98c/0xde4)
    Dec 1 18:46:36 raspberrypi kernel: [511504.291161] r7:00000055 r6:d5a303a8 r5:00000040 r4:00000001
    Dec 1 18:46:36 raspberrypi kernel: [511504.291167] [<c03c77e0>] (handle_mm_fault) from [<c0aa46c0>] (do_page_fault+0x148/0x38c) 
````

blkid still shows all three partitions.

umount /dev/sda1: target is busy (also when using -l), or the command hangs (had both situations).

Is there anything else I can check?

**Additional context**
Also discussed in RPi [Forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=292426&p=1771419#p1771419).

BigMuscle85, 2020-12-02:
I don't know if it is the same problem but we have old Raspberry Pi 2 that streams from IP camera and stores video to mounted SMB storage. With kernel 4.19, everything runs for weeks without any problem. After update to kernel 5.4, it runs for a while but after several hours the access to the remote storage (even simple "ls /mnt") freezes the Raspberry. Similar backtrace appears in kernel log. I don't have the exact backtrace now, because we downgraded back to kernel 4.19 and everything runs without problem again.

githubharald, 2020-12-02:
Thanks for the information. I just downgraded to 4.19.118, let's see if this changes anything. It might take days until the issue occurs, but I'll keep you updated.

popcornmix, 2020-12-03:
Possibly a duplicate of https://github.com/raspberrypi/linux/issues/3210
does arm_64bit=1 in config.txt make a difference?

githubharald, 2020-12-04:
> 
> 
> Possibly a duplicate of #3210
> does arm_64bit=1 in config.txt make a difference?

I still have the test running with the downgraded kernel version. If the issue occurs again I will try `arm_64bit=1`.

yoyojacky, 2020-12-10:
Could it be caused by the automatic hibernation of the hard disk, and the hard disk may enter a low power consumption mode without data collection for a long time? Similar to hard disk hibernation?

yoyojacky, 2020-12-10:
I am running Raspbian OS version 2020-08-19 and kernel version as following :
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```
and dmesg info:
```

pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.79-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:27:40 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1864704 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49804 r8192 d23924 u81920
[    0.000000] pcpu-alloc: s49804 r8192 d23924 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1360x768M@60 video=HDMI-A-2:640x480M@60 smsc95xx.macaddr=DC:A6:32:B0:81:7D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=69850c05-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x14c00000-0x18c00000] (64MB)
[    0.000000] Memory: 7800748K/8245248K available (10240K kernel code, 739K rwdata, 2816K rodata, 2048K init, 854K bss, 182356K reserved, 262144K cma-reserved, 7458816K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31111 entries in 61 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x518 with crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000023] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000100] bcm2835: system timer (irq = 17)
[    0.000752] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000767] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000782] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000794] Switching to timer-based delay loop, resolution 18ns
[    0.001037] Console: colour dummy device 80x30
[    0.001071] printk: console [tty1] enabled
[    0.001122] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001144] pid_max: default: 32768 minimum: 301
[    0.001292] LSM: Security Framework initializing
[    0.001484] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001508] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002876] Disabling memory control group subsystem
[    0.002998] CPU: Testing write buffer coherency: ok
[    0.003516] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004437] Setting up static identity map for 0x200000 - 0x20003c
[    0.004644] rcu: Hierarchical SRCU implementation.
[    0.005334] smp: Bringing up secondary CPUs ...
[    0.006556] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007908] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009187] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009344] smp: Brought up 1 node, 4 CPUs
[... 613 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

```
lsblk infor: 
```
 lsblk -t
NAME     ALIGNMENT MIN-IO OPT-IO PHY-SEC LOG-SEC ROTA SCHED    RQ-SIZE  RA WSAME
sda              0   4096      0    4096     512    1 mq-deadline
│                                                                   58 128   32M
└─sda1           0   4096      0    4096     512    1 mq-deadline
                                                                    58 128   32M
mmcblk0          0    512      0     512     512    0 mq-deadline
│                                                                  128 128    0B
├─mmcblk0p1
│                0    512      0     512     512    0 mq-deadline
│                                                                  128 128    0B
└─mmcblk0p2
                 0    512      0     512     512    0 mq-deadline
                                                                   128 128    0B
```

githubharald, 2020-12-10:
> 
> 
> Could it be caused by the automatic hibernation of the hard disk, and the hard disk may enter a low power consumption mode without data collection for a long time? Similar to hard disk hibernation?

As I wrote, I have a script which writes a file each minute, so that the disk does no go to sleep. But still the problem occurred.
Could you please move the dmesg output to a text file? This would keep this thread more readable.

P.S.: With the downgraded kernel, the problem did no occur until now.

BigMuscle85, 2020-12-11:
In our case, HDD does not hibernate either. Data is being written to it permanently. It is remote network HDD connected via SMB.
I tried to put the kernel 5.4 back. After several hours it got stuck. This is dmesg:

```
[   21.182264] FS-Cache: Netfs 'cifs' registered for caching
[   21.214956] Key type cifs.spnego registered
[   21.215212] Key type cifs.idmap registered
[   21.217919] CIFS: Attempting to mount //IP/Kamery
[   21.479274] CIFS VFS: Autodisabling the use of server inode numbers on new server.
[   21.479294] CIFS VFS: The server doesn't seem to support them properly or the files might be on different servers (DFS).
[   21.479304] CIFS VFS: Hardlinks will not be recognized on this mount. Consider mounting with the "noserverino" option to silence this message.
[36693.762505] CIFS VFS: \\IP sends on sock 0bcf5229 stuck for 15 seconds
[36693.762560] CIFS VFS: \\IP Error -11 sending data on socket to server
[36862.083309] INFO: task cifsd:509 blocked for more than 122 seconds.
[36862.083333]       Tainted: G         C        5.4.72-v7+ #1356
[36862.083343] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[36862.083354] cifsd           D    0   509      2 0x00000000
[36862.083372] Backtrace:
[36862.083407] [<809277c0>] (__schedule) from [<8092803c>] (schedule+0x5c/0xd0)
[36862.083426]  r10:7f3e8330 r9:00000000 r8:acedbd54 r7:00000002 r6:aceda000 r5:aceda000
[36862.083435]  r4:acef9e80
[36862.083453] [<80927fe0>] (schedule) from [<809284a8>] (schedule_preempt_disabled+0x18/0x1c)
[36862.083463]  r5:aceda000 r4:7f3e832c
[36862.083483] [<80928490>] (schedule_preempt_disabled) from [<8092a7fc>] (__mutex_lock.constprop.0+0x2c4/0x55c)
[36862.083500] [<8092a538>] (__mutex_lock.constprop.0) from [<8092abb0>] (__mutex_lock_slowpath+0x1c/0x20)
[36862.083515]  r10:00000012 r9:7f3c67ec r8:aeb1fe00 r7:a9d93800 r6:aeb1e100 r5:7f3e82a8
[36862.083524]  r4:7f3e832c
[36862.083539] [<8092ab94>] (__mutex_lock_slowpath) from [<8092ac10>] (mutex_lock+0x5c/0x60)
[36862.084010] [<8092abb4>] (mutex_lock) from [<7f3b77c8>] (dfs_cache_update_vol+0x38/0xb0 [cifs])
[36862.084025]  r5:7f3e82a8 r4:a9d93800
[36862.084436] [<7f3b7790>] (dfs_cache_update_vol [cifs]) from [<7f36e11c>] (cifs_reconnect+0x5cc/0x950 [cifs])
[36862.084456]  r9:7f3c67ec r8:aeb1fe00 r7:a9d93974 r6:00000000 r5:00000000 r4:a9d93800
[36862.084835] [<7f36db50>] (cifs_reconnect [cifs]) from [<7f36e69c>] (cifs_readv_from_socket+0x1fc/0x244 [cifs])
[36862.084853]  r10:aceda000 r9:80e03d00 r8:80ed7a4c r7:ffffff99 r6:acedbe9c r5:a9d93970
[36862.084861]  r4:a9d93800
[36862.085239] [<7f36e4a0>] (cifs_readv_from_socket [cifs]) from [<7f36e748>] (cifs_read_from_socket+0x64/0x8c [cifs])
[36862.085255]  r10:80ed7a4c r9:a9d9393c r8:aceda000 r7:a9d93800 r6:00000023 r5:a9d67c00
[36862.085264]  r4:a9d93800 r3:00000000
[36862.085641] [<7f36e6e4>] (cifs_read_from_socket [cifs]) from [<7f36eb68>] (cifs_demultiplex_thread+0x124/0xa48 [cifs])
[36862.085650]  r4:00000000
[36862.085851] [<7f36ea44>] (cifs_demultiplex_thread [cifs]) from [<801449cc>] (kthread+0x144/0x170)
[36862.085867]  r10:acd2fd1c r9:aeb1e19c r8:a9d93800 r7:aceda000 r6:00000000 r5:aeb1e0c0
[36862.085875]  r4:aeb1e180
[36862.085891] [<80144888>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[36862.085901] Exception stack(0xacedbfb0 to 0xacedbff8)
[36862.085913] bfa0:                                     00000000 00000000 00000000 00000000
[36862.085927] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[36862.085940] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[36862.085954]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80144888
[36862.085962]  r4:aeb1e0c0
[36862.085982] INFO: task kworker/1:0:14486 blocked for more than 122 seconds.
[36862.085992]       Tainted: G         C        5.4.72-v7+ #1356
[36862.085999] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[36862.086008] kworker/1:0     D    0 14486      2 0x00000000
[36862.086217] Workqueue: cifsiod cifs_uncached_writev_complete [cifs]
[36862.086229] Backtrace:
[36862.086250] [<809277c0>] (__schedule) from [<8092803c>] (schedule+0x5c/0xd0)
[36862.086265]  r10:b0ec8018 r9:00000000 r8:a9f79c5c r7:00000002 r6:a9f78000 r5:a9f78000
[36862.086274]  r4:9946bd00
[36862.086290] [<80927fe0>] (schedule) from [<809284a8>] (schedule_preempt_disabled+0x18/0x1c)
[36862.086301]  r5:a9f78000 r4:b0ec8014
[36862.086317] [<80928490>] (schedule_preempt_disabled) from [<8092a7fc>] (__mutex_lock.constprop.0+0x2c4/0x55c)
[36862.086335] [<8092a538>] (__mutex_lock.constprop.0) from [<8092abb0>] (__mutex_lock_slowpath+0x1c/0x20)
[36862.086350]  r10:b0ec8014 r9:7f181000 r8:00000001 r7:b0ec8000 r6:b2b86000 r5:0000002f
[36862.086359]  r4:b0ec8014
[36862.086374] [<8092ab94>] (__mutex_lock_slowpath) from [<8092ac10>] (mutex_lock+0x5c/0x60)
[36862.086574] [<8092abb4>] (mutex_lock) from [<7f3618c8>] (cifs_reconnect_tcon+0x9c/0x5a0 [cifs])
[36862.086585]  r5:0000002f r4:a9d93800
[36862.086961] [<7f36182c>] (cifs_reconnect_tcon [cifs]) from [<7f361dfc>] (small_smb_init+0x30/0x84 [cifs])
[36862.086977]  r10:00000010 r9:a9d93800 r8:0000000e r7:a9f79d7c r6:0000002f r5:b2b86000
[... 8 lines cut ...]
[36862.090491] [<7f37b610>] (cifs_uncached_writev_complete [cifs]) from [<8013d1ac>] (process_one_work+0x214/0x538)
[36862.090517]  r10:00000000 r9:00000040 r8:00000000 r7:ff7c6e00 r6:b37b62c0 r5:a9c32b80
[36862.090526]  r4:a68408a0
[... cut ...]
[36862.090742]  r4:a6bee740
[36862.090763] INFO: task kworker/1:1:14597 blocked for more than 122 seconds.
[36862.090775]       Tainted: G         C        5.4.72-v7+ #1356
[36862.090783] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[36862.090791] kworker/1:1     D    0 14597      2 0x00000000
[36862.091092] Workqueue: cifsiod cifs_uncached_writev_complete [cifs]
[36862.091108] Backtrace:
[36862.091137] [<809277c0>] (__schedule) from [<8092803c>] (schedule+0x5c/0xd0)
...
```

githubharald, 2021-01-06:
FYI: after downgrading the kernel to 4.19 there were no more problems with the HDD so far.
Any comments from the RPi kernel team on this?

debuti, 2021-03-19:
Same issue here. 
```
$ dmesg | grep -i xhci
[    1.484040] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.484077] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    1.487297] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0000030000000890
[    1.488633] usb usb1: Product: xHCI Host Controller
[    1.488650] usb usb1: Manufacturer: Linux 5.10.17-v7l+ xhci-hcd
[    1.490203] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.490243] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    1.490269] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    1.490806] usb usb2: Product: xHCI Host Controller
[    1.490823] usb usb2: Manufacturer: Linux 5.10.17-v7l+ xhci-hcd
[    1.853064] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    2.373077] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[    2.773152] usb 1-1.4: new high-speed USB device number 4 using xhci_hcd
[    7.143092] usb 1-1.4.1: new high-speed USB device number 5 using xhci_hcd
[    7.513159] usb 1-1.4.2: new high-speed USB device number 6 using xhci_hcd
[    7.813079] usb 1-1.4.3: new high-speed USB device number 7 using xhci_hcd
[    8.093820] usb 1-1.4.4: new high-speed USB device number 8 using xhci_hcd
[38771.059433] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[38771.059449] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[38771.169873] usb 1-1.4.1: reset high-speed USB device number 5 using xhci_hcd
[38801.779938] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[38801.779954] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[38801.890222] usb 1-1.4.1: reset high-speed USB device number 5 using xhci_hcd
[38832.500293] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[38832.500309] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[38832.620588] usb 1-1.4.1: reset high-speed USB device number 5 using xhci_hcd
```

@githubharald how did you downgraded your pi to 4.x kernel?

Thanks

githubharald, 2021-03-19:
`sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2 `

AaronFaltesek, 2021-03-20:
I believe I'm having this same issue. I'll try downgrading as well. It typically loses access to external HDD after 1-3 days of uptime. I'm writing data to it ~every 10 minutes.

```Mar 20 00:50:46 raspberrypi kernel: [63618.625860] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Mar 20 00:50:46 raspberrypi kernel: [63618.625874] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Mar 20 00:50:46 raspberrypi kernel: [63618.726207] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Mar 20 00:51:17 raspberrypi kernel: [63649.025975] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Mar 20 00:51:17 raspberrypi kernel: [63649.025988] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Mar 20 00:51:17 raspberrypi kernel: [63649.126248] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Mar 20 00:51:17 raspberrypi kernel: [63649.257638] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=0x00
Mar 20 00:51:17 raspberrypi kernel: [63649.257661] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 2e 0f be 38 00 00 10 00
Mar 20 00:51:24 raspberrypi kernel: [63656.305916] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Mar 20 00:51:24 raspberrypi kernel: [63656.305930] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Mar 20 00:51:24 raspberrypi kernel: [63656.406243] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Mar 20 00:51:24 raspberrypi udisksd[364]: Error performing housekeeping for drive /org/freedesktop/UDisks2/drives/TOSHIBA_MQ04UBD200_Y04FT0PYT: Error updating SMART data: sk_disk_smart_read_data: Operation not supported (udisks-error-quark, 0)```

I will try the downgrading from 5.4.79 to 4.19 using your command.

@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

githubharald, 2021-03-23:
@debuti 
@AaronFaltesek 
Would be interesting to hear from you if downgrading also solved the problem.

debuti, 2021-03-24:
So far the downgrading solves the problem yes, its been 3 days without issues (while usually in 1-2 days it was failing)

AaronFaltesek, 2021-03-25:
After 5 days without issue it's safe to say this has solved my issue.

githubharald, 2021-03-25:
would be cool if someone from the RPi team could comment on this. It seems that some bug was introduced in the new kernel.
I guess more people will be confronted with this issue as soon as they upgrade. Many people use the RPi as NAS, so even if it takes 2 or 3 days until the HDD hangs, one will notice it at some point.

Are all of you with the problem also using a Toshiba ext HDD (at least from the logs AaronFaltesek seems to also have one).

bretonium, 2021-03-25:
No, i have WD or Seagate (not sure which one fails)

t-paul, 2021-03-25:
Same issue with 2 WD disks running as raid on the USB3 connectors:
```
Bus 002 Device 003: ID 1058:25a3 Western Digital Technologies, Inc. Elements Desktop (WDBWLG)
Bus 002 Device 002: ID 1058:25a3 Western Digital Technologies, Inc. Elements Desktop (WDBWLG)
```
```
# cat /proc/mdstat 
Personalities : [raid1] 
md127 : active raid1 sdb[0] sda[1]
      3906853888 blocks super 1.2 [2/2] [UU]
      bitmap: 0/30 pages [0KB], 65536KB chunk
```
This does not seem related to #3210 as mentioned above, I'm running on a 4MB PI headless and there's no OOM in the logs.

bassistguy, 2021-03-30:
I just wanted to reply and state that downgrading to 4.19.118-v7l+ fixed my issue. I was having this same issue for MONTHS where the connection to my external (powered) USB hard drive would just randomly freeze (within a day or 2 of rebooting my Pi). It would still show as mounted, but I couldn't' access it and the drive light would constantly blink. I would have to reboot my Pi 4 B in order for the drive to be accessible again for a day or so. I downgraded to 4.19.118-v7l+ via the command githubharald posted. That was 4 days ago and the drive is still connected.Thank you everyone for this post!! It was driving me nuts.

Has anyone tried to contact the Raspberry Pi folks to let them know? I'm assuming they have to know about this, as it's been going on for a number of months now.

sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2

a123xxsp, 2021-03-30:
I am having the same issues (external USB HDD stop responding randomly, usually after a day uptime)  :
 uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

I will try to downgrade to 4.19 and let you know , I am just curious why I am on version 5.1 and everyone else is on 5.4 (I don't get 5.4 via apt upgrade)

jimbasilio, 2021-03-31:
This isn't just raspberrypi, I have this issue on my mythbuntu installation with an external drive. Kernel is 5.4.

Trying to figure out how to downgrade to 4.19. I was going to upgrade to 5.10 but it seems that has an issue too.

Frustrating!

a123xxsp, 2021-03-31:
> This isn't just raspberrypi, I have this issue on my mythbuntu installation with an external drive. Kernel is 5.4. 
> Trying to figure out how to downgrade to 4.19. I was going to upgrade to 5.10 but it seems that has an issue too.
> Frustrating!
Thanks for the notice, for now, I bypassed the problem by rebooting myPi every day with a CRON job, but I am also testing 4.19 with another microSD and let you know.

PetrozPL, 2021-03-31:
the same issue on my RPI4

Linux rpi4tvhead 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

The USB 3.0 drive is : 

Bus 002 Device 002: ID 1058:1230 Western Digital Technologies, Inc. My Book (WDBFJK)

The same problem was on other HDD i've previously tested. The kernel log as below:

`[819629.648204] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[819629.648220] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[819629.799065] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[819636.858297] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[819636.858313] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[819637.009119] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[820229.583028] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[820229.583045] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[820229.733735] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[820829.657287] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[820829.657303] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[820829.808092] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[821429.661124] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[821429.661140] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[821429.811913] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[822029.664256] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[822029.664272] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[822029.814936] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[822629.667958] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[822629.667974] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[822629.818665] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[823229.631802] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[823229.631818] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[823229.782543] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[823829.606896] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[823829.606911] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[823829.757699] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[824429.681450] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[824429.681466] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[824429.832298] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[825029.685789] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[825029.685805] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[825029.836537] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[825629.679042] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[825629.679057] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[825629.829764] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[826229.621928] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[826229.621944] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[826229.772715] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[826829.625498] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[826829.625513] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[826829.776227] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[827429.699264] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[827429.699280] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[827429.849932] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[827495.104002] cxd2880: cxd2880_set_frontend: sys:3 freq:586000000 bw:8
[827496.136567] cxd2880: cxd2880_set_frontend: tune result 0
[827555.499900] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[827555.499916] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[827555.650710] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[827555.682449] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=0x00 cmd_age=30s
[827555.682476] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 03 1c 43 93 00 00 02 00
[827555.682502] blk_update_request: I/O error, dev sda, sector 417471640 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
[827586.220071] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[827586.220087] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[827586.370843] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[827723.160680] INFO: task usb-storage:93 blocked for more than 122 seconds.
[827723.160702]       Tainted: G         C        5.10.17-v7l+ #1403
[827723.160717] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[827723.160734] task:usb-storage     state:D stack:    0 pid:   93 ppid:     2 flags:0x00000000
[... 338 lines cut ...]
[828214.685682] [<c0438a54>] (sys_openat) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
[828214.685699] Exception stack(0xcfa2bfa8 to 0xcfa2bff0)
[828214.685718] bfa0:                   00000000 0007b098 ffffff9c 0185f2a0 00020800 00000000
[828214.685739] bfc0: 00000000 0007b098 00000000 00000142 0007b098 b57fec10 b57feb6c b57feb6c
[828214.685756] bfe0: 00000002 b57feab8 00000000 b6990da0`

When I try to downgrade kernel with 
[... cut ...]
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 5.10.17-v7l+
Partition size 255M may not be sufficient for new Pi4 files
This could result in a system that will not boot.
256M FAT partition is recommended. Ensure you have a backup if continuing.
Would you like to proceed? (y/N)

Is it safe to proceed?

pelwell, 2021-03-31:
Provided you've got about 60MB free in your boot partition you should be fine. It's disappointing that the creator of the distribution didn't make it 1MB larger just to avoid that warning.

PetrozPL, 2021-03-31:
Both HDDs were externally powered. The USB <-> SATA adaptor is ASM1151, but with dedicated WD firmware programmed. That's why it's recognized as : 

Bus 002 Device 002: ID 1058:1230 Western Digital Technologies, Inc. My Book (WDBFJK)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x1058 Western Digital Technologies, Inc.
  idProduct          0x1230 My Book (WDBFJK)
  bcdDevice           10.65
  iManufacturer           2 Western Digital
  iProduct                3 My Book 1230
  iSerial                 1 574343344E36444356545039
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us
    Deep BESL value    61440 us
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled

Previous HDD I've tested was using the same chip, but with cleared EEPROM, so It wasn't recognized as WD anymore. 

The same configuration was working great on RPI2. I've switched to RPI4 few months ago, for better wr/rd speeds over USB. And all this time I've struggled with this random USB crashes every 1-3 days. Few days ago I've switched to other HDD and the problem still occures.

pelwell, 2021-03-31:
And which power supply is the Pi 4 using?

PetrozPL, 2021-03-31:
> And which power supply is the Pi 4 using?

It uses 5V 3A dedicated RPI4 power supply, same as this one: 

[https://kamami.pl/zasilacze/575564-zasilacz-5v3a-usb-c-do-raspberry-pi-4-czarny.html](url)

githubharald, 2021-03-31:
I tested using an external powered USB hub, but this did not make a difference.
So it is really just the kernel and not something related to power.

BigMuscle85, 2021-03-31:
The problem is not related to USB hub nor power supply, because it happens also for SMB drives connected over Ethernet and even for any mounted drive in the network.

PetrozPL, 2021-03-31:
if kernel downgrade is working, then it's strictly software related problem. I've successfully downgraded to : 

Linux rpi4tvhead 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

I'll report if it helps in my case in 4-7 days or so. 

Thanks and regards!

hermanhobnob, 2021-03-31:
My Pi NAS was perfectly stable until about a couple of months ago but not any more, so I suspect I'm seeing the same problem. How did you downgrade @PetrozPL? I don't see that kernel version available via apt anymore.

Artnal, 2021-04-01:
I have the same issue described in this thread, I just downgraded to 4.19 and will report back if the problem still arises.

@hermanhobnob `sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` still works to downgrade to 4.19.118.

PetrozPL, 2021-04-01:
> My Pi NAS was perfectly stable until about a couple of months ago but not any more, so I suspect I'm seeing the same problem. How did you downgrade @PetrozPL? I don't see that kernel version available via apt anymore.

as Artnal wrote, I've also used 
` sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2 `
and it worked

Artnal, 2021-04-01:
> 
> 
> I have the same issue described in this thread, I just downgraded to 4.19 and will report back if the problem still arises.
> 
> @hermanhobnob `sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` still works to downgrade to 4.19.118.

As a side-question to this solution, would anyone know how to downgrade the appropriate raspberrypi-kernel-headers as well? The downgrade of the kernel version but not the headers broke my wireguard install, and 
`sudo apt-get --reinstall install raspberrypi-kernel-headers` 
result in an error message, since it attempts to download the latest kernel headers (1.202103003-1).

Thanks!

alpha9191, 2021-04-02:
Hello guys.
That downgrade worked for me fine to solve the problem.
In addition the write performance increased was instantly.
Best regards

Artnal, 2021-04-05:
As a short update, that definitely seems to have solved the issue. It did not appear since I downgraded. However, when I tried to go back to the latest RC (5.10), the issue immediately reappeared, with both the access to the drive hanging and, upon reboot, the SSH connection refused. 

I went back to the 4.19 version, even though I still haven't managed to fix the kernel headers issue.

a123xxsp, 2021-04-05:
How do you reverted it back to 5.1 ?   (I am currently with 4.19)
I tried apt-get upgrade but no update appeared 
> As a short update, that definitely seems to have solved the issue. It did not appear since I downgraded. However, when I tried to go back to the latest RC (5.10), the issue immediately reappeared, [...]

Artnal, 2021-04-05:
> 
> 
> How do you reverted it back to 5.1 ? (I am currently with 4.19)
> I tried apt-get upgrade but no update appeared
> 

Running 
`sudo apt-get update`
`sudo apt install --reinstall libraspberrypi0 libraspberrypi-{bin,dev,doc} raspberrypi-bootloader raspberrypi-kernel`
and a reboot put me back on the latest release version. 

From this [bug](https://bugzilla.kernel.org/show_bug.cgi?id=202541), I understand that there have been some substantial changes to the xhci drivers for linux kernel v5.12, which might solve the issue. However, the latest official RPI release is still 5.10, and these drivers don't seem to be updated yet (the error message should change if they were updated). I haven't found how to change to the 5.12 release channel, so I'm still on 4.19.

a123xxsp, 2021-04-06:
I can confirm that with kernel 4.19 my Rasp-pi 4  (with a 3.5 HDD attached in USB3) works like a charm,
and I suppose we'll have to wait for the next kernel (5.12) to check if the problem is fixed.
Until then, What do you think about a shell script that frequently checks if a file is readable (to check if USB HDD is accessible) and if not reboot the pi, its kind of dangerous to enter an endless loop but with some delay in starting it could be a temporary solution for someone who really can't rollback to 4.19

popcornmix, 2021-04-06:
@Artnal we don't build and distribute the 5.12 kernel, but you can build from [source tree](https://github.com/raspberrypi/linux/tree/rpi-5.12.y) following these instructions: https://www.raspberrypi.org/documentation/linux/kernel/building.md

a123xxsp, 2021-04-06:
> 
> 
> @Artnal we don't build and distribute the 5.12 kernel, but you can build from [source tree](https://github.com/raspberrypi/linux/tree/rpi-5.12.y) following these instructions: https://www.raspberrypi.org/documentation/linux/kernel/building.md

yes, but eventually ... in the near future (in 2021 Q2-Q3) it will be a next kernel compiled/checked kernel for apt upgrade, correct?

bassistguy, 2021-04-08:
For anyone new looking at this post, just an FYI, it's been 9 days since I downgraded to 4.19 and external hard drive has had no disconnects.

maszaa, 2021-04-11:
I updated raspberrypi-kernel from `1.20210201-1 ` to `1.20210303-1` on 1st April and the problem started for me. Strangely it only affects one WD Elements... I have four of them via USB 3.0 hubs and the affected one is newer because the previous one started to have problems reported by SMART.

dmesg:
```
[Apr11 17:50] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[  +0.000007] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[  +0.100534] usb 2-1.2: reset SuperSpeed Gen 1 USB device number 4 using xhci_hcd
[ +30.620188] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[  +0.000007] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[  +0.100479] usb 2-1.2: reset SuperSpeed Gen 1 USB device number 4 using xhci_hcd
[  +0.032586] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=0x00 cmd_age=30s
[  +0.000012] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 46 08 00 00 00 08 00 00
[  +0.000011] blk_update_request: I/O error, dev sda, sector 17928 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0

[Apr11 17:55] INFO: task scsi_eh_0:107 blocked for more than 122 seconds.
[  +0.000021]       Tainted: G         C        5.10.17-v7l+ #1403
[  +0.000016] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  +0.000018] task:scsi_eh_0       state:D stack:    0 pid:  107 ppid:     2 flags:0x00000000
[  +0.000033] Backtrace:
[  +0.000050] [<c0b6d050>] (__schedule) from [<c0b6da28>] (schedule+0x68/0xe4)
[  +0.000024]  r10:00000000 r9:00000002 r8:c1205048 r7:c2e82000 r6:c2e83e44 r5:c22c0000
[  +0.000016]  r4:ffffe000
[  +0.000027] [<c0b6d9c0>] (schedule) from [<c0b6de70>] (schedule_preempt_disabled+0x18/0x1c)
[  +0.000017]  r5:ffffe000 r4:c22bbd40
[  +0.000027] [<c0b6de58>] (schedule_preempt_disabled) from [<c0b700f8>] (__mutex_lock.constprop.9+0x344/0x58c)
[  +0.000027] [<c0b6fdb4>] (__mutex_lock.constprop.9) from [<c0b7045c>] (__mutex_lock_slowpath+0x1c/0x20)
[  +0.000021]  r10:c1203d00 r9:c22bb800 r8:c1203d00 r7:c1203d00 r6:c22bd800 r5:c22bb800
[  +0.000015]  r4:c22bbd40
[  +0.000025] [<c0b70440>] (__mutex_lock_slowpath) from [<c0b704bc>] (mutex_lock+0x5c/0x60)
[  +0.000026] [<c0b70460>] (mutex_lock) from [<c0938ec0>] (device_reset+0x28/0x54)
[  +0.000017]  r5:c22bb800 r4:c22bbd40
[  +0.000025] [<c0938e98>] (device_reset) from [<c087d604>] (scsi_eh_ready_devs+0x4ec/0x8fc)
[  +0.000017]  r5:c41ae0a8 r4:c2e83f34
[  +0.000026] [<c087d118>] (scsi_eh_ready_devs) from [<c087e978>] (scsi_error_handler+0x3d8/0x3e0)
[  +0.000020]  r10:c087c2f4 r9:c22bb974 r8:c1205048 r7:c2e83f34 r6:ffffe000 r5:c2e83f3c
[  +0.000016]  r4:c22bb800
[  +0.000026] [<c087e5a0>] (scsi_error_handler) from [<c0245a50>] (kthread+0x170/0x174)
[  +0.000019]  r10:c22e9944 r9:c22bb800 r8:c087e5a0 r7:c2e82000 r6:00000000 r5:c2e3a800
[  +0.000016]  r4:c2e3a840
[  +0.000022] [<c02458e0>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[  +0.000017] Exception stack(0xc2e83fb0 to 0xc2e83ff8)
[  +0.000018] 3fa0:                                     00000000 00000000 00000000 00000000
[  +0.000020] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0.000018] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0.000020]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02458e0
[  +0.000015]  r4:c2e3a800

[  +0.000018] INFO: task usb-storage:109 blocked for more than 122 seconds.
[  +0.000017]       Tainted: G         C        5.10.17-v7l+ #1403
[  +0.000014] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  +0.000016] task:usb-storage     state:D stack:    0 pid:  109 ppid:     2 flags:0x00000000
[  +0.000031] Backtrace:
[  +0.000037] [<c0b6d050>] (__schedule) from [<c0b6da28>] (schedule+0x68/0xe4)
[  +0.000022]  r10:0000001f r9:00000002 r8:7fffffff r7:c1205048 r6:c22bbdf0 r5:c22c1f00
[  +0.000015]  r4:ffffe000
[  +0.000025] [<c0b6d9c0>] (schedule) from [<c0b71eac>] (schedule_timeout+0x2e8/0x384)
[  +0.000017]  r5:c1205048 r4:c22bbdec
[  +0.000025] [<c0b71bc4>] (schedule_timeout) from [<c0b6efdc>] (wait_for_completion+0xb8/0x138)
[  +0.000020]  r9:00000002 r8:7fffffff r7:ffffe000 r6:c22bbdf0 r5:c1205048 r4:c22bbdec
[  +0.000024] [<c0b6ef24>] (wait_for_completion) from [<c08dd608>] (usb_sg_wait+0x168/0x190)
[  +0.000021]  r9:00001000 r8:00000000 r7:00000000 r6:c22bbdec r5:00000001 r4:c22bbdcc
[  +0.000022] [<c08dd4a0>] (usb_sg_wait) from [<c0939a88>] (usb_stor_bulk_transfer_sglist.part.2+0x80/0xdc)
[  +0.000020]  r9:00001000 r8:c22bbd64 r7:00001000 r6:c0008480 r5:c22bbdcc r4:c22bbd40
[  +0.000023] [<c0939a08>] (usb_stor_bulk_transfer_sglist.part.2) from [<c0939b3c>] (usb_stor_bulk_srb+0x58/0x84)
[  +0.000019]  r8:c1205048 r7:00000000 r6:c41ae0a8 r5:c41ae0a8 r4:c1205048
[  +0.000022] [<c0939ae4>] (usb_stor_bulk_srb) from [<c0939c90>] (usb_stor_Bulk_transport+0x128/0x384)
[  +0.000017]  r5:ded53000 r4:c22bbd40
[  +0.000022] [<c0939b68>] (usb_stor_Bulk_transport) from [<c093a5ac>] (usb_stor_invoke_transport+0x40/0x4c4)
[  +0.000019]  r10:c22e997c r9:c22bbd40 r8:00000000 r7:c1205048 r6:c1205048 r5:c41ae0a8
[  +0.000016]  r4:c22bbd40
[... 7 lines cut ...]
[  +0.000016] Exception stack(0xc2da9fb0 to 0xc2da9ff8)
[  +0.000018] 9fa0:                                     00000000 00000000 00000000 00000000
[  +0.000019] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0.000018] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[... cut ...]
Unplugging USB or power of the affected HDD and re-plugging it does nothing.
Nothing appears in dmesg.

The problem can't be due to HDD going sleep as I have a cron job running dd every 5 minutes to keep it spinning.

Would be nice to have previous install candidates on package repository also:
```
sudo apt-cache policy raspberrypi-kernel
raspberrypi-kernel:
  Installed: 1.20210303-1
  Candidate: 1.20210303-1
  Version table:
 *** 1.20210303-1 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
```

jdtsmith, 2021-04-11:
Same issue here for several months with a WD Elements Desktop (WDBWLG) 10TB on 5.4.83-v7l+.  Trying the 4.19 downgrade.

PetrozPL, 2021-04-12:
I'm stopping by, to report that downgrade in my case resolved the issue (system is up for 11 days now, previous crash on new HDD was after ~9 days). I will try to switch once more to old HDD, where crashes were every 1-3 days. Unfortunately, downgrading to 4.19 result in crashes of cxd2880 (tv hat). Although, after every crash system fully recovers by itself (tvheadend is also still fully functional). So Its definitely better than HDD crashes that require system reboot and HDD re-plug.

maszaa, 2021-04-12:
I'm trying arm_64bit=1 in config.txt at the moment. If it doesn't help I'll start to use the backup I made before 1st April.

Annoying issue, please address this.

maszaa, 2021-04-16:
arm_64bit=1 in config.txt for the fifth day now. No problems so far. Not a single USB related error in dmesg.

a123xxsp, 2021-04-16:
> 
> 
> arm_64bit=1 in config.txt for the fifth day now. No problems so far. Not a single USB related error in dmesg.
which kernel?  (4.19 - 5.11 etc)
Currently I am on 4.19 for about a week with no problems and high USB3 speeds

AnthonyGiorgio, 2021-04-16:
I downgraded to 4.19 as well, and I haven't seen the USB error in almost 48 hours of uptime.  Previously I was lucky if the Pi was stable for 24 hours.

Artnal, 2021-04-16:
> 
> 
> arm_64bit=1 in config.txt for the fifth day now. No problems so far. Not a single USB related error in dmesg.

Thanks for the update, same question as @a123xxsp , is it on 4.19 or 5.10?

Out of curiosity, which config.txt did you edit? Thanks

maszaa, 2021-04-16:
```
$ uname -a
Linux RasPiNAS 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
```

I edited `/boot/config.txt` and rebooted.

wangnick, 2021-04-22:
Same issue here on RPI 4B 4G with a HDD in an externally powered USB3 case (note `usb-storage` not `uas`) used as NAS on `/mnt/ext4-14t`:
```
[    3.129542] scsi host0: usb-storage 2-1:1.0
[    4.155773] scsi 0:0:0:0: Direct-Access     ASMT     2115             0    PQ: 0 ANSI: 6
```
```
[204455.505479] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[204455.505494] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[204455.656350] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[204470.865917] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[204470.865932] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[204471.016595] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
```
```
[219762.720778] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[219762.720793] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[219762.871516] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[219957.904809] INFO: task scsi_eh_0:86 blocked for more than 122 seconds.
[219957.904833]       Tainted: G         C        5.10.23-v7l+ #1406
```
```
Bus 002 Device 002: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
If I then unplug the HDD:
```
Apr 22 11:03:14 database kernel: [240633.864693] usb 2-1: USB disconnect, device number 2
Apr 22 11:03:14 database kernel: [240633.865294] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
Apr 22 11:03:14 database kernel: [240633.865306] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
Apr 22 11:03:14 database kernel: [240633.865478] sd 0:0:0:0: Device offlined - not ready after error recovery
Apr 22 11:03:15 database kernel: [240634.244305] sd 0:0:0:0: [sdb] Synchronizing SCSI cache
Apr 22 11:03:15 database kernel: [240634.244392] sd 0:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
```
When I then re-plug the HDD my root/boot disk (NVMe SDD also on USB3) is lost:
```
Apr 22 11:03:22 database kernel: [240641.194300] usb 2-2: USB disconnect, device number 3
Apr 22 11:03:22 database kernel: [240641.200169] print_req_error: 180 callbacks suppressed
Apr 22 11:03:22 database kernel: [240641.200277] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019347 starting block 3540039)
Apr 22 11:03:22 database kernel: [240641.200336] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019347 starting block 3540043)
Apr 22 11:03:22 database kernel: [240641.200398] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019421 starting block 3541015)
Apr 22 11:03:22 database kernel: [240641.200448] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019421 starting block 3541016)
Apr 22 11:03:22 database kernel: [240641.200505] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019428 starting block 3541032)
Apr 22 11:03:22 database kernel: [240641.200578] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019427 starting block 3311591)
Apr 22 11:03:22 database kernel: [240641.200627] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019427 starting block 3311594)
Apr 22 11:03:22 database kernel: [240641.200678] EXT4-fs warning (device sda2): ext4_end_bio:349: I/O error 10 writing to inode 3019391 starting block 3344549)
Apr 22 11:03:22 database kernel: [240641.224925] buffer_io_error: 59 callbacks suppressed
```

Very strange ...

carlonluca, 2021-04-27:
I'm also experiencing the same issue recently:

```
[84653.896302] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[84653.896312] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[84654.024946] usb 2-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[86454.106692] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[86454.106703] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[86454.235482] usb 2-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[88254.316788] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[88254.316798] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[88254.445543] usb 2-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[90054.530698] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[90054.530705] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[90054.659325] usb 2-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[90150.783661] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
[90150.783673] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
[90150.912451] usb 2-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[90263.413181] INFO: task fdisk:334979 blocked for more than 120 seconds.
[90263.419916]       Tainted: G         C  E     5.8.0-1016-raspi #19-Ubuntu
[90263.426887] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[90263.434908] fdisk           D    0 334979 334978 0x00000001
[90263.434918] Call trace:
[90263.434932]  __switch_to+0xf8/0x12c
[90263.434940]  __schedule+0x2a8/0x72c
[90263.434945]  schedule+0x84/0x120
[90263.434949]  io_schedule+0x48/0x6c
[90263.434956]  wait_on_page_bit_common+0x11c/0x334
[90263.434962]  do_read_cache_page+0x234/0x474
[90263.434967]  read_cache_page+0x20/0x2c
[90263.434974]  scsi_bios_ptable+0x38/0x114
[90263.434979]  scsi_partsize+0x2c/0x14c
[90263.434983]  scsicam_bios_param+0x28/0x1a0
[90263.434989]  sd_getgeo+0xdc/0xe4
[90263.434994]  blkdev_ioctl+0x194/0x2d0
[90263.435000]  block_ioctl+0x44/0x54
[90263.435006]  ksys_ioctl+0xb4/0xf4
[90263.435011]  __arm64_sys_ioctl+0x2c/0xbc
[90263.435016]  el0_svc_common.constprop.0+0x84/0x200
[90263.435021]  do_el0_svc+0x30/0xa0
[90263.435026]  el0_svc+0x18/0x50
[90263.435031]  el0_sync_handler+0x98/0x354
[90263.435036]  el0_sync+0x17c/0x180
```

Is this issue the same in https://bugzilla.kernel.org/show_bug.cgi?id=202541?
I'm on Ubuntu 64 bit:

`Linux pihome 5.8.0-1016-raspi #19-Ubuntu SMP PREEMPT Tue Feb 9 20:12:43 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`

Anyone who tested on 5.12? Unfortunately I'm not sure if there is a way in Ubuntu to go back to 4.19.

pelwell, 2021-04-27:
That bugzilla thread includes a link to an interesting patch: https://bugzilla.kernel.org/attachment.cgi?id=295055

What do you make of it, @P33M? The aim send6 to be to avoid soft resets on some XHCI controllers.

wangnick, 2021-04-28:
Ok, I put arm_64bit=1 in config.txt and rebooted. Uptime now 6 days without any issues.

P33M, 2021-04-28:
The patch adds a quirk to not use the xhci soft-retry mechanism on an AMD platform. Normally, bulk endpoints have a Cerr parameter which is the number of retries that the controller performs automatically in the event of a corrupted packet before giving up (and flagging a transaction error). The soft retry clears the endpoint halt without advancing or disturbing transfer state, and a doorbell write sets the transaction off again.

The xhci driver error in the bugzilla is different to the ones here. There's no evidence to suggest the VLI controller can't handle an endpoint soft retry.

Artnal, 2021-04-28:
> Ok, I put arm_64bit=1 in config.txt and rebooted. Uptime now 6 days without any issues.

Was that on the 5.10 kernel? I might give it a shot if so.

wangnick, 2021-04-28:
> Was that on the 5.10 kernel? I might give it a shot if so.

Yes.
pi@database:~ $ uname -a
Linux database 5.10.23-v8+ #1406 SMP PREEMPT Mon Mar 15 15:49:01 GMT 2021 aarch64 GNU/Linux

maszaa, 2021-04-30:
Seems to be that `arm_64bit=1` doesn't solve the issue for me. 10 days uptime and the problem appeared again. Frustrating.

thebellenik, 2021-04-30:
> Seems to be that `arm_64bit=1` doesn't solve the issue for me. 10 days uptime and the problem appeared again. Frustrating.

Same here, it lasts longer than before (24-48 hours vs 10 days), but the same issue happened.

[256 more replies]